istoreos/istoreos #2216: 双wan口，指定端口流量从wan1走需要怎么设置

dddd64693, 2025-04-11:
由于PT原因导致原来的宽带ipv6地址被运营商收回了，现在新接了一条有ipv4的宽带，想把2条宽带利用起来，现在需要通过双wan并且指定PT下载器端口走原先宽带，需求如下：
路由网关：192.168.100.1
eth0：有固定IP宽带，设置了二级路由192.168.0.0/24网段的固定地址
eth1:原先的被回收ipv6地址的宽带，pppoe拨号
现在需要双wan口上网，并且指定PT下载器流量（端口55555）走原来的宽带也就是eth1，其他流量走固定IP宽带也就是eth1，请问需要如何设置。

[image]

[image]

feidaodu, 2025-04-21:
装个mwan啊。。就是那个“多线多拨”

jjm2473, 2025-05-21:
涉及到端口，需要防火墙打标，然后用策略路由根据标志路由。

如果你只是想让某个宽带专门下载，还有更容易理解的方式，就是istoreos不要使用下载专用的宽带，也就是不要配置任何协议。
假设eth1是专门下载的宽带，docker创建一个macvlan网络，将eth1作为macvlan的父接口，然后docker创建个容器使用这个网络，容器里面就可以拨号，并使用下载软件了。容器可以使用istoreos或者openwrt，方便拨号。

jjm2473, 2025-05-21:
我发现策略路由可以用uid匹配，应该只要让下载工具以特定的uid运行，就可以很简单配置策略路由了。网络-路由-ipv4规则，添加规则时，高级设置里的“用户标识符”