istoreos/istoreos #2217: 连接移远RM500U-CNV 5G模块后会在开机启动阶段崩溃

NonbiriKumo, 2025-04-12:
固件版本： iStoreOS 22.03.7 2025040711 

问题描述：移远RM500U-CNV 5G模块用USB连接，NCM模式，会在导致系统在启动阶段崩溃，启动完毕后再连接正常，但用AT命令触发模块重启也会导致系统崩溃

崩溃前控制台

[image]

jjm2473, 2025-04-18:
下次崩溃时提取日志，实在无法提取日志就拍视频。或者试试24.10测试固件。

Fly6688Fei, 2025-04-21:
我们这里测试RM520N-CN没问题

[image]

[image]

MA-Guodong, 2025-06-04:
> 我们这里测试RM520N-CN没问题
> 
> <img alt="Image" width="1920" src="https://private-user-images.githubusercontent.com/198524096/435522190-9586c305-38e5-4dc8-83a7-e614508b4612.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDkwMjEwMzMsIm5iZiI6MTc0OTAyMDczMywicGF0aCI6Ii8xOTg1MjQwOTYvNDM1NTIyMTkwLTk1ODZjMzA1LTM4ZTUtNGRjOC04M2E3LWU2MTQ1MDhiNDYxMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYwNFQwNzA1MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OTVkOTRlMmVjNzZjNTQwZTFlOTIzZTNhNzRhM2MwYjdiYzNkZDM0OTVkNDk0Mjc5MTdkZWUyNzUxMDUwYjQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iUWIu7o2EzJCF6CE3k5BXX4iY1LUp1SsZx8Uhn-8oLQ"> <img alt="Image" width="1920" src="https://private-user-images.githubusercontent.com/198524096/435522209-4ac77666-fc2d-46bf-80cf-1929cb01f052.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDkwMjEwMzMsIm5iZiI6MTc0OTAyMDczMywicGF0aCI6Ii8xOTg1MjQwOTYvNDM1NTIyMjA5LTRhYzc3NjY2LWZjMmQtNDZiZi04MGNmLTE5MjljYjAxZjA1Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYwNFQwNzA1MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ODI4NGQ5OWY5OTM3MTJhNjdlMzU1OWYyOGYyNzI1OWM4YjRmODE0YmEyM2ZkMmZkYWNlYWFkYjg0YmJjYThlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jyKRO8HiU9am1e0DKs25n_ja06qV3e__TFSzV6cs0x0">

你好，请问你用的QWRT模组管理是从哪里获得的

closed 2025-11-03 by NonbiriKumo