istoreos/istoreos #2218: istoreos系统，小米路由器中继模式会不定期无法加入网络

zhuzhu3456, 2025-04-13:
我的istoreos用小米路由器AP模式扩展与无线网络，WiFi 会不定期的无法加入网络。需要重启小米路由器才行

[image]

Cyber-H, 2025-04-13:
难怪我说怎么进主路由是正常的，但是用小米无线出来就是有时候会神经

alanzhu2019, 2025-04-15:
有找到原因吗

138161, 2025-04-20:
ipv6中继模式设置错误，因为优先选择走ipv6，一般是ipv6续约出现问题，一段时间后不通，546 547端口有没有开放，这两管续约问题的

138161, 2025-04-20:
NAT方式 --------------------------------------
方法1 - 重设路由表

注意，该方法依赖 owipcalc 包来计算子网地址： opkg install owipcalc 。
我们可以手动在 WAN 口获得 IPv6 地址后添加一条路由表，让整个子网重定向到 LAN 口，这个操作可以通过 OpenWrt 的 hotplug 机制来进行，保存以下脚本放在 /etc/hotplug.d/iface/80-reset-route6 并重启 WAN 接口即可：

#!/bin/sh
wan_dev="wan6"
[ "$HOTPLUG_TYPE" = "iface" ] || exit 0
[ "$INTERFACE" = "$wan_dev" ] || exit 0
RTMETRIC=127
. /lib/functions/network.sh
network_get_physdev lan_dev lan || exit 0
ifup_cb() {
        local _lan_dev="$1"
        local _metric="$2"
        local wan_subnet
        network_get_subnet6  wan_subnet  "$wan_dev" || return
        _wan_network=$(owipcalc "${wan_subnet}" network)
        ip -6 route replace "$_wan_network" dev "$_lan_dev" metric "$_metric"
}
ifdown_cb() {
        local _lan_dev="$1"
        local _metric="$2"
        ip -6 route flush dev "$_lan_dev" metric "$_metric"
}
case "$ACTION" in
        ifup)
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifdown)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifupdate)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                sleep 1
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        *)
                ;;
esac
exit 0
#ip -6 route replace "240e:38a:1999:3e00::/64" dev "br-lan" metric "127"
#路由表添加  240e:38a:1999:3e00::/64 dev br-lan metric 127 pref medium
存在的问题：当光猫ipv6改变，ipv4未改变，需要重启软路由来更新此步骤，原因是运营商ipv6分配的地址会改变，但ipv4地址是局域网地址，不会变，光猫重启就可能会出现这样的场景。
 
第零步，安装软件包
0.1 系统--软件包--更新列表
      过滤器输入：owipcalc
     安装
0.2 编辑 80-reset-route6文件，，内容见上面，拷贝到 /etc/hotplug.d/iface/80-reset-route6
0.3 赋予执行权限 777
chmod +X /etc/hotplug.d/iface/80-reset-route6

第一步，先搞定wan6口
1、网络--接口--wan6--编辑
1.1、常规设置
     协议：DHCPv6客户端
         设备：eth0
         开机自动运行：选择
         请求IPv6地址：try
         请求指定长度的IPv6前缀：自动
       
1.2、高级设置
         强制链路：不选
         请求DHCP时发送的客户端ID：空
         使用默认网关：选择
         自动获取DNS服务器：选择
         DNS 权重：空
     使用网关跃点：空
[... 230 lines cut ...]
        option interface 'lan' #端口名称
        option start '100'    #端口号起始--结束
        option limit '150'
        option dhcpv4 'server'
        option leasetime '12h'  # infinite 租期，v4v6均有效
        option ra 'relay'
        option ndp 'relay'        #中继运行模式
        option dhcpv6 'relay'
 #存在下级路由器时，将NDP消息扩展到下级路由
config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config dhcp 'wan6'
        option interface 'wan6'
        option master '1'
        option ra 'relay'
        option ndp 'relay'
        option dhcpv6 'relay'

------------------------------------------------------

6.7、passwall插件，DNS设置，过滤代理域名 IPv6，要勾选，不对IPV6解析，使得IPV6直接连通，不经过passwall。

----------------------------------------

zhuzhu3456, 2025-04-22:
> 方法1 - 重设路由表
> 
> 注意，该方法依赖owipcalc包来计算子网地址：opkg安装owipcalc。我们可以手动在WAN口获得IPv6地址后添加一条路由表，让整个子网重定向到LAN口，这个操作可以通过OpenWrt的热插拔机制来进行，保存以下脚本在/etc/hotplug.d/iface/80-reset-route6并重启WAN接口即可：
> 
> #！/bin/shwan_dev="wan6"[ "$HOTPLUG_TYPE" = "iface" ] || 退出0[ "$INTERFACE" = "$wan_dev" ] || 退出 0RTMETRIC=127./lib/功能/network.shnetwork_get_physdev lan_dev lan || 退出0ifup_cb（）{本地_lan_dev="$1"本地_metric="$2"本地wan_subnetnetwork_get_subnet6 wan_subnet "$wan_dev" || 返回_wan_network=$（owipcalc“${wan_subnet}”网络）ip -6路由替换“$_wan_network”开发“$_lan_dev”指标“$_metric”}Ifdown_cb（）{本地_lan_dev="$1"本地_metric="$2"ip -6 路由刷新 dev "$_lan_dev" 指标 "$_metric"}案例“$ACTION”在Ifup）ifup_cb“$lan_dev”“$RTMETRIC”;;如果下降）ifdown_cb“$lan_dev”“$RTMETRIC”;;如果更新）ifdown_cb“$lan_dev”“$RTMETRIC”睡眠1ifup_cb“$lan_dev”“$RTMETRIC”;;*）;;萨克出口0#ip -6 路由替换“240e:38a:1999:3e00::/64” dev “br-lan” 度量“127”#路由表添加 240e:38a:1999:3e00::/64 dev br-lan metric 127 pref 中等存在的问题：当光猫ipv6改变，ipv4未改变，需要重启软路由来更新此步骤，原因是运营商ipv6分配的地址会改变，但ipv4地址是局域网地址，不会变，光猫重启可能会出现这样的场景。
> 
> 第零步，安装软件包0.1 系统--软件包--更新列表过滤器输入：owipcalc安装0.2 编辑 80-reset-route6文件，，内容见上面，拷贝到 /etc/hotplug.d/iface/80-reset-route60.3 赋予执行权限 777chmod +X /etc/hotplug.d/iface/80-重置路线6
> 
> 第一步，先搞定wan6口1、网络--接口--wan6--编辑1.1、常规设置协议：DHCPv6客户端设备：eth0开机自动运行：选择请求IPv6地址：尝试请求指定长度的IPv6前缀：自动
> 
> 1.2、高级设置强制链路：不选请求DHCP时发送的客户端ID：空使用默认网关：选择自动获取DNS服务器：选择DNS 权重：空使用网关跃点：空覆盖IPv4路由表：未指定覆盖IPv6路由表：未指定IPv6源路由：选择委托IPv6前缀：选择IPv6分配长度：已禁用IPv6 前缀过滤器：--请选择--IPv6 后缀：::1IPv6优先级：空
> 
> 1.3、防火墙设置创建和分配防火墙区域：wan和wan6
> 
> 1.4、DHCP服务器--常规设置忽略这个接口：不选
> 
> 1.5、DHCP服务器--IPv6设置指定的主接口：选择RA服务：中继模式DHCPv6服务：中继模式NDP代理：中继模式学习路线：选择
> 
> 第二步，搞定兰口2、网络--接口--lan--编辑2.1、常规设置协议：静态地址设备：br-lan开机自动运行：选择IPv4 地址：192.168.100.1IPv4子网掩码：255.255.255.0IPv4 网关：灰色无法更改IPv4广播地址：灰色无法更改IPv6地址：空IPv6网关：空IPv6路由前缀：空
> 
> 2.2、高级设置强制链路：选择使用默认网关：选择使用自定义的DNS服务器：空DNS搜索域名：空DNS 权重：空使用网关跃点：空覆盖IPv4路由表：未指定覆盖IPv6路由表：未指定委托IPv6前缀：选择IPv6分配长度：禁用IPv6 前缀过滤器：--请选择--IPv6 后缀：::1IPv6优先级：空
> 
> 2.3、防火墙设置创建和分配防火墙区域：lan
> 
> 2.4、DHCP服务器--常规设置忽略这个接口：不选启动：100客户数：150租期：无限
> 
> 2.5、DHCP服务器--高级设置动态DHCP：选择强制：不选IPv4子网掩码：灰色无法更改DHCP选项：空
> 
> 2.6、DHCP服务器--IPv6设置指定的主接口：不选RA服务：中继模式DHCPv6服务：中继模式NDP代理：中继模式学习路线：选择NDP代理从属设备：不选
> 
> 第三步，网络--接口--全局网络选项IPv6 ULA 前缀：空数据包引导：选择
> 
> 第四步，网络--DHCP/DNS--高级设置过滤 IPv6 AAAA 记录：不选过滤IPv4 A记录：不选
> 
> 第五步，网络--防火墙——自定义规则
> 
> # qbittorrent 19422 tcp udp
> 
> #ip6tables -I input_wan_rule -i eth0 -p tcp --dport 19422 -j 接受#ip6tables -I input_wan_rule -i eth0 -p udp --dport 19422 -j 接受
> 
> # qbittorrent 6883 tcp udp
> 
> ip6tables -I input_wan_rule -i eth0 -p tcp --dport 6883 -j 接受ip6tables -I input_wan_rule -i eth0 -p udp --dport 6883 -j 接受
> 
> # 网络Dav 6086
> 
> ip6tables -I input_wan_rule -i eth0 -p tcp --dport 6086 -j 接受
> 
> # Aira2 6881,6882
> 
> ip6tables -I input_wan_rule -i eth0 -p tcp --dport 6881 -j 接受ip6tables -I input_wan_rule -i eth0 -p udp --dport 6881 -j 接受ip6tables -I input_wan_rule -i eth0 -p tcp --dport 6882 -j 接受ip6tables -I input_wan_rule -i eth0 -p udp --dport 6882 -j 接受
> 
> #ip6tables -I zone_lan_forward -p ipv6-icmp -j 日志
> 
> # 放行入站（wan，lan，docker，单向进入），icmpv6续约
> 
> ip6tables -I input_rule -p udp --sport 546 --dport 547 -j 接受
> 
> # 放行转发（光猫与下级PC，双向），icmpv6续约
> 
> #ip6tables -I forwarding_rule -p udp --sport 546 --dport 547 -j 接受
> 
> ip6tables -I input_rule -p ipv6-nonxt -m length --length 40 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 1 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 2 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 3 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 4 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 128 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 129 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 130 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 131 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 132 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 133 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 134 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 135 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 136 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 137 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 141 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 142 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 143 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 148 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 149 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 151 -m limit --limit 100/s --limit-burst 10 -j 接受ip6tables -I input_rule -p ipv6-icmp --icmpv6-type 152 -m limit --limit 100/s --limit-burst 10 -j ACCEPTip6tables -I input_rule -p ipv6-icmp --icmpv6-type 153 -m limit --limit 100/s --limit-burst 10 -j 接受
> 
[... cut ...] nat 1.8.7-7ip6tables-zz-遗产1.8.7-7kmod-ip6tables 5.10.201-1 不是必须，提供IPv6防火墙kmod-nf-ipt6 5.10.201-1kmod-ipt-nat6 提供IPv6的NAT支持netifd OpenWrt网络接口配置守护程序owipcalc 计算子网地址6.6、文件/etc/config/dhcp如下配置dhcp“lan”选项接口'lan' #端口名称选项开始'100' #端口号起始--结束期权限制“150”选项dhcpv4“服务器”选项租赁时间“12h”#无限租赁期，v4v6均有效选项ra“中继”选项ndp“中继”#中继运行模式选项dhcpv6“中继”#存在下级路由器时，将NDP消息扩展到下级路由配置dhcp“wan”选项界面“wan”选项忽略“1”
> 
> 配置dhcp“wan6”选项接口“wan6”选项主“1”选项ra“中继”选项ndp“继电器”选项dhcpv6“中继”
> 
> 6.7、passwall插件、DNS设置、过滤代理域名IPv6、要勾选、不IPV6解析，使IPV6直接连通，不经过passwall。