istoreos/istoreos #2219: BUG 静态地址分配无效

kjhgy2, 2025-04-14:
硬件：istoreOS 官方 EasePi R1
版本：
[image]

问题： 静态地址分配添加了以后，应用并更新后，IP地址无改动。

[image]

尝试过的方法：重启系统、重启dnsmasq、重启连接在R1上的硬件，设置更短的租期等待更新后IP更新

IP均无跟新，且不止一人存在这个问题

BusangSeven, 2025-04-15:
etc/config/dhcp 有dhcpv4 ‘disabled’吗

kjhgy2, 2025-04-15:
> etc/config/dhcp 有dhcpv4 ‘disabled’吗

无，目前测试过可行的方法仅有重启路由器或者重启终端设备才可让指定的IP地址更新

[image]

Nillforyou, 2025-04-16:
遇到了相同的问题，而且根据我的测试，还是个偶发问题，现在静态地址分配中有一半生效一半不生效。离谱吧。

jjm2473, 2025-05-16:
dnsmasq有个地址分配缓存，如果分配还没过期（默认12小时），会一直使用。重启系统就可以清除缓存。然后客户端应该要重新请求dhcp。

DeMeng33, 2025-09-09:
> [@DeMeng33](https://github.com/DeMeng33) 那你要确认电脑的mac地址是不是变了，以及电脑自己是不是设置了静态ip

感谢回复。都检查过，macOS的wifi连接中设置固定mac，ip来源是DHCP。

和楼主截图结果一样重启过电脑、软路由、ap。额外尝试过手动改dhcp.conf也无效。

BusangSeven, 2025-09-14:
不好意思，我才收到提醒。已经忘记咋解决的了。我记忆里回的时候是改我之前那个设置可以解决，加还是减有效不记得了。不更新好像是用发送HUP还是SIGUSR1命令，比重启这个服务有效。然后同时有另一个bug，就是映射域名的配置文件会被截断最后一部分，导致配置文件出错，更新该服务的配置会不完整。这个我完全忘记了，可能跟上一个bug有关联。

jjm2473, 2025-09-17:
@DeMeng33 确定局域网就只有一个dhcp服务？

DeMeng33, 2025-09-18:
> [@DeMeng33](https://github.com/DeMeng33) 确定局域网就只有一个dhcp服务？

是的，确定。mwan3的wan口都忽略了dhcp，网内路由器是有线桥接的。

jjm2473, 2025-09-18:
@DeMeng33 我不是说istoreos自身有多个dhcp服务，我是说局域网内其他机器有没有提供dhcp服务，例如wifi路由器。最简单的测试方式就是断开istoreos的lan口，看看原本出问题的设备还能不能获取到ip。

jjm2473, 2025-12-15:
注意DHCP默认的动态分配地址池是从100到250（例如192.168.100.100-192.168.100.250），如果要完全避免静态分配跟动态分配冲突，静态分配使用100以下的地址（例如192.168.100.2-192.168.100.99）。