istoreos/istoreos #2222: R4se软路由启动分身以后网页显示不完全

lff110, 2025-04-16:
R4se软路由启动分身以后网页显示不完全，每次都是这样显示的，导致后续操作无法完成

[image]

jjm2473, 2025-04-18:
电脑浏览器打开看看，如果还是有问题，打开浏览器的console，看看有什么报错没

lff110, 2025-04-18:
> 电脑浏览器打开看看，如果还是有问题，打开浏览器的console，看看有什么报错没

刚开始就是用电脑发现这个问题的，后面发现用手机也有相同问题

lff110, 2025-04-20:
> 电脑浏览器打开看看，如果还是有问题，打开浏览器的console，看看有什么报错没

刚刚看了是这个错误

[image]

jjm2473, 2025-04-22:
截图这不是错误。

你这种情况是全新刷机也会出现吗？固件版本是什么？

lff110, 2025-04-22:
以前还以为是已经用了好久才出现的，上周一左右重新刷了一下还是这样，第一张图就是刚刷机以后截图发的，现在是r4se最新版固件
[image]

lff110, 2025-04-30:
我刚刚看到光猫并未给分身分配IP地址，不知道和这个有没有关系，另外下面的内容是我刚刚在istoredup容器检查中看到的内容，不知道能不能通过这个定位到问题所在
{
  "Path": "\/sbin\/init",
  "ProcessLabel": "",
  "ResolvConfPath": "\/mnt\/usb8-1\/docker\/containers\/cf524e449f582e93d9a850e2b5e42f7bda03cb121942fd108f052dedb85721a1\/resolv.conf",
  "NetworkSettings": {
    "LinkLocalIPv6Address": "",
    "SandboxID": "f4af6c45a064b7a3fe5c1bfd1c4494a279c6f102266dd990b2cd05725e9b9e9d",
    "HairpinMode": false,
    "Networks": {
      "dsm-net": {
        "NetworkID": "e85c94e84d07d1a761c3e5ae461525fff93b55cd1862c04061272f9dd6d77e5e",
        "IPAddress": "172.18.0.2",
        "Aliases": [
          "cf524e449f58",
          "iStoreDuplica"
        ],
        "MacAddress": "02:42:ac:12:00:02",
        "IPv6Gateway": "",
        "IPPrefixLen": 16,
        "Gateway": "172.18.0.1",
        "GlobalIPv6PrefixLen": 0,
        "EndpointID": "252f74da5e0bbf6e13cacb02948c6f8de649f12c4a3bc7305e76ba646f8affe8",
        "GlobalIPv6Address": ""
      }
    },
    "Ports": [
    ],
    "Bridge": "",
    "MacAddress": "",
    "IPv6Gateway": "",
    "IPPrefixLen": 0,
    "IPAddress": "",
    "EndpointID": "",
    "SandboxKey": "\/var\/run\/docker\/netns\/f4af6c45a064",
    "Gateway": "",
    "GlobalIPv6PrefixLen": 0,
    "LinkLocalIPv6PrefixLen": 0,
    "GlobalIPv6Address": ""
  },
  "ExecIDs": [
    "1487b5ea8f0057021805d71c95b87c4db7ee59ca3656d999f5e7508c3ceb79a6"
  ],
  "MountLabel": "",
  "HostsPath": "\/mnt\/usb8-1\/docker\/containers\/cf524e449f582e93d9a850e2b5e42f7bda03cb121942fd108f052dedb85721a1\/hosts",
  "LogPath": "\/mnt\/usb8-1\/docker\/containers\/cf524e449f582e93d9a850e2b5e42f7bda03cb121942fd108f052dedb85721a1\/cf524e449f582e93d9a850e2b5e42f7bda03cb121942fd108f052dedb85721a1-json.log",
  "RestartCount": 0,
  "Config": {
    "AttachStdout": false,
    "StopTimeout": 120,
    "AttachStdin": false,
    "Labels": {
      "org.opencontainers.image.source": "https:\/\/github.com\/linkease\/istoreos-in-docker",
      "org.opencontainers.image.version": "22.03",
      "org.opencontainers.image.created": ""
    },
    "Tty": false,
    "AttachStderr": false,
    "OpenStdin": false,
    "Cmd": [
      "\/sbin\/init"
    ],
    "Image": "linkease\/istoreduprk35xx:latest",
    "User": "",
    "Hostname": "iStoreDuplica",
    "Env": [
      "PATH=\/usr\/local\/sbin:\/usr\/local\/bin:\/usr\/sbin:\/usr\/bin:\/sbin:\/bin"
    ],
    "Domainname": "",
    "WorkingDir": "\/",
[... 105 lines cut ...]
    },
    "Name": "overlay2"
  },
  "State": {
    "Pid": 21201,
    "FinishedAt": "2025-04-30T15:03:24.144430297Z",
    "StartedAt": "2025-04-30T15:03:37.602442262Z",
    "Error": "",
    "Running": true,
    "Paused": false,
    "OOMKilled": false,
    "Status": "running",
    "ExitCode": 0,
    "Restarting": false,
    "Dead": false
  },
  "Driver": "overlay2",
  "Name": "\/istoredup",
  "Args": [
  ],
  "HostnamePath": "\/mnt\/usb8-1\/docker\/containers\/cf524e449f582e93d9a850e2b5e42f7bda03cb121942fd108f052dedb85721a1\/hostname",
  "Created": "2025-04-20T17:12:23.572653514Z",
  "AppArmorProfile": "",
  "Image": "sha256:a5d907083a3d3512e6a45607e3c7aed472328b6783a62c4694bb224ce6a7f35e"
}

lff110, 2025-05-23:
> 如果可以远程的话，联系方式发我邮箱 [jjm2473@gmail.com](mailto:jjm2473@gmail.com)

刚看到，已经发了