istoreos/istoreos #2227: 20250403、0407更新破坏了我的IPv6体验

Small-UU, 2025-04-17:
网络环境：
河南联通，光猫桥接、物理机iSOS拨号（x86）、华硕路由器AP模式
（装有Passwall2 25.4.11，但未开启）

问题描述：
以前硬路由用IPv6好几年了正常，后来软路由从24年初用小宝大佬视频（BV1mT4y1b73p）一步步配置IPv6也用了一年左右正常。期间一直更新iSOS保持最新也没问题。

25年初更新到20241227版本，也一切正常。后来农历年期间想返璞归真就把软路由重置了出厂，偷懒用小宝大佬新视频（BV1MRraYDES4）里的系统便利工具设置了IPv6（Full不行、Full Try2也不行、NAT3选项成功了）。然后用了俩月IPv6一直正常。
系统只改了密码、开启UPnP、FC NAT选高性能、应用商店装了个“访问AP/光猫”开启了Wan访问光猫后台，用完就关闭了访问。

后来更新到0403版本就不对劲了，更新完软路由重启所有设备无法联网，也进不去后台。手动重启软路由好了，以前更新没遇到过。然后看公告马上更新了0407，这次更新完正常了。但testipv6.cn测试“双栈域名大数据包传输测试”有概率通不过，用系统便利工具（Full、Full 2、NAT3顺序一遍还是NAT3能用）重新开一遍v6也这样。
全家手机（iOS 18最新）打开淘宝、京东App首页下拉加载商品（图片、视频、详情）都慢的很。手机浏览器看IT之家首页打开慢、加载新闻图片也慢，有时候甚至打不开it之家、百度首页，得刷新一下。

尝试解决：
起初怀疑是dns问题，首页改了阿里dns也这样，所以恢复了自动获取。然后用系统便利工具执行了“关闭IPv6”，马上正常了全部秒开。

运营商应该也是正常的，同样河南联通，公司的硬路由一切正常也有IPv6。
看这个表现有点像网上说的IPv6 PMTU黑洞的问题（https://zhuanlan.zhihu.com/p/621371177）但找不到防火墙页面有WAN口的“MSS钳制”MSS Clamping选项框，不过iSOS默认应该是打开的吧？

猜想：
“临时更新 20250407：修复上个版本 WAN 如果从默认的 DHCP 客户端切换到 PPPoE，LAN 不会分配 IPv6 的问题（“系统便利工具”也能解决此问题）。”
“IPv6: 默认情况下，IPv6是中继模式，LAN 口可能分配到 ULA 地址，对于某些系统如macOS，可能不会响应 ULA 来源的 NS 请求，导致无法联网（这时Ping一下网关就通网了）。现在改为 LAN 口默认只能从 WAN6/WAN_6 的 PD 地址分配。 一般情况下，即使 WAN6 没有 PD 地址，LAN 口无 IPv6 地址，局域网内还是能分配到 IPv6 地址，无需 NAT6。 如果的确需要配置 NAT6，请使用 iStore 提供的最新版“系统便利工具”；如手动配置 NAT6 或者其他原因 LAN 需要 ULA，将 LAN 口高级设置“IPv6 前缀过滤器”列表清空即可。”

看20250407更新日志动了IPv6相关，所以怀疑这些更改造成了无法预料的bug。我暂时关闭IPv6使用几天了一切正常。特此提出反馈，希望对大家还有开发者有所帮助！

haobinnan, 2025-04-18:
uci del network.lan.ip6class
uci commit network

执行后就和以前一样了，你可以试试。

jjm2473, 2025-04-18:
系统便利工具 更新到最新版了吗？

Small-UU, 2025-04-18:
> 系统便利工具 更新到最新版了吗？

更新完系统就顺手更新了，确定是最新的情况下操作的。

另外更正一下：testipv6.cn是“ IPv6 大数据包传输测试”有概率通不过，当时应该是记错了。但www.test-ipv6.com基本每次都能满分，给我整迷糊了……

closed 2025-04-18 by Small-UU

Small-UU, 2025-04-18:
另外，24年初小宝大佬视频（BV1mT4y1b73p）。
按这个配置是IPv6 Full么？为什么系统便利工具的Full反而不能用？
还有我记得24年初刚装好iSOS的时候，发现把DNS设置里的“屏蔽AAA记录”关了就能直接使用v6。这种什么模式？
这几个能用的（默认只打开AAA解析、小宝24年初视频配置、系统便利工具里的NAT3模式）哪个更好？

jjm2473, 2025-04-22:
除了NAT最差，其他都一样，只是不同场景的配置方式不一样而已。

Small-UU, 2025-04-22:
> 除了NAT最差，其他都一样，只是不同场景的配置方式不一样而已。

感谢解答🙏
请问我这种情况目前有头绪了吗？会在后续24.10版本修复？
还是我需要采取什么操作？或者提供更多信息？

现在关闭IPv6的情况下，如果我想再开启并且不用最差的NAT模式的话，需要怎么操作？
先用系统便利工具的Full开一遍，再按24年初小宝大佬视频（BV1mT4y1b73p）操作？
或者直接重置配置到默认状态，然后打开dns的AAA解析？

Small-UU, 2025-04-23:
> 不用再看视频，重置以后先配置好上网，然后用系统便利工具的Full开一遍，最后尝试NAT，不行就不行了。
> 
> 如果你自己配置了上网的接口，不是默认的wan，那把你的`/etc/config/network`提取出来看看。
> 
> 还可以执行这两个命令恢复到之前固件的配置：
> 
> ```
> uci del network.lan.ip6class
> uci commit network
> ```

我自己没有动过接口配置，只用过系统当时版本最新的系统便利工具操作过。如果暂时无解我就先用着不折腾了，等24.10正式版一起重置吧，才用了几个月再折腾一遍挺伤的。😭