istoreos/istoreos #2230: 部分菜单打开报错_(...).format is not a function

GoodBoyCoder, 2025-04-19:
使用固件版本：istoreos-22.03.7-2025040711-x86-64-squashfs-combined.img.gz（无额外安装插件）

打开部分菜单会出现_(...).format is not a function异常导致页面空白，使用页面debug调试一步步放行后有概率正常，下面是一些debug截图信息

[image]

[image]

jjm2473, 2025-04-30:
清空浏览器缓存还会出现吗？是否有资源加载失败

AZwei-X, 2025-04-30:
[image]
我也碰到这个问题，清空浏览器缓存没用，很影响使用体验！

jjm2473, 2025-05-16:
只有网络-接口页面有这个问题吗？浏览器的无痕模式下，或者其他浏览器会有这个问题吗？

AZwei-X, 2025-05-18:
这个问题只能ctrl+f5解决，直接f5刷新或者变更后提交很容易出现我说的页面刷新出错的问题。我不知道什么时候这个问题才出现的，不知道istoreOS有做过什么更新导致这个问题出现，还是windows或者chrome更新导致的问题。以前没碰到过，但是我一直维持软件更新到最新，具体是什么导致的无法找到根源。

> 只有网络-接口页面有这个问题吗？浏览器的无痕模式下，或者其他浏览器会有这个问题吗？