istoreos/istoreos #2231: r66s在原iStoreOS 22.03.5基础上“升级备份”刷入24.10 alpha r6xs固件后，passwall、istore软件商店等luci界面报错

ralopa673, 2025-04-19:
前情：r66s在原iStoreOS 22.03.5基础上“升级备份”刷入24.10 alpha r6xs固件（保留配置+其他默认）后，passwall、openclash、kms，以及istore软件商店、docker管理页等插件的luci界面全部出错。浏览器界面显示如下：

```
Runtime error
Unhandled exception during request dispatching

/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/http.lua:145: attempt to index field 'request' (a nil value)

In error(), file [C]
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
  called from function  ((tail call))

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:803:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:777:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:804:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1017:48)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:840:78)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1017:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

```

而且在系统—软件里卸载掉passwall等上述插件，并通过.run格式重新安装一遍后，报错问题依旧存在。

ralopa673, 2025-04-19:
另外，也不是所有含luci的插件都出错，例如tailscale、动态 DNS、frp客户端、网络唤醒、samba等插件仍然正常。

而且，虽然docker LuCI的概览、容器打开报错，但原本挂载的docker容器仍然正常运转，且可以通过命令行正常操作。

jjm2473, 2025-04-19:
估计你以前自己在软件包升级过luci-compat之类的软件包，导致不兼容了。

ralopa673, 2025-04-19:
> 估计你以前自己在软件包升级过luci-compat之类的软件包，导致不兼容了。

我印象中是没有的。因为这个r66s我闲置很久了，目前是挂在另一个房子里，过年期间重刷了iStoreOS后就只装过几个常用插件。从过年到现在，几乎就没怎么动过它，也几乎不用它，所以今天看到有24.10版本就抱着got nothing to lose的心态立马刷了。

ralopa673, 2025-04-19:
> 估计你以前自己在软件包升级过luci-compat之类的软件包，导致不兼容了。

问了Claude 3.7，按它报告的流程做了一遍，问题终于解决。

————

如果后面有谁遇到问题，可以如下操作：

1️⃣ 首先，重新安装 LuCI 的核心组件和 ucode 桥接模块：

```
    opkg remove --force-depends luci-base
    opkg remove --force-depends luci-ucodebridge
    opkg update
    opkg install luci-base luci-ucodebridge
```
2️⃣ 修复 LuCI 的依赖关系：
```
    opkg install --force-reinstall luci-lib-base luci-lib-ip luci-lib-jsonc luci-lib-nixio
```

3️⃣ 重新安装所有 LuCI 应用：
```
    opkg install --force-reinstall luci luci-mod-admin-full luci-theme-bootstrap
```

4️⃣ 清除所有 LuCI 缓存：
```
    rm -rf /tmp/luci-*
```

5️⃣ 重启所有相关服务：
```
    /etc/init.d/uhttpd restart
    /etc/init.d/rpcd restart
```

closed 2025-04-20 by ralopa673

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/1734/4

rayyume, 2026-08-19:
mark