istoreos/istoreos #2234: 【24.10测试】升级后web界面无法打开

chinesepy, 2025-04-19:
X86 固件，22.03.07 升级 24.10 
使用istoreos-24.10.0-2025041811-x86-64-squashfs-combined-efi.img.gz 在网页端进行升级，选择了保存当前配置
升级后系统启动正常，可以上网，但web页面无法打开 提示ERR_CONNECTION_REFUSED

查看日志发现大量nginx报错日志
 /usr/sbin/nginx:  symbol not found

搜索下尝试按照https://github.com/istoreos/istoreos/discussions/860 中的步骤重新安装 nginx ，安装后网页可以打开，但除了网络，系统等几个模块外其他页面包括自行安装的插件打开均显示空白或错误。

[image]

请教一下应该如何处理？我印象中默认openwrt是用uhttpd的，可以切回去吗？

chinesepy, 2025-04-19:
附上目前日志中的err 
root@P5-N100:/etc/backup# logread |grep err
Sun Apr 20 07:17:50 2025 kern.err kernel: [    6.470458] kvm_amd: CPU 3 isn't AMD or Hygon
Sun Apr 20 07:17:50 2025 kern.info kernel: [    6.700294] usbcore: registered new interface driver sierra
Sun Apr 20 07:17:50 2025 kern.info kernel: [    6.700320] usbserial: USB Serial support registered for Sierra USB modem
Sun Apr 20 07:17:50 2025 kern.info kernel: [    6.863517] usbcore: registered new interface driver sierra_net
Sun Apr 20 07:17:50 2025 kern.err kernel: [    7.463208] ccp_crypto: Cannot load: there are no available CCPs
Sun Apr 20 07:17:50 2025 kern.err kernel: [    7.682499] kvm_amd: CPU 2 isn't AMD or Hygon
Sun Apr 20 07:17:50 2025 user.err kernel: [    7.788112] kmodloader: 3 modules could not be probed
Sun Apr 20 07:17:50 2025 user.err kernel: [    7.788139] kmodloader: - ccp-crypto - 0
Sun Apr 20 07:17:50 2025 user.err kernel: [    7.788159] kmodloader: - it87 - 0
Sun Apr 20 07:17:50 2025 user.err kernel: [    7.788167] kmodloader: - kvm-amd - 0
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:53 2025 daemon.err rmmod: module is not loaded
Sun Apr 20 07:17:54 2025 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Sun Apr 20 07:17:54 2025 daemon.notice procd: /etc/rc.d/S45nfs: exportfs: could not open /var/lib/nfs/.etab.lock for locking: errno 2 (No such file or directory)
Sun Apr 20 07:17:54 2025 cron.err crond[9093]: crond (busybox 1.36.1) started, log level 8
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: [uWSGI] getting INI configuration from /etc/uwsgi/emperor.ini
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: *** Starting uWSGI 2.0.26 (64bit) on [Sun Apr 20 07:17:54 2025] ***
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: compiled with version: 13.3.0 on 14 April 2025 17:32:05
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: os: Linux-6.6.73 #0 SMP Thu Apr 17 12:06:28 2025
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: nodename: P5-N100
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: machine: x86_64
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: clock source: unix
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: pcre jit disabled
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: detected number of CPU cores: 4
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: current working directory: /
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: writing pidfile to /var/run/uwsgi.pid
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: detected binary path: /usr/sbin/uwsgi
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: *** starting uWSGI Emperor ***
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: [uwsgi-emperor] /etc/uwsgi/vassals/luci-cgi_io.ini -> "on demand" instance detected, waiting for connections on socket "/var/run//luci-cgi_io.socket" ...
Sun Apr 20 07:17:54 2025 daemon.err uwsgi[9358]: [uwsgi-emperor] /etc/uwsgi/vassals/luci-webui.ini -> "on demand" instance detected, waiting for connections on socket "/var/run//luci-webui.socket" ...
Sun Apr 20 07:17:55 2025 daemon.notice procd: /etc/rc.d/S93startdhns: run command failed, err= dial unix /var/run/quickstart/local.sock: connect: no such file or directory
Sun Apr 20 07:17:55 2025 daemon.crit smartd[10280]: Configuration file /etc/smartd.conf has fatal syntax errors.
Sun Apr 20 07:17:55 2025 daemon.err nmbd[10471]: [2025/04/20 07:17:55.810325,  0] ../../source3/nmbd/nmbd.c:901(main)
Sun Apr 20 07:17:55 2025 daemon.err nmbd[10471]:   nmbd version 4.18.8 started.
Sun Apr 20 07:17:55 2025 daemon.err nmbd[10471]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Sun Apr 20 07:17:55 2025 daemon.err smbd[10470]: [2025/04/20 07:17:55.870081,  0] ../../source3/smbd/server.c:1746(main)
Sun Apr 20 07:17:55 2025 daemon.err smbd[10470]:   smbd version 4.18.8 started.
Sun Apr 20 07:17:55 2025 daemon.err smbd[10470]:   Copyright Andrew Tridgell and the Samba Team 1992-2023
Sun Apr 20 07:17:56 2025 daemon.err uwsgi[9358]: *** has_emperor mode detected (fd: 7) ***
Sun Apr 20 07:17:56 2025 daemon.err uwsgi[9358]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
Sun Apr 20 07:17:56 2025 daemon.err uwsgi[9358]: Sun Apr 20 07:17:56 2025 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Sun Apr 20 07:17:56 2025 daemon.err uwsgi[9358]: Sun Apr 20 07:17:56 2025 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Sun Apr 20 07:17:56 2025 daemon.err uwsgi[9358]: Sun Apr 20 07:17:56 2025 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sun Apr 20 07:17:57 2025 daemon.err nmbd[10471]: [2025/04/20 07:17:57.131895,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Sun Apr 20 07:17:57 2025 daemon.err nmbd[10471]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.1 for name WORKGROUP<1d>.
Sun Apr 20 07:17:57 2025 daemon.err nmbd[10471]:   This response was from IP 192.168.1.200, reporting an IP address of 192.168.1.200.
Sun Apr 20 07:17:57 2025 daemon.err nmbd[10471]: [2025/04/20 07:17:57.132256,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
[... cut ...]
Sun Apr 20 07:45:51 2025 daemon.err uwsgi[9358]: [uwsgi-emperor] /etc/uwsgi/vassals/luci-webui.ini -> back to "on demand" mode, waiting for connections on socket "/var/run//luci-webui.socket" ...
Sun Apr 20 07:47:57 2025 daemon.err nmbd[12730]: [2025/04/20 07:47:57.112113,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Sun Apr 20 07:47:57 2025 daemon.err nmbd[12730]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.1 for name WORKGROUP<1d>.
Sun Apr 20 07:47:57 2025 daemon.err nmbd[12730]:   This response was from IP 192.168.1.200, reporting an IP address of 192.168.1.200.

wzken, 2025-04-20:
备份配置后执行重置

ralopa673, 2025-04-20:
这不就是我昨天的同款bug, #2231 里面有解决方案。不需要重置

chinesepy, 2025-04-20:
> 这不就是我昨天的同款bug, [#2231](https://github.com/istoreos/istoreos/issues/2231) 里面有解决方案。不需要重置

对我来说不太行，按照这个命令执行一遍后luci更是炸得透彻，直接连首页都狂报错了。
不过之前就已经对跨版本升炸了情况有心里预期了，之前也把clash的yaml，lucky，dhcp、防火墙配置文件，wireguard密钥都备份了，重置对docker也无影响 ，直接reset 了

closed 2025-04-20 by chinesepy