istoreos/istoreos #2237: istoreos拨号,硬路由连接软路由,下级设备分配不到公网ipv6

zhongziqiang, 2025-04-20:
设备:HINLINK OPC-H68K  刷istoreos
硬路由:Redmi路由器 AX3000

连接方式是:光猫桥接,软路由istoreos拨号,下级小米硬路由连接软路由,终端设备连接硬路由

软路由能获取ipv6地址,以下是软路由的ip信息

[image]

[image]

以下是软路由设置的截图

[image]

[image]

[image]

[image]

最终我的硬路由的lan获取不到ipv6,导致,连接硬路由的设备都没有ipv6地址,以下是我路由器ip信息

[image]

求官方大佬修复这个ipv6无法分配的问题

138161, 2025-04-20:
后缀64位情况下，无法修复，不要说小米，大神都没办法

138161, 2025-04-20:
最上层，就是拨号得到的前缀要小于64位，下面一级级设备才有可能全部获得ipv6地址，
如果一开始就是64位，代表下级都是终端设备，无法再为下下级（就是你现在这样）分配ipv6地址，如果要强行分配，只有使用ipv6的NAT方式（小米路由器wan口得到240e开头的地址，其下级得到fe80开头的地址），但实现这个功能，要修改小米路由器的固件系统，根本没办法的

zhongziqiang, 2025-04-21:
> 后缀64位情况下，无法修复，不要说小米，大神都没办法

istoreos最新固件,我试过设置成64位和60位都无法获得ipv6的公网,现在我刷了lean大佬的固件,已经能正确获取ipv6公网了,我什么都没动,我甚至在之前试过在istoreos上按照lean固件的设置来一模一样的设置都无法获取,下图是使用lean固件得到了完整的ipv6图片

[image]

[image]

是刷好固件之后,直接拨号,什么都没调整就获取到了

jjm2473, 2025-05-28:
参考 #1172

closed 2025-05-28 by jjm2473