istoreos/istoreos #2238: 【24.10测试】直接升级到24.10后aria2无法启动

NEA147, 2025-04-20:
从22.03直接升级到24.10后aria2服务无法启动，具体如图

[image]

[image]

YinHeng89, 2025-04-20:
同样无法启动，错误日志：
```
Sun Apr 20 17:08:41 2025 daemon.err aria2c[10373]:
Sun Apr 20 17:08:41 2025 daemon.err aria2c[10373]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sun Apr 20 17:08:41 2025 daemon.err aria2c[10373]: Exception caught
Sun Apr 20 17:08:41 2025 daemon.info aria2c[10373]: jail: exec-ing /usr/bin/aria2c
Sun Apr 20 17:08:36 2025 daemon.info aria2c[10036]: jail: jail (10037) exited with exit: 1
```

jjm2473, 2025-04-20:
在istore卸载重装试试，会保留配置的

NEA147, 2025-04-20:
> 在istore卸载重装试试，会保留配置的

依然无法启动，

[image]

jjm2473, 2025-04-20:
根据这个讨论 https://github.com/aria2/aria2/issues/2152 ，应该是openssl的库权限问题，里面也有个临时解决方法

NEA147, 2025-04-20:
> 根据这个讨论 [aria2/aria2#2152](https://github.com/aria2/aria2/issues/2152) ，应该是openssl的库权限问题，里面也有个临时解决方法。
> 
> --- a/aria2.init
> +++ b/aria2.init
>  	procd_add_jail "$NAME.$section" log
> +	procd_add_jail_mount "/usr/lib/ossl-modules"
>  	procd_add_jail_mount "$ca_certificate" "$certificate" "$rpc_certificate" "$rpc_private_key"

解决了确实这个问题，加入之后就正常了

YinHeng89, 2025-04-21:
> 解决了确实这个问题，加入之后就正常了

请问一下这个是加到 /root/etc/init.d/aria2 里面吗？

NEA147, 2025-04-21:
> 请问一下这个是加到 /root/etc/init.d/aria2 里面吗？

NEA147, 2025-04-21:
> 请问一下这个是加到 /root/etc/init.d/aria2 里面吗？

另外貌似在主线24.10.1内核6.6.86似乎修复了这一问题，istoreos后面更新应该也能解决。

jjm2473, 2025-04-21:
@NEA147 主线修复了吗？这个问题跟内核无关，就是openssl新的库需要动态加载特定文件，jail配置又不允许读取这些文件。如果你在openwrt 24.10.1上正常运行aria2，可能只是因为没安装 procd-ujail，jail未生效而已。

NEA147, 2025-04-21:
> [@NEA147](https://github.com/NEA147) 主线修复了吗？这个问题跟内核无关，就是openssl新的库需要动态加载特定文件，jail配置又不允许读取这些文件。如果你在openwrt 24.10.1上正常运行aria2，可能只是因为没安装 procd-ujail，jail未生效而已。

原来是这样啊，我看到主线提到更新openssl，有设备在24.10.1正常运行aria2，误解了以为是内核的问题，原来如此。感谢解答。

[image]

jjm2473, 2025-04-21:
补丁PR给openwrt了 https://github.com/openwrt/packages/pull/26361

jjm2473, 2025-04-22:
在openwrt更新之前，可以安装后使用这段脚本修复：
```bash
rm -f /tmp/aria2.init
wget -O /tmp/aria2.init 'https://github.com/jjm2473/packages/raw/refs/heads/istoreos-24.10/net/aria2/files/aria2.init'

[ -s /tmp/aria2.init ] && cat /tmp/aria2.init > /etc/init.d/aria2

```

NEA147, 2025-04-22:
> 在openwrt更新之前，可以安装后使用这段脚本修复：
> ```bash
> rm -f /tmp/aria2.init
> wget -O /tmp/aria2.init 'https://github.com/jjm2473/packages/raw/refs/heads/istoreos-24.10/net/aria2/files/aria2.init'
> 
> [ -s /tmp/aria2.init ] && cat /tmp/aria2.init > /etc/init.d/aria2
> 
> ```
太厉害，感谢提供修复脚本🙏

jjm2473, 2025-04-30:
openwrt 那边已经合并到24.10分支，大概一两天编译编译以后就完全没问题了

jjm2473, 2025-05-01:
openwrt已经编译，之后安装的不会有问题。

closed 2025-05-01 by jjm2473