istoreos/istoreos #2239: [24.10]首页显示网络错误

poorest1212, 2025-04-20:
[image]

chongqidewawa, 2025-04-20:
+1，设备是x86，istoreos-22.03.7-2025040711直接升级过来也是这样

jjm2473, 2025-04-20:
修复系统组件试试 https://github.com/istoreos/istoreos/wiki/FAQ

lichao0223, 2025-04-22:
> 修复系统组件试试 https://github.com/istoreos/istoreos/wiki/FAQ

X86也是这样，之前是22.03最新版本，修复了系统组件之后保留配置升级就是这样，通过Pve快照回滚了，升级了两次都是一样的

jjm2473, 2025-04-22:
电脑浏览器打开后台，然后打开浏览器的开发者工具，看看console那边有什么报错没

jjm2473, 2025-04-22:
或者谁方便远程桌面，或者远程打开后台页面？

simer11-jing, 2025-04-25:
我升级两回了，等待稳定版本吧，感觉升级后，基本软件都需要重新安装

poorest1212, 2025-04-26:
r2s刷新到0422还是老问题。而且store也打不开咧。是不是需要恢复出厂？

simer11-jing, 2025-04-26:
好像是需要重置后，插件重新部署

发自我的iPhone

bs112138, 2025-04-27:
出现同样问题，命令修复了没用

jjm2473, 2025-04-27:
已经确认此问题是由于安装过openclash导致的，并且卸载openclash不能修复此问题。openclash插件安装/卸载时（不知道是否所有版本都这样），它会修改lua的核心文件`http.lua`，修改过的文件在固件升级时将会保留下来，导致升级到24.10以后，还是使用老版本的`http.lua`，产生错误。

之后的固件在升级时会自动修复这个文件。

目前临时可以执行以下命令修复：
```bash
rm -f /overlay/upper/usr/lib/lua/luci/http.lua /ext_overlay/upper/usr/lib/lua/luci/http.lua
reboot
```
等待重启后就正常了。

simer11-jing, 2025-04-27:
那升级之前我是否可以先删除这个文件，或者卸载openclash，然后重新安装

发自我的iPhone

jjm2473, 2025-04-29:
已经说过了，不管升级前后，卸载openclash都不能修复此问题。我提供的方案不就是删除文件吗？但是要删除这两个路径的`rm -f /overlay/upper/usr/lib/lua/luci/http.lua /ext_overlay/upper/usr/lib/lua/luci/http.lua`

jjm2473, 2025-04-30:
20250430 版本已经增加自动修复机制，只要升级到这个版本即可解决此问题。

closed 2025-04-30 by jjm2473