istoreos/istoreos #2241: 【24.10测试】通过备份升级后进不了管理页面

lito12345, 2025-04-20:
我的istoreos从22.03.7 2025040711通过备份与升级升到24.10后不能进入管理界面，出现Bad Gateway，The process did not produce any response这个提示。
设备为j4125，esxi6.7虚拟机，ikuai主路由，istoreos是旁路由。

jjm2473, 2025-04-20:
打开7681端口的终端，登录，打开另一个窗口，进管理页面直到出现错误，然后回到终端窗口执行`logread | tail`看看

lito12345, 2025-04-21:
> 7681端口的终端，登录，打开另一个窗口，进入管理页面直到出现错误，然后返回终端窗口执行`logread | tail`查看

root@iStoreOS:~# logread | tail
Mon Apr 21 20:24:01 2025 auth.info login[23997]: root login on 'pts/0'
Mon Apr 21 20:24:04 2025 daemon.notice ttyd[18505]: [2025/04/21 20:24:04:7822] N: __lws_lc_untag:  -- [wsisrv|5|adopted] (1) 10.003s
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]: Runtime error: Unable to dlopen file '/usr/lib/ucode/ubus.so': Error loading shared library libubus.so.20220601: No such file or directory (needed by /usr/lib/ucode/ubus.so)
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]: In module(), file /usr/share/ucode/luci/dispatcher.uc, line 1, byte 1:
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:   called from anonymous function (/www/cgi-bin/luci:7:21)
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:  `// Copyright 2022 Jo-Philipp Wich <jo@mein.io>`
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:   ^-- Near here
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:
Mon Apr 21 20:24:07 2025 daemon.err uhttpd[9855]:

jjm2473, 2025-04-22:
参考 https://github.com/istoreos/istoreos/wiki/FAQ 第2点

closed 2025-04-22 by jjm2473