istoreos/istoreos #2244: 【24.10测试】

jeefv, 2025-04-20:
X86/64机型，1、更换软件源后很多有效插件加载不出来（使用南京大学源和immortalwrt以前的源，版本正确无误）。2、手动上传PASSwall安装提示缺少很多依赖，但软件包搜不出来（更换软件源也是）。3、Wan-6有时会无效，目前没有找到规律。

jjm2473, 2025-04-20:
软件包试试这里的24.10 https://github.com/bcseputetto/Are-u-ok/releases

jeefv, 2025-04-20:
感谢，我试试 

书中落叶
***@***.***

kennylee60, 2025-04-21:
> 软件包试试这里的24.10 https://github.com/bcseputetto/Are-u-ok/releases

我这里树莓派4试了这个包也是不行，还是缺少

kennylee60, 2025-04-21:
> 软件包试试这里的24.10 https://github.com/bcseputetto/Are-u-ok/releases

这个是报错日志，麻烦看看是什么问题

-rw-r--r--    1 root     root      59639471 Apr 21 10:01 /tmp/is-root/tmp/PassWall_25.4.10_aarch64_generic_all_sdk_24.10.0.run
1b56b7e57fe83d4b22465b0c6110bef0  /tmp/is-root/tmp/PassWall_25.4.10_aarch64_generic_all_sdk_24.10.0.run
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing PassWall_25.4.10_with_sdk_24.10.0  100%  
Downloading https://istore.istoreos.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://istore.istoreos.com/repo/all/compat/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/targets/bcm27xx/bcm2711/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/targets/bcm27xx/bcm2711/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/base/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/targets/bcm27xx/bcm2711/kmods/6.6.73-1-2577896cea679d46fe670142cc9703c1/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/targets/bcm27xx/bcm2711/kmods/6.6.73-1-2577896cea679d46fe670142cc9703c1/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/24.10.0/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Unknown package 'chinadns-ng'.
Unknown package 'dns2socks'.
Unknown package 'geoview'.
Unknown package 'hysteria'.
Unknown package 'ipt2socks'.
Unknown package 'microsocks'.
Unknown package 'naiveproxy'.
Unknown package 'shadowsocks-libev-ss-local'.
Unknown package 'shadowsocks-libev-ss-redir'.
Unknown package 'shadowsocks-libev-ss-server'.
Unknown package 'shadowsocks-rust-sslocal'.
Unknown package 'shadowsocks-rust-ssserver'.
Unknown package 'shadowsocksr-libev-ssr-local'.
Unknown package 'shadowsocksr-libev-ssr-redir'.
Unknown package 'shadowsocksr-libev-ssr-server'.
Unknown package 'simple-obfs'.
Unknown package 'sing-box'.
Unknown package 'tcping'.
Unknown package 'trojan-plus'.
Unknown package 'tuic-client'.
Package v2ray-geoip (202504050136.1) installed in root is up to date.
Package v2ray-geosite (20250407044718.1) installed in root is up to date.
Unknown package 'v2ray-plugin'.
Unknown package 'xray-core'.
Unknown package 'xray-plugin'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for chinadns-ng found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package chinadns-ng.
 * pkg_hash_fetch_best_installation_candidate: Packages for dns2socks found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package dns2socks.
[... 34 lines cut ...]
 * pkg_hash_fetch_best_installation_candidate: Packages for tuic-client found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package tuic-client.
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-plugin found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-plugin.
 * pkg_hash_fetch_best_installation_candidate: Packages for xray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package xray-core.
 * pkg_hash_fetch_best_installation_candidate: Packages for xray-plugin found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package xray-plugin.
Package haproxy (3.0.9-r1) installed in root is up to date.
正在安装passwall 25.4.10-r1
Unknown package 'luci-app-passwall'.
Installing luci-i18n-passwall-zh-cn (25.100.41496~3281fc6) to root...
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency chinadns-ng for luci-app-passwall
 * pkg_hash_check_unresolved: cannot find dependency dns2socks for luci-app-passwall
[... cut ...] l
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-passwall found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-passwall.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
 *      chinadns-ng
 *      dns2socks
 *      tcping
 * opkg_install_cmd: Cannot install package luci-i18n-passwall-zh-cn.
uci: Entry not found
./install.sh: line 26: /etc/init.d/passwall: not found

bcseputetto, 2025-04-21:
那边的issue里我也回复了，稍晚我会提供 a72的包

bcseputetto, 2025-04-21:
@kennylee60 已提供 aarch64_a72，测试一下

kennylee60, 2025-04-21:
> [@kennylee60](https://github.com/kennylee60) 已提供 aarch64_a72，测试一下

测试通过，谢谢了

jjm2473, 2025-04-22:
x86肯定没问题，恢复出厂设置以后看看能不能复现

closed 2025-04-29 by jjm2473