istoreos/istoreos #2245: COU卡IO

Swear1207, 2025-04-20:
我目前的硬件是康耐信的J4125，安装了iStoreOS 22.03.7 2024080210，硬件是系统盘SSD64G搭配一块2Thdd，发现往hdd下载文件或者先下载到SSD再转移到hdd都会卡io，我找AI排除了一下问题，发现是"系统（iStoreOS）内核太精简，可能去掉了 IO 调度器支持",https://chatgpt.com/share/6803c073-f9ac-8012-9f28-739bdb983b93，请问这个有办法解决不，期盼回复，感谢~~

jjm2473, 2025-04-20:
可能是硬盘挂了。看看内核日志有没有错误

Swear1207, 2025-04-20:
> 可能是硬盘挂了。看看内核日志有没有错误

大佬，您好，我之前排查过hdd硬盘，是显示健康的，以下是硬盘状态：
root@iStoreOS:~# smartctl -a /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.221] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     WDC WD20SPZX-00UA7T0
Serial Number:    WD-WXA2A63LDUYL
LU WWN Device Id: 5 0014ee 2c0bb27cf
Firmware Version: 01.01A01
User Capacity:    2,000,398,934,016 bytes [2.00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5400 rpm
Form Factor:      2.5 inches
TRIM Command:     Available, deterministic
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   ACS-3 T13/2161-D revision 5
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu Apr  3 13:01:49 2025 CST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00) Offline data collection activity
                                        was never started.
                                        Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever 
                                        been run.
Total time to complete Offline 
data collection:                (21480) seconds.
Offline data collection
capabilities:                    (0x71) SMART execute Offline immediate.
                                        No Auto Offline data collection support.
                                        Suspend Offline collection upon new
                                        command.
                                        No Offline surface scan supported.
                                        Self-test supported.
                                        Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        General Purpose Logging supported.
Short self-test routine 
recommended polling time:        (   2) minutes.
Extended self-test routine
recommended polling time:        ( 370) minutes.
Conveyance self-test routine
recommended polling time:        (   3) minutes.
SCT capabilities:              (0x3035) SCT Status supported.
                                        SCT Feature Control supported.
                                        SCT Data Table supported.

SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x002f   200   200   051    Pre-fail  Always       -       0
  3 Spin_Up_Time            0x0027   226   225   021    Pre-fail  Always       -       1666
  4 Start_Stop_Count        0x0032   100   100   000    Old_age   Always       -       45
  5 Reallocated_Sector_Ct   0x0033   200   200   140    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x002e   200   200   000    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   088   088   000    Old_age   Always       -       9485
 10 Spin_Retry_Count        0x0032   100   253   000    Old_age   Always       -       0
 11 Calibration_Retry_Count 0x0032   100   253   000    Old_age   Always       -       0
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       44
192 Power-Off_Retract_Count 0x0032   200   200   000    Old_age   Always       -       36
193 Load_Cycle_Count        0x0032   091   091   000    Old_age   Always       -       327668
194 Temperature_Celsius     0x0022   108   094   000    Old_age   Always       -       39
196 Reallocated_Event_Count 0x0032   200   200   000    Old_age   Always       -       0
197 Current_Pending_Sector  0x0032   200   200   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0030   100   253   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x0032   200   200   000    Old_age   Always       -       121
200 Multi_Zone_Error_Rate   0x0008   100   253   000    Old_age   Offline      -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
No self-tests have been logged.  [To run self-tests, use: smartctl -t]

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.

jjm2473, 2025-04-20:
看内核日志，电源或者数据线都可能导致问题

Swear1207, 2025-04-20:
> 看内核日志，电源或者数据线都可能导致问题

我看了一下内核日志，好像也没有啥问题，让AI帮忙提取了一下关于硬盘部分的，大佬看看

[    0.182601] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.182608] Spectre V2 : Mitigation: Enhanced / Automatic IBRS
[    0.182613] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.182619] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.182625] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl and seccomp
[    0.182631] Register File Data Sampling: Vulnerable: No microcode
[    0.214414] ... version:                4
[    0.214425] ... max period:             00007fffffffffff
[    0.214576] rcu: Hierarchical SRCU implementation.
[    0.214908] x86: Booting SMP configuration:
[    0.220725] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.220832] PCI: Using configuration type 1 for base access
[    0.225847] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.225857] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    0.293454] ACPI: SSDT 0xFFFF888101028C00 0001E6 (v02 PmRef  ApIst    00003000 INTL 20160930)
[    0.295264] ACPI: Using IOAPIC for interrupt routing
[    0.295632] ACPI: Enabled 10 GPEs in block 00 to 7F
[    0.308174] acpi PNP0A08:00: _OSC failed (AE_ERROR)
[    0.308829] pci_bus 0000:00: root bus resource [io  0x0070-0x0077]
[    0.308833] pci_bus 0000:00: root bus resource [io  0x0000-0x006f window]
[    0.308837] pci_bus 0000:00: root bus resource [io  0x0078-0x0cf7 window]
[    0.308841] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    0.309333] pci 0000:00:02.0: reg 0x20: [io  0xf000-0xf03f]
[    0.310999] pci 0000:00:12.0: reg 0x18: [io  0xf090-0xf097]
[    0.311009] pci 0000:00:12.0: reg 0x1c: [io  0xf080-0xf083]
[    0.311019] pci 0000:00:12.0: reg 0x20: [io  0xf060-0xf07f]
[    0.314825] pci 0000:00:1f.1: reg 0x20: [io  0xf040-0xf05f]
[    0.319304] iommu: Default domain type: Translated 
[    0.319341] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.319535] SCSI subsystem initialized
[    0.319544] libata version 3.00 loaded.
[    0.319791] Registered efivars operations
[    0.320219] hpet0: at MMIO 0xfed00000, IRQs 2, 8, 0, 0, 0, 0, 0, 0
[    0.322406] system 00:00: [io  0x0a00-0x0a1f] has been reserved
[    0.322406] system 00:00: [io  0x0a20-0x0a2f] has been reserved
[    0.322406] system 00:00: [io  0x0a30-0x0a3f] has been reserved
[    0.322406] system 00:02: [io  0x0680-0x069f] has been reserved
[    0.322406] system 00:02: [io  0x0400-0x047f] has been reserved
[    0.322406] system 00:02: [io  0x0500-0x05fe] has been reserved
[    0.322406] system 00:02: [io  0x0600-0x061f] has been reserved
[    0.322406] system 00:02: [io  0x164e-0x164f] has been reserved
[    0.324570] pci_bus 0000:00: resource 4 [io  0x0070-0x0077]
[    0.324574] pci_bus 0000:00: resource 5 [io  0x0000-0x006f window]
[    0.324578] pci_bus 0000:00: resource 6 [io  0x0078-0x0cf7 window]
[    0.324581] pci_bus 0000:00: resource 7 [io  0x0d00-0xffff window]
[    0.325316] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.325322] software IO TLB: mapped [mem 0x00000000721d9000-0x00000000761d9000] (64MB)
[    0.331869] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.332105] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 248)
[    0.360885] AMD-Vi: AMD IOMMUv2 functionality not available on this system - This is not a bug.
[    0.370328] Loading iSCSI transport class v2.0-870.
[    0.371783] VMware PVSCSI driver - version 1.0.7.0-k
[    0.374417] ahci 0000:00:12.0: version 3.0
[    0.384810] ahci 0000:00:12.0: AHCI 0001.0301 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
[    0.385962] ahci 0000:00:12.0: flags: 64bit ncq sntf pm clo only pmp pio slum part deso sadm sds apst 
[    0.388724] scsi host0: ahci
[    0.390217] scsi host1: ahci
[    0.391454] ata1: SATA max UDMA/133 abar m2048@0xa1917000 port 0xa1917100 irq 128
[    0.392623] ata2: SATA max UDMA/133 abar m2048@0xa1917000 port 0xa1917180 irq 128
[    0.393933] VMware vmxnet3 virtual NIC driver - version 1.5.0.0-k-NAPI
[    0.396212] Fusion MPT base driver 3.04.20
[    0.397312] Copyright (c) 1999-2008 LSI Corporation
[    0.398403] Fusion MPT SPI Host driver 3.04.20
[    0.399486] Fusion MPT SAS Host driver 3.04.20
[    0.410059] xhci_hcd 0000:00:15.0: hcc params 0x200077c1 hci version 0x100 quirks 0x0000000000009810
[    0.431008] usbcore: registered new interface driver ftdi_sio
[... 297 lines cut ...]
[2679529.316441] docker0: port 5(veth5482abd) entered blocking state
[2765085.983279] docker0: port 5(veth8f10141) entered blocking state
[2765087.114338] docker0: port 5(veth8f10141) entered blocking state
[2851738.864133] docker0: port 5(veth128e50e) entered blocking state
[2851902.402243] docker0: port 5(veth128e50e) entered blocking state
[2937842.152194] docker0: port 5(veth1560757) entered blocking state
[2937843.769253] docker0: port 5(veth1560757) entered blocking state
[3024232.830211] docker0: port 5(veth7ae29e7) entered blocking state
[3024234.317079] docker0: port 5(veth7ae29e7) entered blocking state
[3110635.808743] docker0: port 5(veth33038c3) entered blocking state
[3110637.644414] docker0: port 5(veth33038c3) entered blocking state
[3197034.466518] docker0: port 5(veth7e418fe) entered blocking state
[3197036.046400] docker0: port 5(veth7e418fe) entered blocking state
[... cut ...]
[3369834.339253] docker0: port 5(veth319e8f3) entered blocking state
[3369836.282282] docker0: port 5(veth319e8f3) entered blocking state
[3456236.607586] docker0: port 5(veth6d9b8eb) entered blocking state
[3456238.822162] docker0: port 5(veth6d9b8eb) entered blocking state
[3490510.630731] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[3490510.636990] ata1.00: configured for UDMA/133
[3490511.130763] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[3490511.132336] ata2.00: configured for UDMA/133

Swear1207, 2025-04-20:
我的2块硬盘，sda是固态，sdb是机械盘

jjm2473, 2025-04-22:
日志看起来没问题，如果只是创建文件的时候卡（包括下载或者拷贝到hdd），那可能只是因为你的文件系统不支持立即分配空间，需要把空间都写0，例如创建一个40G的文件就要先写40G的空数据。执行`mount | grep sdb`看看是什么文件系统。

Swear1207, 2025-04-22:
> 日志看起来没问题。
> 
> ~如果只是创建文件的时候卡（包括下载或者拷贝到hdd），那可能只是因为你的文件系统不支持立即分配空间，需要把空间都写0，例如创建一个40G的文件就要先写40G的空数据。执行`mount | grep sdb`看看是什么文件系统。~
> 
> 哦，我看日志sdb1是ext4文件系统，应该支持立即分配。下载或者拷贝一个大文件进去，然后再提取日志看看。

大佬，我运行mount | grep sdb显示：/dev/sdb1 on /mnt/sata2-1 type ext4 (rw,relatime)，目前正在Transmission下载大文件

[image]

jjm2473, 2025-04-22:
执行这个命令安装io调度器 `opkg update; opkg install kmod-iosched-bfq`，然后执行`echo bfq >/sys/block/sdb/queue/scheduler`，之后再测试复制文件，看有没有效果

Swear1207, 2025-04-22:
> 就是看内核日志有没有报错

我重新添加个大的下载任务试了下，内核日志没有变化

观察了一下，下载20个G的电影文件，大概前1.5G不会卡，1.5G之后io飙升

[image]

jjm2473, 2025-04-22:
安装调度器以后有效果吗

Swear1207, 2025-04-22:
> 安装调度器以后有效果吗

root@iStoreOS:~# root@iStoreOS:~# echo bfq >/sys/block/sdb/queue/scheduler
-ash: root@iStoreOS:~#: not found

第二段代码出现这个问题

jjm2473, 2025-04-22:
安装以后，这个执行没`echo bfq >/sys/block/sdb/queue/scheduler`

jjm2473, 2025-04-22:
> 大佬 我发现内核日志里有报错，但是不知道是什么方面的

没看到有什么报错，大部分是docker在启动容器。

可能就是硬盘本身有问题，要么你再试试其他系统吧，x86很容易用ventoy测试其他系统。

Swear1207, 2025-04-22:
> > 大佬 我发现内核日志里有报错，但是不知道是什么方面的
> 
> 没看到有什么报错，大部分是docker在启动容器。
> 
> 可能就是硬盘本身有问题，要么你再试试其他系统吧，x86很容易用ventoy测试其他系统。

好的 感谢大佬我再试试吧

jjm2473, 2025-04-22:
io调度器应该是很多程序同时读写硬盘才用得上，如果只是单独复制文件都能卡，跟这就没啥关系了。你的系统在没下载文件的时候，io占用多吗？`iostat 2`命令可以检查io负载