istoreos/istoreos #2249: 【24.10测试】nat66速度慢

HongJie-Master, 2025-04-21:
设备r68s
nat66下ipv6速度奇慢无比，ipv4测速正常，搜了下github.com/openwrt issues/15847
每次重启后需要执行以下命令可以解决：
ifconfig | grep "^[^ ]" | grep -v lo | awk '{print $1}' | while read iface; do ethtool -K $iface gro off; done
解决方法来自https://github.com/openwrt/openwrt/issues/15847  ptpt52

HongJie-Master, 2025-04-21:
希望下次更新能够整合一下GRO/GRO_FRAGLIST is really the issue for NAT66

jjm2473, 2025-04-22:
关闭gro会降低性能，增加cpu使用率。你把这段命令写到系统-启动项-本地启动脚本里就行了

jjm2473, 2025-04-22:
看这个提交已经解决此问题，只是还没合并到24.10 https://github.com/openwrt/openwrt/commit/00e4b23e27bd4ed3e0370b9b474ec53f72bd1282

HongJie-Master, 2025-04-22:
目前就是这么干的，期待下次更新

jjm2473, 2025-05-21:
24.10.1 20250521 应该修复了，测试下

HongJie-Master, 2025-05-21:
看到istoreos-24.10分支，你们19号有合并补丁，自己拉了一份今天刚编译好，准备下班回去测一下的，收到你的通知，我就不用我打包的测试了，我回去就测你们的，谢谢！

HongJie-Master, 2025-05-21:
测试ok,nat66下ipv6网速正常了[image]

jjm2473, 2025-05-21:
好，我看还有个补丁修复udp的也合并了。

closed 2025-05-21 by jjm2473