istoreos/istoreos #2250: 【24.10测试】R2S无网络

wuhui-9527, 2025-04-21:
试过重新和升级写入24.10固件 WAN口无网络且显示灯不亮  LAN口正常亮灯 网络连接显示一直检测中

jjm2473, 2025-04-22:
打开终端，执行下这些命令，把输出都发出来：
```bash
cat /etc/config/network
cat /etc/board.json
ls -l /sys/class/net/ | grep eth
lspci -nnk
ip link
```

wuhui-9527, 2025-04-22:
iStoreOS 24.10.0, 2025041811
 -------------------------------------------------------
root@iStoreOS:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd74:8f25:5099::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config device
        option name 'eth1'
        option macaddr 'ce:3d:91:ab:c7:24'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        list ip6class 'wan6'

config device
        option name 'eth0'
        option macaddr 'ce:3d:91:ab:c7:23'

config interface 'wan'
        option device 'eth0'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'

root@iStoreOS:~# cat /etc/board.json
{
        "model": {
                "id": "friendlyelec,nanopi-r3s",
                "name": "FriendlyElec NanoPi R3S"
        },
        "led": {
                "wan": {
                        "name": "WAN",
                        "sysfs": "wan_led",
                        "type": "netdev",
                        "device": "eth0",
                        "mode": "link tx rx"
                },
                "lan": {
                        "name": "LAN",
                        "sysfs": "lan_led",
                        "type": "netdev",
                        "device": "eth1",
                        "mode": "link tx rx"
                }
        },
        "network": {
                "lan": {
                        "device": "eth1",
                        "protocol": "static",
                        "macaddr": "ce:3d:91:ab:c7:24"
                },
                "wan": {
                        "device": "eth0",
                        "protocol": "dhcp",
                        "macaddr": "ce:3d:91:ab:c7:23"
                }
        },
        "system": {
                "hostname": "iStoreOS",
                "ntpserver": [
                        "0.cn.pool.ntp.org",
                        "1.cn.pool.ntp.org",
                        "2.cn.pool.ntp.org",
                        "3.cn.pool.ntp.org"
                ]
        }
}
root@iStoreOS:~# ls -l /sys/class/net/ | grep eth
lrwxrwxrwx    1 root     root             0 May  1 15:50 eth0 -> ../../devices/platform/fe010000.ethernet/net/eth0
lrwxrwxrwx    1 root     root             0 May  1 15:50 eth1 -> ../../devices/platform/3c0000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/eth1
root@iStoreOS:~# lspci -nnk
00:00.0 PCI bridge [0604]: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor [1d87:3566] (rev 01)
        Kernel driver in use: pcieport
lspci: Unable to load libkmod resources: error -2
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8111/8168 PCI Express Gigabit Ethernet controller [10ec:8168]
        Kernel driver in use: r8168
root@iStoreOS:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether ce:3d:91:ab:c7:23 brd ff:ff:ff:ff:ff:ff permaddr 62:07:32:40:66:3f
3: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd :: permaddr 5a2d:36a9:9d60::
4: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
5: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
6: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether d6:08:77:04:ea:d2 brd ff:ff:ff:ff:ff:ff
7: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether ce:3d:91:ab:c7:24 brd ff:ff:ff:ff:ff:ff permaddr 1e:35:ed:ad:95:52
8: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
9: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
10: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre6 :: brd :: permaddr 2a2a:f572:dd3a::
12: teql0: <NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 100
    link/void 
13: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether ce:3d:91:ab:c7:24 brd ff:ff:ff:ff:ff:ff
14: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:bc:b0:6b:d5 brd ff:ff:ff:ff:ff:ff
root@iStoreOS:~#

wuhui-9527, 2025-04-22:
@jjm2473 

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.73 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-bf7bd4c) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Thu Apr 17 12:06:28 2025
[    0.000000] Machine model: FriendlyElec NanoPi R3S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x000000007ca00000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s77992 r8192 d32600 u118784
[    0.000000] pcpu-alloc: s77992 r8192 d32600 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 515592
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007c600000-0x000000007c800000] (2MB)
[    0.000000] Memory: 2010060K/2095104K available (10944K kernel code, 3086K rwdata, 7672K rodata, 3072K init, 768K bss, 68660K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001885] Console: colour dummy device 80x25
[    0.002379] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.003376] pid_max: default: 32768 minimum: 301
[    0.008708] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.009439] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.015087] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.016272] rcu: Hierarchical SRCU implementation.
[    0.016739] rcu: 	Max phase no-delay instances is 1000.
[    0.018814] smp: Bringing up secondary CPUs ...
[... 597 lines cut ...]
[   46.700730] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   46.700747] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   46.700754] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   46.700759] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  164.648091] r8168: eth1: Link is Down
[  164.661090] br-lan: port 1(eth1) entered disabled state
[... cut ...]
[  262.254836] br-lan: port 2(dheth0) entered disabled state
[  262.254870] dheth0: entered allmulticast mode
[  262.255067] dheth0: entered promiscuous mode
[  262.306315] br-lan: port 2(dheth0) entered blocking state
[  262.306337] br-lan: port 2(dheth0) entered forwarding state
[  274.888083] r8168: eth1: Link is Down
[  274.900713] br-lan: port 1(eth1) entered disabled state
[  290.501785] r8168: eth1: Link is Up - 1Gbps/Full
[  290.502048] br-lan: port 1(eth1) entered blocking state
[  290.502091] br-lan: port 1(eth1) entered forwarding state

wuhui-9527, 2025-04-22:
[image]

jjm2473, 2025-04-22:
强制升级就行了，因为官方的dts改了设备id。

jjm2473, 2025-04-22:
> [    0.522731] mdio_bus stmmac-0: MDIO device at address 1 is missing.

内核日志说明WAN口的PHY芯片没识别到。

分别测试这两个固件，看看WAN口是否正常：
* https://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/openwrt-24.10.0-rockchip-armv8-friendlyarm_nanopi-r3s-squashfs-sysupgrade.img.gz
* https://downloads.openwrt.org/releases/24.10.1/targets/rockchip/armv8/openwrt-24.10.1-rockchip-armv8-friendlyarm_nanopi-r3s-squashfs-sysupgrade.img.gz

wuhui-9527, 2025-04-22:
@jjm2473 

24.10.0 正常无异
24.10.1 WAN. lan口指示灯正常 有网络连接  但是网站加载很慢很慢 像是网络不稳定

jjm2473, 2025-04-24:
@wuhui-9527 要全新刷机测试哦。你测试openwrt的24.10.0没问题吗

jjm2473, 2025-04-24:
之前测试从 iStoreOS 22.03 后台升级到 iStoreOS 24.10.0 也出现WAN口无法用吗

wuhui-9527, 2025-04-24:
@jjm2473 24.10没问题 后台升级和全新刷机WAN口都反应  显示已断开 指示灯 也没亮

jjm2473, 2025-04-24:
openwrt的24.10.0正常启动的内核日志也发出来看看吧

wuhui-9527, 2025-04-24:
@jjm2473 刚测试了24.10也没有网络  wan. lan口指示灯有亮

closed 2025-04-25 by wuhui-9527

jjm2473, 2025-04-30:
已经修复了

x2358473, 2025-05-01:
我使用改了4g内存的r3s无法启动24.10的固件。

x2358473, 2025-05-01:
> 我使用改了4g内存的r3s无法启动24.10的固件。

安装22的istoreos可以启动