istoreos/istoreos #2251: 【24.10测试】R2S 保留配置升级 LUCI页面异常

cellistV, 2025-04-21:
设备R2S 保留配置升级。
除了首页错误外，服务页面的软件选项会提示：
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/http.lua:145: attempt to index field 'request' (a nil value)

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:803:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:777:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:804:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1017:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
以及系统便利工具、istroe点开是空白页。

lichao0223, 2025-04-22:
X86有一样的问题

ralopa673, 2025-04-22:
参照 #2231 处理

cellistV, 2025-04-22:
> 参照 [#2231](https://github.com/istoreos/istoreos/issues/2231) 处理

有效，升级测试版建议关掉代理后再操作，不然恢复luci界面会遇到软件源问题。

closed 2025-04-22 by cellistV