istoreos/istoreos #2263: 【24.10测试】

Amethyonst, 2025-04-24:
自编译x86 https://github.com/istoreos/istoreos/commit/d725c2ee933d1b254987977ddc2275a716208da4

**wifi重启后显示设备未激活**

一张AX200 运行```/sbin/wifi up ```可恢复
另一张usb网卡（之前的推荐型号，MT7921 COMFAST）识别为```Generic unknown```,插拔+/sbin/wifi up，识别为```Aukey USBAC1200 802.11ac/b/g/n```

jjm2473, 2025-04-29:
intel的无线网卡全部不支持5G AP模式，后台的无线配置将AP模式改成客户端一般就行了

jjm2473, 2025-04-29:
usb的无线网卡插上去以后，执行 `wifi config`，看看后台的无线配置有没有出现新的网卡。

Amethyonst, 2025-04-29:
**该问题似乎与新的命名有关**

[image]
[image]

AX200 始终工作在Client模式，系统启动后显示为radio0，执行```/sbin/wifi up``` 则会变成phy-sta0

[image]

usb网卡工作为ap模式，插拔后运行命令，恢复为phy1-ap0

[image]

Amethyonst, 2025-04-30:
是的