istoreos/istoreos #2264: 【24.10测试】简单网络管控v1.2功能失效

lzjh136789, 2025-04-25:
开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。

jjm2473, 2025-05-06:
翻墙插件应该要配置这些客户端不代理

lzjh136789, 2025-05-06:
这个我知道，但是旧版是不用设置翻墙软件就可以控制ip上不上了网的，现在没法了。还有访问限制，设置好了开启了也不工作的。希望出一个对应24.10的网络管理控制，

jjm2473, 2025-05-07:
找作者更新个支持fw4版本的

lzjh136789, 2025-05-07:
istore商店下载了不知道作者是谁

gaobin89, 2025-05-27:
> 开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。

在Lienol原版的基础上，自己改了一个fw4版本。https://github.com/gaobin89/luci-app-timecontrol

[luci-app-timecontrol.zip](https://github.com/user-attachments/files/20448266/luci-app-timecontrol.zip)

[image]

[image]

gaobin89, 2025-05-27:
> > > 开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。
> > 
> > 在Lienol原版的基础上，自己修改了一个fw4版本。https://github.com/gaobin89/luci-app-timecontrol
> > 
> > [luci-app-时间控制.zip](https://github.com/user-attachments/files/20448266/luci-app-timecontrol.zip)
> > 
> > [image]
> > 
> > [image]
> 
> 
> 
> > > 开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。
> > 
> > 在Lienol原版的基础上，自己修改了一个fw4版本。https://github.com/gaobin89/luci-app-timecontrol
> > 
> > [luci-app-时间控制.zip](https://github.com/user-attachments/files/20448266/luci-app-timecontrol.zip)
> > 
> > [image]
> > 
> > [image]
> 
> 谢谢，不过我不知道如何安装你人软件？？可以执教一下吗

在https://github.com/gaobin89/luci-app-timecontrol 的Release页面下载最新的两个ipk手动在系统-软件包-上传软件包，安装就行了。

lzjh136789, 2025-05-28:
[image]
[image]

> > > > > 开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。
> > > > 
> > > > 
> > > > 在Lienol原版的基础上，自己修改了一个fw4版本。[https://github.com/gaobin89/luci-app-timecontrol ](https://github.com/gaobin89/luci-app-timecontrol)
> > > > [luci-app-时间控制.zip](https://github.com/user-attachments/files/20448266/luci-app-timecontrol.zip)
> > > > [image]
> > > > [image]
> > > 
> > > 
> > > > > 开启上网时间控制后，还可以看油管，外网。国内的网站就可以控制上不了。访问限制运行不了设置好后还是显示未运行。。
> > > > 
> > > > 
> > > > 在Lienol原版的基础上，自己修改了一个fw4版本。[https://github.com/gaobin89/luci-app-timecontrol ](https://github.com/gaobin89/luci-app-timecontrol)
> > > > [luci-app-时间控制.zip](https://github.com/user-attachments/files/20448266/luci-app-timecontrol.zip)
> > > > [image]
> > > > [image]
> > > 
> > > 
> > > 谢谢，不过我不知道如何安装你的软件？？可以执教一下吗
> > 
> > 
> > 在https://github.com/gaobin89/luci-app-timecontrol的Release页面下载最新的两个ipk手动在系统-均衡-上传分区，安装就行了。
> > 谢谢，已经安装好了，但是显示未运行，不知道是那个错误了