istoreos/istoreos #2265: 【24.10测试】Tailscale组网异常

NJYZ-96, 2025-04-25:
使用固件如图：

[image]

* PVE平台全新安装，未导入旧配置，也没有安装乱七八糟插件

[image]

* 安装Tailscale并配置广播子网路由和出口模式后，连接是正常的，但无法使用，外网无法访问局域网设备

* 同样的操作，在istoreos-22.03.7上是完全正常的

* 同时，在istoreos-24.10.0安装时，多了两条报错，如下，这个提示在istoreos-22.03.7上安装是没有的
>Warning: UDP GRO forwarding is suboptimally configured on eth0, UDP forwarding throughput capability will increase with a configuration change.
See https://tailscale.com/s/ethtool-config-udp-gro

* 有尝试多种方式安装，（通过一键脚本或者iStore商店）问题依旧

```
-------------------------

To authenticate, visit:

        https://login.tailscale.com/a/1a63aef4014770

Success.
INSTALL: -------------------------------------------------------------------------
你可以运行 tailscale up --advertise-routes=192.168.1.0/24 以启用子网中继，请自行修改网段
----------------------------------------------------------------------------------
root@Z7-iStoreOS-alpha:~# tailscale up --advertise-routes=192.168.50.0/24
Warning: UDP GRO forwarding is suboptimally configured on eth0, UDP forwarding throughput capability will increase with a configuration change.
See https://tailscale.com/s/ethtool-config-udp-gro
root@Z7-iStoreOS-alpha:~# tailscale up --advertise-exit-node --advertise-routes=192.168.50.
0/24
Warning: UDP GRO forwarding is suboptimally configured on eth0, UDP forwarding throughput capability will increase with a configuration change.
See https://tailscale.com/s/ethtool-config-udp-gro
```

lichao0223, 2025-04-28:
把tailscale的网卡加到防火墙设置里去允许转发

NJYZ-96, 2025-04-28:
> 把tailscale的网卡加到防火墙设置里去允许转发

可以详细说说嘛？

jjm2473, 2025-04-29:
这可能是因为openwrt 24.10开始防火墙默认禁止入站。

防火墙需要配置这个接口，允许入站。或者打开全局的允许入站。

jjm2473, 2025-04-29:
注意，最好不要参考前面的文档建立tailscale接口，直接去防火墙建立tailscale区域，然后区域的高级设置里面就可以将“tailscale+”添加到“涵盖的设备”中，“tailscale+”的意思是所有“tailscale“开头的接口都生效，如果只要“tailscale0”，那就直接填“tailscale0”。常规设置中允许入站和转发，视情况可能要勾选“IP 动态伪装”（也就是NAT）。

NJYZ-96, 2025-04-30:
> 注意，最好不要参考前面的文档建立tailscale接口，直接去防火墙建立tailscale区域，然后区域的高级设置里面就可以将“tailscale+”添加到“涵盖的设备”中，“tailscale+”的意思是所有“tailscale“开头的接口都生效，如果只要“tailscale0”，那就直接填“tailscale0”。常规设置中允许入站和转发，视情况可能要勾选“IP 动态伪装”（也就是NAT）。

实测有效，感谢回复。

* openwrt 24.10+ 开始防火墙默认禁止入站，需要配置防火墙来开放`Tailscale`接口；否则能连接上但无法实际传输数据
>`openwrt-网络-防火墙-常规设置-区域-添加`
>
>常规设置：
>>名称：Tailscale
>>
>>入站数据：接受
>>
>>区域内转发：接受
>>
>>IP 动态伪装（NAT）：视情况开启
>
>高级设置：
>>涵盖的设备：自定义添加`“tailscale+”`
>>>`“tailscale+”`的意思是所有`“tailscale”`开头的接口都生效
>>>
>>>如果只要`“tailscale0”`，那就直接填`“tailscale0”`
>
>保存并应用

closed 2025-04-30 by NJYZ-96

londbell, 2025-05-04:
请问你的tailscale是什么版本？

londbell, 2025-05-05:
> > 请问你的tailscale是什么版本？
> 
> 目前我升级到最新版了

我也升级到1.82.5了，但是只要一注册headscale成功，网络就全挂了，路由器也无法访问

jhdh487, 2025-05-25:
> > > 注意，最好不要参考前面的文档建立tailscale接口，直接去防火墙建立tailscale区域，然后区域的高级设置里面就可以将“tailscale+”添加到“涵盖的设备”中，“tailscale+”的意思是所有“tailscale“开头的接口都生效，如果只要“tailscale0”，那就直接填“tailscale0”。常规设置中允许入站和转发，视情况可能要勾选“IP 动态伪装”（也就是NAT）。
> > 
> > 实测有效，感谢回复。
> > 
> > * openwrt 24.10+ 开始防火墙默认禁止入站，需要配置防火墙来开放`Tailscale`接口；否则能连接上但无法实际传输数据
> > >`openwrt-网络-防火墙-常规设置-区域-添加`
> > >
> > >常规设置：
> > >>名称：Tailscale
> > >>
> > >>入站数据：接受
> > >>
> > >>区域内转发：接受
> > >>
> > >>IP 动态伪装（NAT）：视情况开启
> > >
> > >高级设置：
> > >>涵盖的设备：自定义添加`“tailscale+”`
> > >>>`“tailscale+”`的意思是所有`“tailscale”`开头的接口都生效
> > >>>
> > >>>如果只要`“tailscale0”`，那就直接填`“tailscale0”`
> > >
> > >保存并应用
> > 
> 
> 添加完这个之后，其他设备连接tailscale可以访问tailscale分配的100.的ip进入路由器后台，但是无法访问192.168.2.1。
> tailscale已经进行了advertise-routes=192.168.2.0/24

路由器版本iStoreOS 24.10.1 2025052311

jhdh487, 2025-05-25:
路由器版本iStoreOS 24.10.1 2025052311

jhdh487, 2025-05-25:
> > > > 注意，最好不要参考前面的文档建立tailscale接口，直接去防火墙建立tailscale区域，然后区域的高级设置里面就可以将“tailscale+”添加到“涵盖的设备”中，“tailscale+”的意思是所有“tailscale“开头的接口都生效，如果只要“tailscale0”，那就直接填“tailscale0”。常规设置中允许入站和转发，视情况可能要勾选“IP 动态伪装”（也就是NAT）。
> > > 
> > > 
> > > 实测有效，感谢回复。
> > > 
> > > * openwrt 24.10+ 开始防火墙默认禁止入站，需要配置防火墙来开放`Tailscale`接口；否则能连接上但无法实际传输数据
> > > 
> > > > `openwrt-网络-防火墙-常规设置-区域-添加`
> > > > 常规设置：
> > > > > 名称：Tailscale
> > > > > 入站数据：接受
> > > > > 区域内转发：接受
> > > > > IP 动态伪装（NAT）：视情况开启
> > > > 
> > > > 
> > > > 高级设置：
> > > > > 涵盖的设备：自定义添加`“tailscale+”`
> > > > > > `“tailscale+”`的意思是所有`“tailscale”`开头的接口都生效
> > > > > > 如果只要`“tailscale0”`，那就直接填`“tailscale0”`
> > > > 
> > > > 
> > > > 保存并应用
> > 
> > 添加完这个之后，其他设备连接tailscale可以访问tailscale分配的100.的ip进入路由器后台，但是无法访问192.168.2.1。 tailscale已经进行了advertise-routes=192.168.2.0/24
> 
> 需要到tailscale网页面板批准子网路由

已经批准，还是不行

NJYZ-96, 2025-05-25:
那就不清楚了，我这边是正常使用的。

怀疑是你的某些插件修改了DNS相关的配置；全新安装一个istoreos试试？

gaoshulaoshi, 2025-05-28:
> 已经批准，还是不行

24.10版本的istore安装tailscale后怎么弄都无法路由内网，装回旧版本就没这个情况

DemoJ, 2025-06-03:
[image]
亲测有效