istoreos/istoreos #2269: 安装插件市场的pve虚拟机后屏幕控制台输出异常

865258951, 2025-04-26:
pve会和openwer互相抢占控制台输出
[image]
下边是采集卡的画面
[image]

jjm2473, 2025-05-06:
`cat /proc/cmdline`看看是什么

865258951, 2025-05-06:
[root@Kwrt:03:27 PM ~] # cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=e6506b27-bdbb-4199-fde8-8efa58f7c202 rootwait console=ttyS0,115200n8 noinitrd

root@pve:~# cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=e6506b27-bdbb-4199-fde8-8efa58f7c202 rootwait console=ttyS0,115200n8 noinitr

jjm2473, 2025-05-07:
不要使用 `console=ttyS0,115200n8`，应该就没问题了。

closed 2025-05-07 by jjm2473

865258951, 2025-09-28:
最新发现的完美解决方案
在pve下执行
systemctl disable getty@tty1.service
systemctl enable getty@tty2.service
即可解决这个问题，同时可以在显示器上使用alt+F2切换到pve的控制台

jjm2473, 2025-09-28:
alt+F1可以切换回openwrt的控制台吗

865258951, 2025-09-28:
可以，alt+F1就回到了openwrt的控制台