istoreos/istoreos #2270: 无法解析IPV6

LianghaiLi-root, 2025-04-26:
网口已经正确获得IPV6公网，DHCP也正常，PC均获得了IPV6地址，nslookup发现无法解析IPV6，过滤器中（过滤 IPv6 AAAA 记录）并未打勾，在路由器内部也无法解析IPV6。

138161, 2025-04-28:
1、wan口获得的ipv6有PD前缀吗？64位？
2、ipv6是中继模式还是NAT6模式？
3、lan口和PC获得的ipv6地址开头是2409  fe80  fd00，哪一种

LianghaiLi-root, 2025-04-28:
很抱歉，今天晚上我又重新安装了系统，重新配置了，已经解决了，好像是我的DNS配置错误了，昨天安装了两次没解决才来发的条目，希望你们做得越来越好；

今天配置完成后自己安装了一些插件，在配置的过程中保存一些东西的时候，发现经常卡死web无法进入，过一会儿又可以进入web页面，在卡死的时候网络功能是正常的，使用SSH进入后台stop uhttp，在start uhttp是可以马上进入路由管理界面的，当然我的cpu应该是够用的，我安装的方式J4125安装的esxi，分配2核cpu和1G内存给路由，是卡死了多次，我只能把我遇到的情况给你们说，目前其他的并没有发现什么问题。

LianghaiLi-root, 2025-04-28:
后面我使用nginx替代luhppd，测试了5到10分钟左右，没有web页面卡死的状况。

jjm2473, 2025-04-29:
试试公共的dns，例如`nslookup taobao.com 223.5.5.5`，看看能不能取得ipv6的地址。如果没有，关掉任何翻墙软件再试试。如果还是没有，那可能你的电脑的ipv6地址不是有效的。

closed 2025-04-29 by jjm2473