istoreos/istoreos #2271: 【24.10测试】zerotier无法启动

lito12345, 2025-04-27:
一直运行良好，系统从23.07.11升级24.10后也没有问题，最近出现了一次插件更新后目前zerotier无法启动，版本号1.2.0，商店里最新版本

jjm2473, 2025-04-29:
看看zerotier的luci配置页面是不是空的，如果是空的，更新就能解决。但是更新以后，要看看配置是不是正确的。

lito12345, 2025-04-29:
> 看看zerotier的luci配置页面是不是空的，如果是空的，更新就能解决。但是更新以后，要看看配置是不是正确的。

还是不行，不是空的，看了一下没有接口信息，问题应该在这里

jjm2473, 2025-04-30:
看看日志zerotier有没有报错，如果没有，截图看看你的配置是什么样的

lito12345, 2025-04-30:
> 看看日志zerotier有没有报错，如果没有，截图看看你的配置是什么样的
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error loading shared library libminiupnpc.so.18: No such file or directory (needed by /usr/bin/zerotier-one)
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: UPNP_GetSpecificPortMappingEntry: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: UPNP_AddPortMapping: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: UPNP_GetValidIGD: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: freeUPNPDevlist: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: FreeUPNPUrls: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: UPNP_GetExternalIPAddress: symbol not found
Wed Apr 30 01:49:54 2025 daemon.err zerotier-one[20079]: Error relocating /usr/bin/zerotier-one: upnpDiscoverAll: symbol not found

fengyaochen, 2025-05-03:
zerotier很多nat规则都是用iptables的，失效以后就启动不了了，就算能启动，功能也不完整，比如组网以后想要访问外网的就不行，只能访问openwrt所在整个局域网其它设备，假如openwrt是旁路由，那只能访问走旁路由流量的设备，走主路由的设备就访问不了，都是新版防火墙的锅

lito12345, 2025-05-03:
> zerotier很多nat规则都是用iptables的，失效以后就启动不了了，就算能启动，功能也不完整，比如组网以后想要访问外网的就不行，只能访问openwrt所在整个局域网其它设备，假如openwrt是旁路由，那只能访问走旁路由流量的设备，走主路由的设备就访问不了，都是新版防火墙的锅

我看了istoreos的430版本及最新市场ui还是有问题，主要在config下这个zerotier的配置文件有问题，缺globe，另外防火墙需要自己设接口、区域、通信规则可正常。ui在2410下不可用，重启又会䨱盖错误配置，等更新或自己手动装吧

jjm2473, 2025-05-06:
> Wed Apr 30 12:46:20 2025 daemon.err zerotier-one[23802]: /usr/bin/zerotier-one: fatal error: cannot bind to local control interface port 9993

看起来只是因为9993端口被占用了

jjm2473, 2025-05-06:
恢复出厂设置，不要添加任何第三方的软件源，安装zerotier，把安装日志发出来看看，我怀疑你安装的是其他版本的luci。

fengyaochen, 2025-05-07:
> 恢复出厂设置，不要添加任何第三方的软件源，安装zerotier，把安装日志发出来看看，我怀疑你安装的是其他版本的luci。或者卸载重装，看看安装日志
我这边新装6.0或者从5.0升级上来zerotier都可以正常启动，但是在zerotier官网设定的路由表无法正常工作，zerotier官网我设定访问外网的路由表是0.0.0.0/0，在5.0下连接成功就能访问外网，6.0就不行,还有一个问题，zerotier官网设定对方内网的路由表只能访问到对方openwrt以及这个openwrt网关下面的设备，如果对方的openwrt没有作为主路由而是作为旁路由的话，我只能访问到它下面走旁路由的设备，这个局域网内其它设备的网关没有走旁路由就不能访问，这些问题在5.0内核下面都是不存在的

jjm2473, 2025-05-07:
@fengyaochen 可能需要自己配置防火墙规则

lito12345, 2025-05-07:
> [@fengyaochen](https://github.com/fengyaochen) 可能需要自己配置防火墙规则。22.03时还是使用iptables，插件可能自动配置了防火墙，现在是nftables，插件可能不支持自动配置。

谢谢，问题已解决，升级到430的istoreos24.10，卸载zero，还需要ssh进入删掉原来zerotier的所有配置后再重装否则无法进入组网，确实nft需要配置，可以粗暴在常规打开转发或配置zerotier的接口与添加防火墙区域与lan区域进行互访，否则组网后无线防问局域网设备

lito12345, 2025-05-08:
> 恢复出厂设置，不要添加任何第三方的软件源，安装zerotier，把安装日志发出来看看，我怀疑你安装的是其他版本的luci。或者卸载重装，看看安装日志 [...]

现在只剩外网不访问了，内网无问题，我是旁路由。怎么折腾防火墙，zero的控制中心，接口都不行

SyunSS, 2025-05-10:
今天刚刚装上24.10发现zerotier也挂了 我也试着重装了下 luci界面一直报错 没办法添加配置 只能强制降级到22.03 还好强制降级后重新装上zerotier配置都还读的到

kaige1116, 2025-05-13:
也是遇到一样的问题 不知道怎么处理了。等作者更新吧

spellyaohui, 2025-05-17:
> 也是遇到一样的问题 不知道怎么处理了。等作者更新吧

配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上@lito12345 的办法即可解决~我试过了,好使

spellyaohui, 2025-05-17:
> > > 也是遇到一样的问题 不知道怎么处理了。等作者更新吧
> > 
> > 
> > 配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上[@lito12345](https://github.com/lito12345) 的办法即可解决~我试过了,好使
> 
> ssh进入也卸载不干净，能否搞个步骤或者命令？

我也不会,直接重置路由器重新搞的

KimmyLueng, 2025-05-20:
> > 也是遇到一样的问题 不知道怎么处理了。等作者更新吧
> 
> 配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上[@lito12345](https://github.com/lito12345) 的办法即可解决~我试过了,好使

同样是拨号的设置后可以互访，但假如istore os设置为旁路由的就无法访问

kaige1116, 2025-05-20:
> > > 也是遇到一样的问题 不知道怎么处理了。等作者更新吧
> > 
> > 
> > 配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上[@lito12345](https://github.com/lito12345) 的办法即可解决~我试过了,好使
> 
> 同样是拨号的设置后可以互访，但假如istore os设置为旁路由的就无法访问

我的就是旁路由

KimmyLueng, 2025-05-20:
> > > > 也是遇到一样的问题 不知道怎么处理了。等作者更新吧
> > > 
> > > 
> > > 配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上[@lito12345](https://github.com/lito12345) 的办法即可解决~我试过了,好使
> > 
> > 
> > 同样是拨号的设置后可以互访，但假如istore os设置为旁路由的就无法访问
> 
> 我的就是旁路由

我开了访问AP/光猫，光猫的ip段跟旁路由的ip段一样，关了就行

[image]

kaige1116, 2025-05-21:
> > > > > 也是遇到一样的问题 不知道怎么处理了。等作者更新吧
> > > > 
> > > > 
> > > > 配置zerotier的接口与添加防火墙区域与lan区域进行互访 用楼上[@lito12345](https://github.com/lito12345) 的办法即可解决~我试过了,好使
> > > 
> > > 
> > > 同样是拨号的设置后可以互访，但假如istore os设置为旁路由的就无法访问
> > 
> > 
> > 我的就是旁路由
> 
> 我开了访问AP/光猫，光猫的ip段跟旁路由的ip段一样，关了就行
> 
> [image]

我是光猫跟主路由旁路由不同网端

fengyaochen, 2025-06-01:
> 找到暂时不需要手动配置防火墙的方法。就是安装 immortalwrt 那边编译好的 2410 zerotier 包。具体步骤如下
> 
> 1. 卸载已经安装的 zerotier，在商店安装的可以在商店卸载，或者在软件包已安装软件里面卸载所有带 zerotier 的package
> 2. 到 /etc/config 里面删除 zerotier 配置文件，这一步很关键，不然 后面安装新包可能无法正常运行。
> 3. 到 immortalwrt 的仓库找到相应的 zerotier包。主要是三个，zerotier.ipk, luci-app-zerotier.ipk, luci-i18n-zerotier-zh-cn.ipk.
> 4. 然后就可以添加 zerotier 配置了。注意这里加入网络后[，会当成一个新的设备，所以需要到 zerotier 官网的设备管理去重新，而且分到一个新的 IP。可以自己绑定旧的IP。然后就大功告成了。
> 5. iStore 商店的 zerotier  应该很快就能适配 24.10 版。翘首期盼中。
> 6. 我自己的软路由是 x86 的，几个软件包的链接放在下面，不是 x86 的可以在仓库里找找对应的版本。[zerotier](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/packages/zerotier_1.14.1-r8_x86_64.ipk)
>    [luci-app-zerotier](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/luci/luci-app-zerotier_25.116.62431~e7f6f1e_all.ipk)
>    [luci-i18n-zerotier-zh-cn](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/luci/luci-i18n-zerotier-zh-cn_25.116.62431~e7f6f1e_all.ipk)

不行，结果和istore os里装的zerotier是一样的，zerotier官网的路由表都失效，我是旁路由

118119019, 2025-06-03:
有没有大神指导下怎么处理 具体点 弄了好几天了没效果

118119019, 2025-06-03:
root@iStoreOS:~# zerotier-one -d
root@iStoreOS:~# Starting Control Plane...
Starting V6 Control Plane...
一直报这个错误

fengyaochen, 2025-06-03:
> > > 找到暂时不需要手动配置防火墙的方法。就是安装 immortalwrt 那边编译好的 2410 zerotier 包。具体步骤如下
> > > 
> > > 1. 卸载已经安装的 zerotier，在商店安装的可以在商店卸载，或者在软件包已安装软件里面卸载所有带 zerotier 的package
> > > 2. 到 /etc/config 里面删除 zerotier 配置文件，这一步很关键，不然 后面安装新包可能无法正常运行。
> > > 3. 到 immortalwrt 的仓库找到相应的 zerotier包。主要是三个，zerotier.ipk, luci-app-zerotier.ipk, luci-i18n-zerotier-zh-cn.ipk.
> > > 4. 然后就可以添加 zerotier 配置了。注意这里加入网络后[，会当成一个新的设备，所以需要到 zerotier 官网的设备管理去重新，而且分到一个新的 IP。可以自己绑定旧的IP。然后就大功告成了。
> > > 5. iStore 商店的 zerotier  应该很快就能适配 24.10 版。翘首期盼中。
> > > 6. 我自己的软路由是 x86 的，几个软件包的链接放在下面，不是 x86 的可以在仓库里找找对应的版本。[zerotier](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/packages/zerotier_1.14.1-r8_x86_64.ipk)
> > >    [luci-app-zerotier](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/luci/luci-app-zerotier_25.116.62431~e7f6f1e_all.ipk)
> > >    [luci-i18n-zerotier-zh-cn](https://downloads.immortalwrt.org/releases/packages-24.10/x86_64/luci/luci-i18n-zerotier-zh-cn_25.116.62431~e7f6f1e_all.ipk)
> > 
> > 
> > 不行，结果和istore os里装的zerotier是一样的，zerotier官网的路由表都失效，我是旁路由
> 
> zerotier 能正常启动吗？看一下日志，有没有别的错误信息。

能正常启动，就和istoreos里安装的一样没有错误就是官网设定的路由表都不起作用，ping 私有地址能通其它都不通，我也是旁路由

jjm2473, 2025-06-05:
zerotier的自动NAT支持fw4了，从istore卸载重装或者更新

jjm2473, 2025-06-05:
upnp导致无法启动的问题openwrt那边应该已经修复了，从istore卸载重装即可。如果还有问题，执行`opkg info zerotier | grep Version:`确保显示`Version: 1.14.1-r2`或者比这更新。

jjm2473, 2025-06-09:
zerotier的upnp还有问题，在这跟进 https://github.com/istoreos/istoreos/issues/2358

kaige1116, 2025-06-09:
> zerotier的upnp还有问题，在这跟进 [#2358](https://github.com/istoreos/istoreos/issues/2358)

x86最新版本istoreos跟最新版本的zerotier没见有这情况。