istoreos/istoreos #2272: 【24.10测试】检测到旧版规则 系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。

lzjh136789, 2025-04-27:
如题，这个如何处理？

Huaizh, 2025-04-28:
目前，我测出来的是如果自己安装了luci-app-adguardhome 1.8.xx版本,以及luci-app-wrtbwon任何版本，就会导致这个问题，从根本上来讲，是他俩还在用iptables进行流量监控
adguardhome可以不用luci或者有个针对nft的1.9版本，wrtbwon就没办法了，装了就会混用防火墙
不过，以上仅针对个人经验，如果你没有装这俩插件，就慢慢找吧，全新安装的24.10我特意看了，是没问题的

lzjh136789, 2025-04-28:
收到谢谢，

jjm2473, 2025-04-29:
混用一般没问题的，现在只是有些插件还在使用iptables。如果是翻墙插件，那么需要设置防火墙为nftables

lzjh136789, 2025-05-06:
问题找到了，就是开启了简单网络控制这个软件导致的，这软件的访问限制在24.10系统里面也是失效的，所以我现在不用了。就不会出现iptables的

heiyebaitian, 2025-05-13:
> 目前，我测出来的是如果自己安装了luci-app-adguardhome 1.8.xx版本,以及luci-app-wrtbwon任何版本，就会导致这个问题，从根本上来讲，是他俩还在用iptables进行流量监控 adguardhome可以不用luci或者有个针对nft的1.9版本，wrtbwon就没办法了，装了就会混用防火墙 不过，以上仅针对个人经验，如果你没有装这俩插件，就慢慢找吧，全新安装的24.10我特意看了，是没问题的

您好，您文中提及针对nft的1.9版本adguardhome，可以给个链接吗，这边好像找不到...

44571366, 2025-09-19:
> 问题找到了，就是开启了简单网络控制这个软件导致的，这软件的访问限制在24.10系统里面也是失效的，所以我现在不用了。就不会出现iptables的

兄台，这个详细是怎么解决的，这个简单网络控制在那儿呀