istoreos/istoreos #2273: 【24.10测试】从istoreos更新alist插件后无法运行

NEA147, 2025-04-27:
测试平台X86 物理机 ，从istoreos商店更新alist插件到alist v3.44.0后无法运行，再尝试卸载重装后依然无法运行。具体错误如图。

[image]

[image]

[image]

xfghvgnfyjssjgte, 2025-04-28:
我也遇到了，以前的cui界面在网络存储，新的更新到了 服务下，用alist server可以启动，但是原配置文件丢失，现在我也是只能回退到了3.43.0版本

jjm2473, 2025-04-29:
现在alist使用的是openwrt官方的版本，配置文件可能跟之前不一样了，需要重新配置。

1. 先重命名老的配置文件`mv /etc/config/alist /etc/config/alist.old`，
2. 如果之前设置过数据目录，将老的数据目录复制到`/etc/alist`，如果之前就是`/etc/alist`，那就不需要进行这一步
3. 然后重新安装alist，勾选启用就行了

NEA147, 2025-04-29:
> 现在alist使用的是openwrt官方的版本，配置文件可能跟之前不一样了，需要重新配置。
> 
>     1. 先重命名老的配置文件`mv /etc/config/alist /etc/config/alist.old`，
> 
>     2. 如果之前设置过数据目录，将老的数据目录复制到`/etc/alist`，如果之前就是`/etc/alist`，那就不需要进行这一步
> 
>     3. 然后重新安装alist，勾选启用就行了

有效，这样配置正常了

jjm2473, 2025-04-29:
执行这段脚本修复：
```bash

mkdir /etc/alist

ALIST_DATA_DIR=$(uci -q get 'alist.@alist[0].data_dir' || echo '/etc/alist')

[ '/etc/alist' = "$ALIST_DATA_DIR" ] || cp -a "$ALIST_DATA_DIR/." /etc/alist/

[ -f /etc/config/alist.old ] || cp -a /etc/config/alist /etc/config/alist.old
[ -f /etc/config/alist-opkg ] && cp -a /etc/config/alist-opkg /etc/config/alist

```

修复以后，后台打开alist的配置页面，勾选启用，端口如果不合适也可以改，然后保存应用就行了。

jjm2473, 2025-04-29:
已经修复了，如果之前有问题，从istore再更新一次即可。

closed 2025-04-29 by jjm2473