istoreos/istoreos #2283: 【24.10测试】R2S配置加入Tailscale网络后必定无法访问管理页面

londbell, 2025-05-03:
使用固件版本：[istoreos-24.10.0-2025043010-r2s-squashfs.img.gz](https://fw0.koolcenter.com/iStoreOS/alpha/24.10/r2s/istoreos-24.10.0-2025043010-r2s-squashfs.img.gz)

全新安装，每次都能复现，排除了其他插件、docker的干扰；到现在已经试了第三次

设备：R2S
模式：PPPOE
LAN地址：192.168.2.1
tailscale网段：192.168.100.0/24

使用istore商店里的tailscale插件，安装后启用，然后tailscale up

再去headscale放行：
headscale nodes register --user default --key nodekey:xxxx

然后网络就会挂掉，完全无法使用，关机后重启，此时无法连接管理页面，但是可以上网；

然后访问不上后台：

[image]

ssh无法访问：
[image]

ping不通：

[image]

此时nslookup baidu正常，但是看dns，应该是ipv6负责解析的，并不是ipv4的192.168.2.1

[image]

另外，解析istoreos官网也是失败的，无法访问

londbell, 2025-05-03:
过一段时间后，再也无法联网

londbell, 2025-05-10:
应该是tailscale的--accept-routes选项问题，新版本的tailscale使用--accept-routes后，lan口就只能通过tailscale访问

closed 2025-05-10 by londbell

londbell, 2025-05-10:
> 应该是tailscale的--accept-routes选项问题，新版本的tailscale使用--accept-routes后，lan口就只能通过tailscale访问

参考：
https://forum.gl-inet.com/t/gl-axt1800-tailscale-remote-subnet-routing-fails/29494/10
https://forum.gl-inet.com/t/tailscale-cannot-reach-subnets-on-other-devices/27063/53
https://github.com/tailscale/tailscale/issues/1164

jjm2473, 2025-06-06:
是不是你在tailsacle配置的路由跟本地路由冲突了？