istoreos/istoreos #2287: istoreos-24.10不支持bnx2x_warpcore_8727_2_5g驱动补丁文件

Jaykwok2999, 2025-05-05:
加入以下补丁编译是出错的

--- a/drivers/net/ethernet/broadcom/bnx2x/bnx2x.h
+++ b/drivers/net/ethernet/broadcom/bnx2x/bnx2x.h
@@ -1593,6 +1593,7 @@ struct bnx2x {
 	uint			num_ethernet_queues;
 	uint			num_cnic_queues;
 	int			disable_tpa;
+	int			mask_tx_fault;
 
 	u32			rx_mode;
 #define BNX2X_RX_MODE_NONE		0
--- a/drivers/net/ethernet/broadcom/bnx2x/bnx2x_link.c
+++ b/drivers/net/ethernet/broadcom/bnx2x/bnx2x_link.c
@@ -151,6 +151,7 @@ typedef int (*read_sfp_module_eeprom_fun
 
 #define SFP_EEPROM_CON_TYPE_ADDR		0x2
 	#define SFP_EEPROM_CON_TYPE_VAL_UNKNOWN	0x0
+	#define SFP_EEPROM_CON_TYPE_VAL_SC	0x1
 	#define SFP_EEPROM_CON_TYPE_VAL_LC	0x7
 	#define SFP_EEPROM_CON_TYPE_VAL_COPPER	0x21
 	#define SFP_EEPROM_CON_TYPE_VAL_RJ45	0x22
@@ -4210,6 +4211,16 @@ static void bnx2x_warpcore_set_sgmii_spe
 					 0x1000);
 		DP(NETIF_MSG_LINK, "set SGMII AUTONEG\n");
 	} else {
+		/* Note that 2.5G works only when used with 1G advertisment */
+		if (fiber_mode && (phy->req_line_speed == SPEED_2500) &&
+			(phy->speed_cap_mask &
+			 (PORT_HW_CFG_SPEED_CAPABILITY_D0_1G |
+			  PORT_HW_CFG_SPEED_CAPABILITY_D0_2_5G))) {
+			bnx2x_cl45_write(bp, phy, MDIO_WC_DEVAD,
+				MDIO_WC_REG_SERDESDIGITAL_MISC1,
+					0x6010);
+		}
+
 		bnx2x_cl45_read(bp, phy, MDIO_WC_DEVAD,
 				MDIO_WC_REG_COMBO_IEEE0_MIICTRL, &val16);
 		val16 &= 0xcebf;
@@ -4220,6 +4231,7 @@ static void bnx2x_warpcore_set_sgmii_spe
 			val16 |= 0x2000;
 			break;
 		case SPEED_1000:
+		case SPEED_2500:
 			val16 |= 0x0040;
 			break;
 		default:
@@ -8174,6 +8186,7 @@ static int bnx2x_get_edc_mode(struct bnx
 		break;
 	}
 	case SFP_EEPROM_CON_TYPE_VAL_UNKNOWN:
+	case SFP_EEPROM_CON_TYPE_VAL_SC:
 	case SFP_EEPROM_CON_TYPE_VAL_LC:
 	case SFP_EEPROM_CON_TYPE_VAL_RJ45:
 		check_limiting_mode = 1;
@@ -8184,7 +8197,8 @@ static int bnx2x_get_edc_mode(struct bnx
 		    (val[SFP_EEPROM_1G_COMP_CODE_ADDR] != 0)) {
 			DP(NETIF_MSG_LINK, "1G SFP module detected\n");
 			phy->media_type = ETH_PHY_SFP_1G_FIBER;
-			if (phy->req_line_speed != SPEED_1000) {
+			if ((phy->req_line_speed != SPEED_1000) &&
+				(phy->req_line_speed != SPEED_2500)) {
 				u8 gport = params->port;
 				phy->req_line_speed = SPEED_1000;
 				if (!CHIP_IS_E1x(bp)) {
@@ -8344,7 +8358,7 @@ static int bnx2x_wait_for_sfp_module_ini
 	 * some phys type ( e.g. JDSU )
 	 */
 
-	for (timeout = 0; timeout < 60; timeout++) {
+	for (timeout = 0; timeout < 1800; timeout++) {
 		if (phy->type == PORT_HW_CFG_XGXS_EXT_PHY_TYPE_DIRECT)
 			rc = bnx2x_warpcore_read_sfp_module_eeprom(
 				phy, params, I2C_DEV_ADDR_A0, 1, 1, &val,
@@ -9238,6 +9252,7 @@ static void bnx2x_8727_config_speed(stru
 	u16 tmp1, val;
 	/* Set option 1G speed */
 	if ((phy->req_line_speed == SPEED_1000) ||
+		(phy->req_line_speed == SPEED_2500) ||
 	    (phy->media_type == ETH_PHY_SFP_1G_FIBER)) {
 		DP(NETIF_MSG_LINK, "Setting 1G force\n");
 		bnx2x_cl45_write(bp, phy,
@@ -9247,6 +9262,22 @@ static void bnx2x_8727_config_speed(stru
 		bnx2x_cl45_read(bp, phy,
 				MDIO_PMA_DEVAD, MDIO_PMA_REG_10G_CTRL2, &tmp1);
 		DP(NETIF_MSG_LINK, "1.7 = 0x%x\n", tmp1);
+		if ((phy->req_line_speed == SPEED_2500) &&
+			(phy->speed_cap_mask &
+			(PORT_HW_CFG_SPEED_CAPABILITY_D0_1G |
+			 PORT_HW_CFG_SPEED_CAPABILITY_D0_2_5G))) {
+			bnx2x_cl45_read_and_write(bp, phy,
+					MDIO_AN_DEVAD,
+					MDIO_AN_REG_8727_MISC_CTRL2,
+					~(1<<5));
+			bnx2x_cl45_write(bp, phy,
+					MDIO_AN_DEVAD,
+					MDIO_AN_REG_8727_MISC_CTRL1, 0x0010);
+		} else {
+			bnx2x_cl45_write(bp, phy,
+					MDIO_AN_DEVAD,
+					MDIO_AN_REG_8727_MISC_CTRL1, 0x001C);
+		}
 		/* Power down the XAUI until link is up in case of dual-media
 		 * and 1G
 		 */
@@ -9268,7 +9299,7 @@ static void bnx2x_8727_config_speed(stru
 
 		DP(NETIF_MSG_LINK, "Setting 1G clause37\n");
 		bnx2x_cl45_write(bp, phy,
-				 MDIO_AN_DEVAD, MDIO_AN_REG_8727_MISC_CTRL, 0);
+				 MDIO_AN_DEVAD, MDIO_AN_REG_8727_MISC_CTRL2, 0);
 		bnx2x_cl45_write(bp, phy,
 				 MDIO_AN_DEVAD, MDIO_AN_REG_CL37_AN, 0x1300);
 	} else {
@@ -9276,9 +9307,12 @@ static void bnx2x_8727_config_speed(stru
 		 * registers although it is default
 		 */
 		bnx2x_cl45_write(bp, phy,
-				 MDIO_AN_DEVAD, MDIO_AN_REG_8727_MISC_CTRL,
+				 MDIO_AN_DEVAD, MDIO_AN_REG_8727_MISC_CTRL2,
 				 0x0020);
 		bnx2x_cl45_write(bp, phy,
+				 MDIO_AN_DEVAD, MDIO_AN_REG_8727_MISC_CTRL1,
+				 0x001C);
+		bnx2x_cl45_write(bp, phy,
 				 MDIO_AN_DEVAD, MDIO_AN_REG_CL37_AN, 0x0100);
 		bnx2x_cl45_write(bp, phy,
 				 MDIO_PMA_DEVAD, MDIO_PMA_REG_CTRL, 0x2040);
@@ -9567,6 +9601,11 @@ static u8 bnx2x_8727_read_status(struct
 		vars->line_speed = SPEED_10000;
 		DP(NETIF_MSG_LINK, "port %x: External link up in 10G\n",
 			   params->port);
+	} else if ((link_status & (1<<1)) && (!(link_status & (1<<14)))) {
+		link_up = 1;
+		vars->line_speed = SPEED_2500;
+		DP(NETIF_MSG_LINK, "port %x: External link up in 2.5G\n",
+			   params->port);
 	} else if ((link_status & (1<<0)) && (!(link_status & (1<<13)))) {
 		link_up = 1;
 		vars->line_speed = SPEED_1000;
@@ -9598,7 +9637,8 @@ static u8 bnx2x_8727_read_status(struct
 	}
 
 	if ((DUAL_MEDIA(params)) &&
-	    (phy->req_line_speed == SPEED_1000)) {
+	    ((phy->req_line_speed == SPEED_1000) ||
+	     (phy->req_line_speed == SPEED_2500))) {
 		bnx2x_cl45_read(bp, phy,
 				MDIO_PMA_DEVAD,
 				MDIO_PMA_REG_8727_PCS_GP, &val1);
@@ -11722,6 +11762,7 @@ static const struct bnx2x_phy phy_warpco
 			   SUPPORTED_100baseT_Full |
 			   SUPPORTED_1000baseT_Full |
 			   SUPPORTED_1000baseKX_Full |
+			   SUPPORTED_2500baseX_Full |
 			   SUPPORTED_10000baseT_Full |
 			   SUPPORTED_10000baseKR_Full |
 			   SUPPORTED_20000baseKR2_Full |
@@ -11908,6 +11949,7 @@ static const struct bnx2x_phy phy_8727 =
 	.tx_preemphasis	= {0xffff, 0xffff, 0xffff, 0xffff},
 	.mdio_ctrl	= 0,
 	.supported	= (SUPPORTED_10000baseT_Full |
+			   SUPPORTED_2500baseX_Full |
 			   SUPPORTED_1000baseT_Full |
 			   SUPPORTED_FIBRE |
 			   SUPPORTED_Pause |
@@ -12255,6 +12297,7 @@ static int bnx2x_populate_int_phy(struct
 			break;
 		case PORT_HW_CFG_NET_SERDES_IF_SFI:
 			phy->supported &= (SUPPORTED_1000baseT_Full |
+					   SUPPORTED_2500baseX_Full |
 					   SUPPORTED_10000baseT_Full |
 					   SUPPORTED_FIBRE |
 					   SUPPORTED_Pause |
@@ -13939,7 +13982,8 @@ void bnx2x_period_func(struct link_param
 		    & PORT_HW_CFG_NET_SERDES_IF_MASK) ==
 		    PORT_HW_CFG_NET_SERDES_IF_SFI) {
 			if (bnx2x_is_sfp_module_plugged(phy, params)) {
-				bnx2x_sfp_tx_fault_detection(phy, params, vars);
+				if(!((params->port + 1) & bp->mask_tx_fault))
+					bnx2x_sfp_tx_fault_detection(phy, params, vars);
 			} else if (vars->link_status &
 				LINK_STATUS_SFP_TX_FAULT) {
 				/* Clean trail, interrupt corrects the leds */
--- a/drivers/net/ethernet/broadcom/bnx2x/bnx2x_main.c
+++ b/drivers/net/ethernet/broadcom/bnx2x/bnx2x_main.c
@@ -112,6 +112,10 @@ static int disable_tpa;
 module_param(disable_tpa, int, 0444);
 MODULE_PARM_DESC(disable_tpa, " Disable the TPA (LRO) feature");
 
+static int mask_tx_fault;
+module_param(mask_tx_fault, int, 0444);
+MODULE_PARM_DESC(mask_tx_fault, " Mask SFP TX fault detection");
+
 static int int_mode;
 module_param(int_mode, int, 0444);
 MODULE_PARM_DESC(int_mode, " Force interrupt mode other than MSI-X "
@@ -12344,6 +12348,8 @@ static int bnx2x_init_bp(struct bnx2x *b
 	if (BP_NOMCP(bp) && (func == 0))
 		dev_err(&bp->pdev->dev, "MCP disabled, must load devices in order!\n");
 
+	bp->mask_tx_fault = mask_tx_fault;
+
 	bp->disable_tpa = disable_tpa;
 	bp->disable_tpa |= !!IS_MF_STORAGE_ONLY(bp);
 	/* Reduce memory usage in kdump environment by disabling TPA */
--- a/drivers/net/ethernet/broadcom/bnx2x/bnx2x_reg.h
+++ b/drivers/net/ethernet/broadcom/bnx2x/bnx2x_reg.h
@@ -7169,7 +7169,8 @@ Theotherbitsarereservedandshouldbezero*/
 #define MDIO_PMA_REG_8727_PCS_GP		0xc842
 #define MDIO_PMA_REG_8727_OPT_CFG_REG		0xc8e4
 
-#define MDIO_AN_REG_8727_MISC_CTRL		0x8309
+#define MDIO_AN_REG_8727_MISC_CTRL1		0x8308
+#define MDIO_AN_REG_8727_MISC_CTRL2		0x8309
 
 #define MDIO_PMA_REG_8073_CHIP_REV			0xc801
 #define MDIO_PMA_REG_8073_SPEED_LINK_STATUS		0xc820

jjm2473, 2025-05-06:
所以编译出啥错，日志看看

Jaykwok2999, 2025-05-06:
[image]

jjm2473, 2025-05-07:
这哪是编译错误，是补丁都打不上，你找的补丁是6.6内核的吗？

Jaykwok2999, 2025-05-07:
是的

jjm2473, 2025-05-19:
因为这个补丁本来就已经打了，你又还弄了个相同的补丁，产生冲突了。