istoreos/istoreos #2301: 【24.10测试】passwall 不正常

flypighd, 2025-05-10:
无法科学上网

2025-05-10 17:12:33: 删除 nftables 规则完成。
2025-05-10 17:12:33: 清空并关闭相关程序和缓存完成。
2025-05-10 17:12:34: TCP节点：[香港06]，监听端口：1041
2025-05-10 17:12:34: UDP节点：与TCP节点相同
2025-05-10 17:12:34: DNS域名解析：
2025-05-10 17:12:34:   * 请确认上游 DNS 支持 TCP/DoT/DoH 查询，如非直连地址，确保 TCP 代理打开，并且已经正确转发！
2025-05-10 17:12:34:   - ChinaDNS-NG(127.0.0.1#15353)：直连DNS：127.0.0.1#7874，可信DNS：tcp://8.8.8.8#53
2025-05-10 17:12:34:   - 节点列表中的域名(vpslist)：127.0.0.1#7874
2025-05-10 17:12:34:   - 域名白名单(whitelist)：127.0.0.1#7874
2025-05-10 17:12:34:   - 代理域名表(blacklist)：tcp://8.8.8.8#53
2025-05-10 17:12:34:   - 防火墙域名表(gfwlist)：tcp://8.8.8.8#53
2025-05-10 17:12:34:   - 中国域名表(chnroute)：127.0.0.1#7874
2025-05-10 17:12:34:   - 默认 DNS ：智能匹配
2025-05-10 17:12:34:   - ChinaDNS-NG已作为Dnsmasq上游，如果你自行配置了错误的DNS流程，将会导致域名(直连/代理域名)分流失效！！！
2025-05-10 17:12:34: 开始加载防火墙规则...
2025-05-10 17:12:35:   - [0]追加ISP IPv4 DNS到白名单：114.114.114.114
2025-05-10 17:12:35:   - [0]追加ISP IPv4 DNS到白名单：202.101.172.46
2025-05-10 17:12:35:   - [1]加入所有IPv4节点到nftset[passwall_vps]直连完成
2025-05-10 17:12:35:   - [1]加入所有IPv6节点到nftset[passwall_vps6]直连完成
2025-05-10 17:12:35:   - [0]追加WAN IP到nftables：60.186.172.191
2025-05-10 17:12:35:   - 【路由器本机】，屏蔽代理 UDP 端口[443]
2025-05-10 17:12:35:   - [0]将上游 DNS 服务器 8.8.8.8:53 加入到路由器自身代理的 TCP 转发链
2025-05-10 17:12:35:   - 访问控制：
2025-05-10 17:12:35:      - 【默认】，屏蔽代理 UDP 端口[443]
2025-05-10 17:12:35:      - 【默认】，使用 TCP 节点[香港06](REDIRECT:1041)
2025-05-10 17:12:35:      - 【默认】，使用 UDP 节点[香港06](TPROXY:1041)
2025-05-10 17:12:35: 防火墙规则加载完成！
2025-05-10 17:12:35: 运行完成！

tcp日志显示
Error loading shared library libsodium.so.26: No such file or directory (needed by /tmp/etc/passwall/bin/ssr-redir)
Error relocating /tmp/etc/passwall/bin/ssr-redir: sodium_init: symbol not found
Error relocating /tmp/etc/passwall/bin/ssr-redir: crypto_stream_salsa20_xor_ic: symbol not found
Error relocating /tmp/etc/passwall/bin/ssr-redir: sodium_memzero: symbol not found
Error relocating /tmp/etc/passwall/bin/ssr-redir: crypto_stream_chacha20_ietf_xor_ic: symbol not found
Error relocating /tmp/etc/passwall/bin/ssr-redir: crypto_stream_chacha20_xor_ic: symbol not found
Error relocating /tmp/etc/passwall/bin/ssr-redir: randombytes_buf: symbol not found

bcseputetto, 2025-05-13:
ssr的SSL算法的问题，ssl3抛弃了一些过时的加密算法，所以你机场的ssr节点还在用过时的加密算法，可能会造成一些额外的问题。
你可以在 tg 联系我 @seputetto 因为我手里没有使用被抛弃的ssl加密算法的节点

bcseputetto, 2025-05-13:
`libsodium`在安装`ssr-libev`时，理论上会自动以依赖形式被安装，也麻烦你确认一下`libsodium`是否有被正常安装

flypighd, 2025-05-13:
> `libsodium`在安装`ssr-libev`时，理论上会自动以依赖形式被安装，也麻烦你确认一下`libsodium`是否有被正常安装

libsodium 有安装 是1.0.18-4版本

bcseputetto, 2025-05-13:
是从 22.03升级上来 `libsodium`包有问题
```
opkg update
opkg install libsodium --force-reinstall
```
重新安装 libsodium 应该可以解决

flypighd, 2025-05-13:
> 是从 22.03升级上来 `libsodium`包有问题
> 
> ```
> opkg update
> opkg install libsodium --force-reinstall
> ```
> 
> 重新安装 libsodium 应该可以解决

已解决，谢谢大佬

closed 2025-05-13 by flypighd

bcseputetto, 2025-06-07:
@shortgun77 你是 24.10 固件重置了？那你重置固件后，用这个 https://github.com/bcseputetto/Are-u-ok 项目的24.10的包安装，还有问题？