istoreos/istoreos #2305: 【24.10测试】  双拨号成功，但是MultiWAN不正常导致速度不叠加

longaga, 2025-05-11:
测试环境:
硬件:X86
固件:24.10 0422 

1、iStoreiStore 里不能直接安装多线多拨、多账号多拨，提示缺IP tables 的一些依赖
2、通过 22.03 升级到24.10，双拨号成功，但是MultiWAN不正常导致速度不叠加

请问这个能解决吗？

Shinku-Chen, 2025-05-12:
关注
目前mwan3 只能实现单线,不能实现叠加

vs2pk0, 2025-05-13:
24.10固件地址在哪

longaga, 2025-05-13:
全新刷写固件不保留配置 ，MultiWAN终于正常工作了

bupahaibao, 2025-05-16:
> 关注 更新固件不保留配置 目前mwan3 只能实现单线,不能实现叠加,,可以通过策略制定一条wan作为出口 但不能速率叠加
> 
> 有时间全盘刷固件试试,

可以叠加，我用测速软件实测过。

bupahaibao, 2025-05-16:
> 测试环境: 硬件:X86 固件:24.10 0422
> 
> 1、iStore商店里不能直接安装多线多拨、多账号多拨，提示缺IP tables 的一些依赖 2、通过 22.03 升级到24.10，双拨号成功，但是MultiWAN不正常导致速度不叠加
> 
> 请问这个能解决吗？
> 
> **全新刷写固件不保留配置 ，MultiWAN终于正常工作了**

安装iptables-nft和ip6tables-nft这两个软件包。

Shinku-Chen, 2025-05-16:
> > 测试环境: 硬件:X86 固件:24.10 0422
> > 1、iStore商店里不能直接安装多线多拨、多账号多拨，提示缺IP tables 的一些依赖 2、通过 22.03 升级到24.10，双拨号成功，但是MultiWAN不正常导致速度不叠加
> > 请问这个能解决吗？
> > **全新刷写固件不保留配置 ，MultiWAN终于正常工作了**
> 
> 安装iptables-nft和ip6tables-nft这两个软件包。

这个默认是已经安装了的

closed 2025-11-28 by jjm2473