istoreos/istoreos #2307: istoreos内置的统一文件共享，开启的SMB服务，一旦istoreos重启，就会导致客户端凭证失效。

kokoko777, 2025-05-12:
istoreos内置的统一文件共享，开启的SMB服务，一旦istoreos重启，就会导致SMB客户端凭证失效。
无论是Windows还是手机上的SMB客户端，都没法连上。会提示密码不正确。

唯一的解决办法就是在“统一文件共享”里随意修改一处设置，然后点 保存并应用 。就能让客户端恢复连接。
我平常就是重启后，把“允许匿名用户”点开，然后保存应用。下次重启了，又关掉 允许匿名用户，再保存应用。每次重启都反复如此操作一遍。简直太心累了。有时候我不在家，路由器重启了。我爸在家就连不上SMB。我又得远程操作一遍。

但是，WebDAV的配置又是正常的。WebDAV配置好之后，随便重启，都不会导致凭证失效，可以随便连上。但是我不喜欢WebDAV，因为这个WebDAV挂载后，它不支持拖入一个同名的文件夹并选择跳过或者覆盖。他会直接拒绝本次操作。WebDAV只能复制同名文件进去并覆盖，想复制同名文件夹做不到。

真的是排查不出这个SMB服务端重启后就连不上，问题在哪。

[image]
[image]
[image]
[image]
[image]

[image]

kokoko777, 2025-05-13:
给各位遇到和我一样的问题的网友支一个招，可以完美解决这个问题。在 系统 - 启动项 - 本地启动脚本 里面，添加一条指令。
(sleep 5; /etc/init.d/unishare restart) &
该指令能做到延迟5秒钟将 unishare 这个程序在开机自启之后，再重启一遍，等同于替你手动操作一次程序重启。这样程序就正常了，SMB客户端不再会提示密码不正确。

Spidhunter, 2025-05-13:
我看这个系统有3处文件共享管理设置，是3个不同的应用吗？我用“统一文件共享”，Windows连不上smb，挂webdav无修改权限，只是无修改权限，可以访问，可以删除，但重命名不行。猜测是账户问题，但已经设置管理员账户。当时因为硬盘休眠用这个系统，但发现文件管理太糟心了，文件夹属性也看不了，账户权限管理不清晰，像有些docker服务，直接可以访问nas所有路径，风险很大

ZhuangJW, 2025-05-18:
我也遇到过，解决不了，是在/etc/passwd上的共享用户的前面多了很多@，把前面的@删除了就好了

kokoko777, 2025-05-21:
> 统一文件共享中，对于一个共享文件夹，如果某个用户已经选择作为“读写用户”，就不需要在“只读用户”勾选了，“读写”是包括了读权限的。

没有用，就在刚刚，我把“网络共享”里的分享路径全部删除，在统一文件共享中，将只读用户全部去勾选了，只在读写用户里选择了“user”，然后保存。再回到 网络共享 里。所有的共享目录，依旧被勾选上了 只读。然后重启一遍。老问题，凭证错误。

kokoko777, 2025-05-21:
> 统一文件共享中，对于一个共享文件夹，如果某个用户已经选择作为“读写用户”，就不需要在“只读用户”勾选了，“读写”是包括了读权限的。

[image]

[image]

jjm2473, 2025-05-23:
这个只读是默认的，因为有另一个参数控制部分用户可写，这里不会显示这个参数。只要统一文件共享配置了samba，那么系统的samba会生成相应的配置项，别去改这些生成的配置项，改了也会被覆盖的。

对于重启以后权限异常的问题，将在新版本固件修复。

kokoko777, 2025-06-10:
> 新固件应该修复了

今天一觉醒来。这个插件彻底失效了...无论我如何设置，Windows都提示凭证不正确。只好删除掉unishare的的一切设置。让chatGPT从头开始教我创建 smb用户，和配置smb共享。现在才一切正常了。unishare这个插件问题真的蛮多的。

jjm2473, 2025-06-10:
@ZhuangJW 

> 我也遇到过，解决不了，是在/etc/passwd上的共享用户的前面多了很多@，把前面的@删除了就好了

下次遇到这个问题截图看看到底是怎样的