istoreos/istoreos #2308: 关于打开istoreos管理页面慢的问题，和进入SSH慢的问题，发现了个临时处理的办法

ZhuangJW, 2025-05-12:
具体的问题表面上是管理页面打开卡慢甚至多多刷新几次才能打开，这个问题蔓延到了SSH，什么是docker的容器端口问题。
        我先说一下目前的配置是我一个口做wan口走ppoe拨号，另一个口分发给ap-wifi , 之前也提过以为是什么连接数的问题，目前测试来看的话应该就不是了，因为我发现我重启了ppoe的接口，我打开管理页面，进入ssh,进入docker的服务就会变回很顺畅。
        我查不出什么问题，也没看到什么日志信息啥的，但是只要我重启了一下ppoe的接口，登录ssh，打开管理页面就会变得顺畅，因为有些容器的端口和lan口IP的端口有绑定的，这已经实质影响到了一些服务了。
        如果其它人有遇到我这样的问题也可以尝试一下，重启wan口看看

jjm2473, 2025-05-16:
有意思，wan口影响到了lan口。先备份下，然后恢复出厂设置或者不保留配置刷机，然后立马改后台密码，复杂点，然后再配置pppoe，看看海有没有这个问题。

ZhuangJW, 2025-05-21:
> 有意思，wan口影响到了lan口。先备份下，然后恢复出厂设置或者不保留配置刷机，然后立马改后台密码，复杂点，然后再配置pppoe，看看还有没有这个问题。

一开始没有，但是重启了几遍之后就又出现了，比较离奇的是我先装PVE然后再在PVE上建个虚拟机装istoreos的话就没有什么问题。可能这是我设备的悬案了

ZhuangJW, 2025-05-23:
> 有看到cpu使用率飙升吗？

没看到，CPU使用率，温度，内存都正常甚至偏低。我装原生openwrt也有这种情况，但是就像时不时出现一样，重装了就好，出现一次的话就一直在了

closed 2025-05-23 by jjm2473