istoreos/istoreos #2311: IP/MAC绑定功能图形界面用不了，只能终端来绑定

MrHaYu, 2025-05-13:
我在 ’网络→DHCP/DNS→静态地址分配→添加‘ 处无法自定义mac和ip地址，用的是 
iStoreOS 22.03.7 2025050912 版本，浏览器源代码有 unselectable 字样，我使用命令
uci add dhcp host
uci set dhcp.@host[-1].name='homeserver.lan'
uci set dhcp.@host[-1].mac='AA:BB:CC:DD:EE:FF' # 设备mac 
uci set dhcp.@host[-1].ip='192.168.5.2' # 设备ip
uci commit dhcp
可以正常设置绑定的mac及ip，均可自定义，可正常生效，浏览器界面如图

[image]

我想达成的效果如图（使用终端命令）

[image]

希望给出解决方法

MrHaYu, 2025-05-13:
补充：浏览器ui无法选择自定义，只能使用终端命令

bcseputetto, 2025-05-15:
跟 `unselectable` 没有关系，我这边测试可以正常输入，输入完后按回车就行了

XReakz-X, 2025-05-15:
> 跟 `unselectable` 没有关系，我这边测试可以正常输入，输入完后按回车就行了

我换了别的浏览器试了一下，貌似是chome浏览器的问题，版本 136.0.7103.114（正式版本） （64 位），但是使用edge浏览器就正常，内核我记得都是谷歌的，有没有什么浏览器设置会出现这种问题的

jjm2473, 2025-05-16:
chrome试试无痕模式，以避免浏览器插件的影响。如果无痕模式没问题，那就是浏览器插件导致的了

XReakz-X, 2025-05-16:
> chrome试试无痕模式，以避免浏览器插件的影响。如果无痕模式没问题，那就是浏览器插件导致的了

感谢指导，找到问题了，我逐一尝试发现是一个允许复制插件导致的，关闭后问题已解决

[image]

closed 2025-05-16 by XReakz-X