istoreos/istoreos #2314: 京东亚瑟JD-AX1800，软件源失效问题如何解决呢？

Byarain, 2025-05-14:
出现以下错误：
root@iStoreOS:~# opkg update
Downloading https://istore.istoreos.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/targets/ipq60xx/generic/packages/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/targets/ipq60xx/generic/packages/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/base/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/base/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/istore/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/istore/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/luci/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/luci/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas_luci/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas_luci/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/packages/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/packages/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/routing/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/routing/Packages.gz

Downloading https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/telephony/Packages.gz
*** Failed to download the package list from https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/targets/ipq60xx/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/istore/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/nas_luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://mirrors.ustc.edu.cn/openwrt/releases/19.07-SNAPSHOT/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 8.

bcseputetto, 2025-05-15:
https://mirrors.ustc.edu.cn/openwrt/releases/
本来就没有 19.07-SNAPSHOT 这个源，这个机型不由KoolCenter维护。

jjm2473, 2025-05-16:
找固件作者，跟istoreos无关。

closed 2025-05-16 by jjm2473