istoreos/istoreos #2315: 连接手机USB共享网络不可用

daadaer999, 2025-05-14:
手机连接USB后打开手机USB共享网络，创建接口时找不到usb接口。
22.03是可以的，所以又回到了22.03

jjm2473, 2025-05-16:
rndis驱动是集成了的。你接上手机以后，执行下`ip link`看看

daadaer999, 2025-05-18:
[image]没有看到USB

jjm2473, 2025-05-21:
`2717:ff40` 是红米手机的MTP模式，不是RNDIS https://devicehunt.com/view/type/usb/vendor/2717/device/FF40

609293964, 2025-05-23:
同一个硬件，为什么22可以识别rndis_host，24固件只能usbfs？ @jjm2473 
Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
    |__ Port 001: Dev 007, If 0, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 1, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 2, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 3, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 4, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 5, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 6, Class=[unknown], Driver=usbfs, 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 7, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 8, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  
    |__ Port 001: Dev 007, If 9, Class=[unknown], Driver=[none], 480M
        ID 1782:4038  

旧的固件
  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/4p, 480M
        ID 05e3:0610  
        |__ Port 4: Dev 6, If 0, Class=, Driver=rndis_host, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 1, Class=, Driver=rndis_host, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 2, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 3, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 4, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 5, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 6, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 7, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 8, Class=, Driver=, 480M
            ID 1782:4038  
        |__ Port 4: Dev 6, If 9, Class=, Driver=, 480M
            ID 1782:4038