istoreos/istoreos #2317: [24.10] miniupnp在防火墙里 不断的重复添加了几百条规则

whetherTsmile, 2025-05-15:
设备是mac， 软件是 parsec 

root@iStoreOS:~# nft list chain inet fw4 upnp_forward | wc
      554      5509     41844

root@iStoreOS:~# nft list chain inet fw4 upnp_forward
table inet fw4 {
        chain upnp_forward {
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 31497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22497 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22498 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22499 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                iif "pppoe-wan" th dport 22500 @nh,128,32 0xc0a80508 @nh,72,8 0x11 accept
                ..........

jjm2473, 2025-05-16:
这个问题在22.03页存在，不知道是不是miniupnp的bug，还是因为打了允许强制转发的补丁。你试试在upnp的高级设置把强制转发关掉，然后重启。不过没有公网ip的话，可能就完全不会建立转发了。

whetherTsmile, 2025-05-16:
> 这个问题在22.03页存在，不知道是不是miniupnp的bug，还是因为打了允许强制转发的补丁。你试试在upnp的高级设置把强制转发关掉，然后重启。不过没有公网ip的话，可能就完全不会建立转发了。

我是公网， 使用stun，强制转发都关了， 是可以建立转发的。  我暂时是定时任务清理一下链

jjm2473, 2025-05-16:
有公网的话，stun也不需要了。今天更新了固件，miniupnp打了补丁，可能解决了这个问题。

jjm2473, 2025-05-19:
测试下最新的固件，看是不是已经解决这个问题了

closed 2025-05-23 by jjm2473