istoreos/istoreos #2325: [24.10] Web界面无法进入 UPnP IGD 和 PCP 管理界面，点击即卡死

Mekiduan, 2025-05-16:
在 OpenWrt 24.10 版本中，点击 LuCI Web 界面中的“UPnP IGD 和 PCP”页面（路径：cgi-bin/luci/admin/services/upnp）时，页面出现 XHR request timed out 错误，随后整个 Web 界面崩溃并无法登录。必须通过重启 rpcd 服务才能恢复 Web 界面访问。

在系统日志中，miniupnpd 进程报错如下：
Sat May 17 00:57:21 2025 daemon.err miniupnpd[9524]: send_batch: mnl_cb_run returned -1  
Sat May 17 00:57:21 2025 daemon.err miniupnpd[9524]: nft_send_rule(0x7f794eb62020, 6, 2) send_batch failed -4  
Sat May 17 00:57:21 2025 daemon.err miniupnpd[9524]: refresh_nft_cache: mnl_cb_run returned -1

推测原因：
根据日志，怀疑是 miniupnpd 与内核 nftables 通信异常，导致 rpcd 服务阻塞，无法正常返回数据，从而导致 LuCI 前端请求卡死，Web 界面崩溃。

复现步骤：
进入 LuCI Web 界面
点击「UPnP IGD 和 PCP」菜单项
页面开始加载，随后出现超时错误  XHR request timed out
Web 界面崩溃且无法继续登录

通过命令 etc/init.d/rpcd restart 重启 rpcd 服务后恢复正常

mengls666, 2025-05-19:
同样遇到了这个问题。同时观察到切换到24.10后，upnp映射数明显增多，nas版迅雷隔一段时间就会映射相同的内部端口，被upnp映射到不同的外部端口，且过期时间为168h，猜测是映射数过多导致upnp崩溃

jjm2473, 2025-05-21:
可能是数据量太大传送不完。使用最新的固件看看修复没

mengls666, 2025-05-21:
更新到最新固件24.10.0后已修复。

closed 2025-05-22 by Mekiduan