istoreos/istoreos #2328: iStore安装alist后，点击打开不了

quanliantechnology, 2025-05-17:
iStore安装alist后，点击打开不了，提示No page is registered at '/admin/services/alist。。。

[image]
服务和网络储存也没有alist的菜单

[image]
删除了缓存文件之后也没有出现，但是启动项管理那里有alist。
就很迷惑，尝试安装其他插件是没有问题的。就是安装alist有这个问题。

2406583758qq, 2025-05-20:
你好，请问解决了吗

derrickchang04, 2025-06-22:
我也是这样，不知道怎么弄

jjm2473, 2025-06-23:
卸载重装，安装过程的日志发出来看看。

derrickchang04, 2025-06-23:
> 卸载重装，安装过程的日志发出来看看。

[image]

[image]

[image]

有劳帮忙看看有没有这么问题 感谢

jjm2473, 2025-06-24:
哦，我知道了，你这个固件太老了，还是luci 1.0，不支持alist的luci

closed 2025-06-24 by jjm2473

jjm2473, 2025-06-24:
检查系统中有没有`/www/luci-static/resources/luci.js`，如果没有这个文件，说明只支持lua，不支持js，现在有些插件已经用js重写了，不再支持老版本的luci

jjm2473, 2025-06-24:
@derrickchang04 可以不用截图，安装完点一下查看日志，可以滚动日志到前面，然后复制下来。