istoreos/istoreos #2332: x86 从22.03.7升级到2424.10.0 登录页显示Bad Gateway

lonhowe, 2025-05-17:
22的固件是22.03.7 2025051615
24的固件是24.10.0 2025051619

在后台使用升级包保留配置升级 升级后就是显示Bad Gateway

Bad Gateway
The process did not produce any response

升级前使用 reset_rom_pkgs修复了下 也不行
配置是i3 N305 装有Docker和OpenClash Passwall等

lonhowe, 2025-05-17:
[image]

Aluan1996, 2025-05-23:
同样的问题，升级以后没法使用了

jjm2473, 2025-05-23:
可能是openclash的问题，卸载掉再升级试试

lonhowe, 2025-05-25:
> 可能是openclash的问题，卸载掉再升级试试

测试了下 卸载了openclash 也是同样问题~

wibi8bo, 2025-05-26:
> 22的固件是22.03.7 2025051615
> 24的固件是24.10.0 2025051619
> 
> 在后台使用升级包保留配置升级 升级后就是显示Bad Gateway并且也无法连接网络 SSH 也无法连接
> 
> Bad Gateway
> The process did not produce any response
> 
> 
> 升级前使用 reset_rom_pkgs修复了下 也不行
> 配置是i3 N305 装有Docker和OpenClash Passwall等
> 
> 
> 

请问解决了吗，我今天刚在后台升级的，到现在还没进得去后台，我哭死

Aluan1996, 2025-05-26:
搞不懂什么原因，已经刷回老系统了

MWT6433, 2025-05-26:
@jjm2473 
我也有同样问题，以下是输出
Syntax error: Unable to resolve path for module 'fs'
In line 1, byte 35:

 `import { stdin, stdout } from 'fs';`
  Near here ------------------------^
我的/usr/lib/ucode/文件夹是空的

jjm2473, 2025-05-26:
`/usr/lib/ucode/` 是空的？有什么程序删掉这个文件夹了吗？

@MWT6433 你使用了沙箱吗？

执行这个命令修复并重启`rm -rf /overlay/upper/usr/lib/ucode /ext_overlay/upper/usr/lib/ucode; reboot`

MWT6433, 2025-05-26:
@jjm2473 不太清楚，在升级之后发现nginx拒绝访问，logread发现是nginx-util不兼容新的库的错误，卸载nginx并重装之后，就显示了502Bad Gateway

MWT6433, 2025-05-26:
@jjm2473 重启之后访问路由器后台，logread报错
`uwsgi-luci: Syntax error: Unable to resolve path for module 'luci.dispatcher' In line 7, byte 39:   `import dispatch from 'luci.dispatcher';`   Near here ----------------------------^   Syntax error: Unable to resolve path for module 'luci.http' In line 8, byte 32:   `import request from 'luci.http';`   Near here ---------------------^
Mon May 26 23:59:09 2025 daemon.err nginx[10445]: 2025/05/26 23:59:09 [error] 10983#0: *3 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.3, server: _redirect2ssl, request: "GET /cgi-bin/luci/ HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.100.1", referrer: "http://192.168.100.1/"`

jjm2473, 2025-05-26:
其他遇到这个问题的人也卸载过nginx吗？

MWT6433, 2025-05-26:
@jjm2473 执行命令后输出
root@FXAL:/usr/lib/ucode# ls -l /overlay/upper/usr/share/ucode/luci/
ls: /overlay/upper/usr/share/ucode/luci/: No such file or directory
重启之后再执行命令还是No such file or directory

jjm2473, 2025-05-26:
@MWT6433 那么看看`ls -l /overlay/upper/usr/share/; ls -l /overlay/upper/usr/share/ucode/`是什么，然后`rm -rf /overlay/upper/usr/share/ucode; reboot`

jjm2473, 2025-05-27:
@MWT6433 执行这个`( cd /overlay/upper/; getfattr -R -P -h --match='trusted.overlay.opaque' upper 2>/dev/null | grep '^# file: ' | cut -c9- )`看看输出什么。

MWT6433, 2025-05-27:
@jjm2473 执行之后什么都没有输出

jjm2473, 2025-05-27:
@MWT6433 先执行`ls -l /overlay/upper/usr/share/rpcd/ucode`看看里面有没有东西，然后`rm -rf /overlay/upper/usr/share/rpcd/ucode; reboot`

jjm2473, 2025-05-27:
为啥都是ucode文件夹出问题，你在卸载nginx的时候把ucode卸载了吗

michaelrrrrrr, 2025-05-30:
> 我也是 Bad Gateway: The process did not produce any response 尝试 https://github.com/istoreos/istoreos/wiki/FAQ 的第二步不行； /overlay/upper/usr/share/rpcd/ucode 这个路径下没有东西， `( cd /overlay/; getfattr -R -P -h --match='trusted.overlay.opaque' upper 2>/dev/null | grep '^# file: ' | cut -c9- )` 输出很多：
> 
> > upper/root/.ssh
> > upper/opt/containerd
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/boot
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/root/.config
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/root/.android
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/overlay/upper
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/opt/containerd
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/host
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/usr/scgi_temp
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/usr/uwsgi_temp
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4/diff/etc/docker
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4-init/diff/dev/pts
> > upper/opt/docker/overlay2/18df663f77d99062a0c2600119cfe377b2d6fd2f0e176baf007e0849e9b17db4-init/diff/dev/shm
> > upper/www/luci-static/tailscaler
> > upper/usr/bin/pushbot
> > upper/usr/share/ucitrack
> > upper/usr/lib/engines-3
> > upper/usr/lib/ucode
> > upper/usr/lib/lua/luci/model/cbi/pushbot
> > upper/usr/lib/lua/luci/view/gogs
> > upper/usr/lib/lua/luci/view/pushbot
> > upper/usr/lib/lua/luci/view/tailscaler
> > upper/usr/lib/ruby
> > upper/etc/tailscale
> 
> 执行 `echo "0.0.0" >/overlay/.upgrading ; touch /overlay/upper/usr/lib/opkg/.upgrading 2>/dev/null ; reboot` 之后，WebUI 还是不能访问。 求助～

执行 `rm -rf /overlay/upper/usr/lib/ucode /ext_overlay/upper/usr/lib/ucode; reboot` 后 WebUI 恢复了

lonhowe, 2025-05-31:
rm -rf /overlay/upper/usr/lib/ucode /ext_overlay/upper/usr/lib/ucode; reboot
使用此命令成功修复

bdingtech, 2025-07-06:
> rm -rf /overlay/upper/usr/lib/ucode /ext_overlay/upper/usr/lib/ucode; reboot 使用此命令成功修复

有效！！！
但是不知道会不会有什么影响

zhangyan56, 2025-07-14:
👍👍👍大版本更新直接提示bad Gateway 按照大佬的操作成功进入操作界面

zhangyan56, 2025-07-14:
> @MWT6433 执行`echo "0.0.0" >/overlay/.upgrading ; touch /overlay/upper/usr/lib/opkg/.upgrading 2>/dev/null ; reboot`让系统自己修复核心组件试试，相当于重新跑一次升级过程

按照这个命令成功进入系统👍👍

itlaohui, 2025-07-20:
> `/usr/lib/ucode/` 是空的？有什么程序删掉这个文件夹了吗？
> 
> [@MWT6433](https://github.com/MWT6433) 你使用了沙箱吗？
> 
> 执行这个命令修复并重启`rm -rf /overlay/upper/usr/lib/ucode /ext_overlay/upper/usr/lib/ucode; reboot`

问题解决了，非常感谢🙏