istoreos/istoreos #2337: rockchip rk3568及rk3588  Motorcomm(yt8531c及yt8511h)的phy网卡驱动均不支持

yizhanghong, 2025-05-19:
目前测试了两款开发板， Motorcomm的网卡驱动均无法识别
其中一款为imb3588 使用两颗 Motorcomm的yt8531c，dts地址在此处 https://github.com/yizhanghong/Actions-iStoreOS-RK35XX/blob/main/configfiles/rk3588-imb3588.dts；可以正常编译成功，编译后的版本在此处 https://github.com/yizhanghong/Actions-iStoreOS-RK35XX/releases/tag/2025.05.16-08.26，日志下载地址 https://results-receiver.actions.githubusercontent.com/rest/runs/fc1389b9-73b8-4205-bd68-5adcfb66204d/logs?filename=logs_38652733967.zip&signature=1747614538.d6d49368b9d528317a89de503ac2ccfc0f71df8b1fc48fcf24eac44b0e975b0d

网卡获取不到地址

另一款主板为jetron jetnet-lu，采用两颗yt8511h芯片，也均无法获取地址
目前istoreos22版本的rk35xx的dts设备库中，并未发现支持 Motorcomm有线网卡的设备，是否目前版本不支持？

jjm2473, 2025-05-21:
Motorcomm 早就支持了，R2C就用了。

* 22.03 https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/files/drivers/net/phy/motorcomm.c
* 24.10 内核已经自带

jjm2473, 2025-05-21:
就像楼上说的，把开机过程内核日志发出来看看，如果phy无法识别会有报错

jjm2473, 2025-05-22:
这东西就没有2.5G网口吗？2.5G口应该能用。如果串口输入命令有效，读取数据乱码，要么就是接触不良，要么就是bit位奇偶校验位不匹配。

yizhanghong, 2025-05-22:
@jjm2473  木有2.5G网口，工控媒体板，能省就省了，目前是2颗yt8531c 无线是ap6275p，看日志，网卡是能识别的，但是获取不到地址。。

jjm2473, 2025-05-22:
R2C 使用的是 YT8521S ，或许你也可以试试换成这个，如果gmac还有问题，那基本就是dts没写对。

yizhanghong, 2025-05-22:
@jjm2473  usb网卡是想查看一下内核日志接的，但是并未生效；看日志eth0是被驱动的，但是网口并不能获取到任何dhcp地址，之前查看此链接[https://blog.csdn.net/weixin_43245577/article/details/141671972](https://blog.csdn.net/weixin_43245577/article/details/141671972)，看此链接有可能是motorcomm协商速率的问题，另外一款开发板，为双yt8511h,编译istoreos22也是同意问题，网卡可被识别，地址一直无法获取，后采用 led-rockchip [https://github.com/DHDAXCW/lede-rockchip.git](https://github.com/DHDAXCW/lede-rockchip.git),可成功识别yt8511h, 猜测是否是motorcomm后续驱动版本有更新？？？
   也曾尝试将imb3588的dts移植到lede-rockchip主线，但是固件未启动且没有ap6275p的驱动。另外上面提到的 nanopi的r2c网卡与yt8531c型号也不同

xiaomeng9597, 2025-05-22:
按下面这个移植步骤移植机型在编译，你没复制 02_network 网络配置文件到target/linux/rockchip/rk35xx/base-files/etc/board.d/02_network路径下，这个文件很重要，你的网口能不能用和位置都是它生成的。

jm10 3588和tn3568移植记录地址
https://github.com/istoreos/istoreos/pull/2027/files

xiaomeng9597, 2025-05-22:
你先配置02_network后，看看能不能用，不能用在修改dts文件

yizhanghong, 2025-05-22:
@xiaomeng9597 感谢提醒，我尝试编译一下

xiaomeng9597, 2025-05-22:
[image]

yizhanghong, 2025-05-23:
@xiaomeng9597 @jjm2473 
1、首先感谢两位大佬的协助，yx-imb3588设备的网口已经设配成功，正如@xiaomeng9597所提醒的，在target/linux/rockchip/rk35xx/base-files/etc/board.d/02_network路径下添加imb3588的接口和mac配置即可。
2、另外关于@jjm2473提醒的eth1没有产生的问题，也得到解决。参考rk3588的基础引用rk3588.dtsi，缺少了gmac1: ethernet@fe1c0000的相关配置，反而rk3588s.dtsi中存在这个配置，将rk3588s.dtsi中的相关配置，移植到rk3588.dtsi，就可解决eth1不出来的问题，查看rockchip kernel的原始dts，也存在这个情况（没有fe1c0000的配置），不确定以后将imb3588的源码合并到主线，对其他rk3588设备有多大影响。
3、该设备目前还有个问题，imb3588所使用的无线模块为ap6275p，与源码库中，armsom_sige7及yijiahe_jm10设备相同，参考这两个设备，在rk35xx.mk中添加了kmod-rkwifi-bcmdhd-pcie rkwifi-firmware-ap6275p的配置，编译成功后的固件有无线入口，但是没有无线信号，搜索也不能搜索到信号，如下图

[image]
[image]

查看系统内核日志，其中config_bcm43752a2_pcie_ag.txt未在驱动中，内核日志如下
`[    6.553150] [dhd-pcie] _dhd_module_init: in Dongle Host Driver, version 101.10.361.25 (wlan=r892223-20220929-2)
[    6.553150] /home/gong/istoreos22/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/rkwifi-pcie/rkwifi-2023-12-06/rkwifi/bcmdhd compiled on May 14 2025 at 02:37:27
[    6.553150] 
[    6.553175] [WLAN_RFKILL]: rockchip_wifi_get_oob_irq: Enter
[    6.553211] [dhd-pcie] dhd_wlan_init_gpio: WL_HOST_WAKE=-1, oob_irq=128, oob_irq_flags=0x4
[    6.553214] [dhd-pcie] dhd_wlan_init_gpio: WL_REG_ON=-1
[    6.553218] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.25 (wlan=r892223-20220928-2)
[    6.553354] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 8138752(7948K)
[    6.553358] [dhd-pcie] dhd_wifi_platform_load: Enter
[    6.553483] [dhd-pcie] dhdpcie_pci_probe : no mutex held
[    6.553486] [dhd-pcie] dhdpcie_pci_probe : set mutex lock
[    6.553490] [dhd-pcie] PCI_PROBE:  bus 21, slot 0,vendor 14E4, device 449D(good PCI location)
[    6.553496] [dhd-pcie] dhdpcie_init: found adapter info 'DHD generic adapter'
[    6.553499] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 3, size 139264
[    6.553519] [dhd-pcie] succeed to alloc static buf
[    6.553522] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 4, size 0
[    6.553547] pcieh-pcie 0002:21:00.0: enabling device (0000 -> 0002)
[    6.553882] [dhd-pcie] Disable CTO
[    6.554544] [dhd-pcie] ******** Perform FLR ********
[    6.630654] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.630726] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.630785] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.630843] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.630901] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.630958] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631015] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631072] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631129] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631186] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631244] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631301] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631358] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631416] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631473] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631530] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631589] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631646] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631703] [dhd-pcie] read_config: reg=0x88 read val=0xb080
[    6.631762] [dhd-pcie] read_config: reg=0x88 read val=0x8080
[    6.632315] [dhd-pcie] ******** FLR Succedeed ********
[    6.632642] [dhd-pcie] DHD: dongle ram size is set to 1310720(orig 1310720) at 0x170000
[    6.632655] [dhd-pcie] dhdpcie_bar1_window_switch_enab: bar1_switch_enab=0 ramstart=0x170000 ramend=0x2affff bar1_size=0x0
[    6.632688] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 7, size 34688
[    6.632720] [dhd-pcie] dhd_conf_set_chiprev : devid=0x449d, chip=0xaae8, chiprev=2, svid=0x14e4, ssid=0xaae8
[    6.632769] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 0, size 1776
[    6.634108] [dhd-pcie] dhd_pktid_map_init:2454: pktid_audit init succeeded 1025
[    6.634198] [dhd-pcie] dhd_pktid_map_init:2454: pktid_audit init succeeded 4097
[    6.634692] [dhd-pcie] dhd_pktid_map_init:2454: pktid_audit init succeeded 36865
[    6.635913] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 5, size 65536
[    6.636493] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 19, size 65712
[    6.636551] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 20, size 262144
[    6.636625] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 22, size 65536
[    6.636968] [dhd-pcie] dhd_attach(): thread:dhd_watchdog_thread:c1c started
[    6.636993] [dhd-pcie] dhd_deferred_work_init: work queue initialized
[    6.637007] [dhd-pcie] dhd_tcpack_suppress_set: TCP ACK Suppress mode 0 -> mode 3
[    6.637018] [dhd-pcie] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[    6.637048] [dhd-pcie] dhd_cpumasks_init CPU masks primary(big)=0xf0 secondary(little)=0xe
[    6.641469] [dhd-pcie] dhdpcie_bus_attach: making DHD_BUS_DOWN
[    6.641496] [dhd-pcie] dhdpcie_init: rc_dev from dev->bus->self (1d87:3588) is 000000002f3aa8ca
[    6.641612] [dhd-pcie] dhdpcie_init: rc_ep_aspm_cap: 1 rc_ep_l1ss_cap: 1
[    6.641633] [dhd-pcie] dhdpcie_request_irq: INTx enabled, irq=118
[... 138 lines cut ...]
[    8.641964] [dhd-pcie] dhd_bus_start, dhd_bus_init failed -1
[    8.641976] [dhd-pcie] dhdpcie_init: dhd_bus_start() failed
[    8.641989] [dhd-pcie] dhdpcie_bus_remove_prep: making DHD_BUS_DOWN
[... cut ...]
[    8.713681] [dhd-pcie] wifi_platform_bus_enumerate device present 0
[    8.713690] [dhd-pcie] ======== Card detection to remove PCIE card! ========
[    8.713700] [dhd] STATIC-MSG) dhd_static_buf_exit : Enter
[    8.714132] [dhd-pcie] dhd_wlan_deinit_gpio: gpio_free(WL_HOST_WAKE 0)
[    8.714181] [dhd-pcie] _dhd_module_init: Failed to load the driver, try cnt 0
[    8.714208] [dhd-pcie] _dhd_module_init: Failed to load driver max retry reached**
[    8.714221] [dhd-pcie] _dhd_module_init: Exit err=-1`

请教一下大佬这样的问题如何解决

yizhanghong, 2025-05-23:
> 日志中有这个：
> 
> > [   13.209283] rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-1:01] driver [YT8531 Gigabit Ethernet] (irq=POLL)
> > [   13.219720] dwmac4: Master AXI performs any burst length
> > [   13.219755] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
> > [   13.219774] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
> > [   13.219797] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
> > [   13.220994] br-lan: port 1(eth0) entered blocking state
> > [   13.221010] br-lan: port 1(eth0) entered disabled state
> 
> 说明YT8531识别到了，而且加载了驱动。但eth0被加入lan了，你试试接gmac接的网口看会不会有反应。eth1倒是没看到后续，可能已经卡住了。
> 
> 另外，看起来还有两个USB转网口，但是很多报错，而且如果你没配置network应该也不会被系统用上：
> 
> > [   12.806502] usb usb6-port1: Cannot enable. Maybe the USB cable is bad?
> > ...
> > [   13.276430] usb 5-1: new high-speed USB device number 4 using xhci-hcd
> > [   14.847481] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   14.858096] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   15.071378] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.182365] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x0006: -32
> > [   15.182397] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): invalid MAC address, using random
> > [   15.192838] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0006: -32
> > [   15.203376] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0005: -32
> > [   15.214085] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.226188] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.236672] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.247192] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.258068] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   15.268693] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.280696] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   15.291231] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x0001: -32
> > [   15.302133] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.312642] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x001f: -32
> > [   15.323218] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x0019: -32
> > [   15.335230] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x001f: -32
> > [   15.346046] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000d: -32
> > [   15.356634] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000e: -32
> > [   15.367209] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000d: -32
> > [   15.377756] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x000e: -32
> > [   15.390095] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000d: -32
> > [   15.400636] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000e: -32
> > [   15.411164] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000d: -32
> > [   15.421670] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to write reg index 0x000e: -32
> > [   15.432313] ax88179_178a 5-1:2.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x0000: -32
> > [   15.433241] ax88179_178a 5-1:2.0 eth2: register 'ax88179_178a' at usb-xhci-hcd.5.auto-1, ASIX AX88179 USB 3.0 Gigabit Ethernet, 42:b1:dc:d5:60:c3
> > [   15.546208] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   15.557126] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0002: -32
> > [   15.774023] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.887541] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to read reg index 0x0006: -32
> > [   15.887584] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): invalid MAC address, using random
> > [   15.898060] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0006: -32
> > [   15.908587] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0005: -32
> > [   15.920864] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.931569] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
> > [   15.942126] ax88179_178a 5-1:2.1 (unnamed net_device) (uninitialized): Failed to write reg index 0x0001: -32
[... cut ...]
> > [   18.556409] usb 5-1: new high-speed USB device number 5 using xhci-hcd
> > [   20.183273] usb 5-1: new high-speed USB device number 6 using xhci-hcd
> > [   21.753257] usb 5-1: new high-speed USB device number 7 using xhci-hcd
> > [   24.706942] ax88179_178a 5-1:1.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x0040: -32
> > [   25.018057] ax88179_178a 5-1:1.0 eth2: register 'ax88179_178a' at usb-xhci-hcd.5.auto-1, ASIX AX88179 USB 3.0 Gigabit Ethernet, 08:26:ae:33:bb:da
> 
> 总之，YT8531驱动是有的。其他问题可能是dts没配置正确，或者usb有问题。

jjm2473, 2025-05-23:
你还想修改 `rk3588.dtsi`？这是上游内核或者厂商提供的文件，还轮不到我们修改。如果你不会开发，就别在这浪费时间了。如果自己都搞不清楚的文件，pr了我也会拒收。