istoreos/istoreos #2358: 24.10的zerotier会自动停止运行，而且没有接口信息

superfamer, 2025-05-24:
24.10的版本上zerotier运行可能10秒左右就会自动停止，然后就无法在启动，卸载重新安装后依然如此，而且没有接口信息，机器是R2S，这种情况在我另一台JCG Q30 PRO的机器上也遇到过，在那台Q30上用immortalwrt的24.10没有这个问题，如果是原版的openwrt的24.10就会有这个问题，只是Q30不会自动停止，只是没有接口信息，造成无法异地组网

uwence, 2025-05-27:
iStoreOS 24.10.1 2025052311
内核版本
6.6.86  
这个版本我用我试了，可以组网，但是istoreos无法被访问

mrlbing, 2025-05-28:
x86版本的24.10下zerotier也不正常，可以组网，但是无法从内部直接访问外部，也就是没有路由成功

sbwml, 2025-05-30:
> 24.10的版本上zerotier运行可能10秒左右就会自动停止，然后就无法在启动，卸载重新安装后依然如此，而且没有接口信息，机器是R2S，这种情况在我另一台JCG Q30 PRO的机器上也遇到过，在那台Q30上用immortalwrt的24.10没有这个问题，如果是原版的openwrt的24.10就会有这个问题，只是Q30不会自动停止，只是没有接口信息，造成无法异地组网 X86上没有此问题

给你个包（nanopi r2s aarch64_generic），下载解压上传到路由器，`opkg install /路径/zerotier_1.14.2-r1_aarch64_generic.ipk` 安装它。

[zerotier_1.14.2-r1_aarch64_generic.zip](https://github.com/user-attachments/files/20523142/zerotier_1.14.2-r1_aarch64_generic.zip)

sbwml, 2025-06-03:
导致这个问题的原因是 zerotier 于 miniupnpd 兼容问题。

如果是自己编译固件可以使用我用的这个 zerotier patch 修复它：

https://github.com/sbwml/feeds_packages_net_zerotier/blob/bcaa2381df8b47a7aecb3b7a8fd8898a2094b71b/patches/0006-fix-UPNP_GetValidIGD-function-call-for-miniupnpc-2.2.patch

sbwml, 2025-06-03:
版本号是  24.10.x  的话，就能用

lff110, 2025-06-03:
确实不了解补丁的安装方法😭

lff110, 2025-06-04:
能发一个安装补丁的软件吗，我这边找不到那个软件

sbwml, 2025-06-04:
理论上这是中国宝宝能读懂的文字。

[image]

lff110, 2025-06-04:
我是说安装补丁的那个，不会你已经把这个补丁放到上面那个安装包里了吧

sbwml, 2025-06-04:
[image]

我认为补丁和你没有什么关系。

jjm2473, 2025-06-05:
openwrt那边应该已经修复了，从istore卸载重装即可。如果还有问题，执行`opkg info zerotier | grep Version:`确保显示`Version: 1.14.1-r2`或者比这更新。

jjm2473, 2025-06-13:
openwrt已经合并，几天之内应该会编译出新版本，到时候从istore卸载重装即可。新版本是`1.14.1-r3`。