istoreos/istoreos #2360: 【24.10正式版】 已经安装了KSpeeder，但是pull一些镜像会出现中断错误

hugbio, 2025-05-24:
Images: pulling guyskk/rssant:latest... latest: Pulling from guyskk/rssant f7a1c6dad281: Pull complete 92c59ec44e08: Pull complete 08935196cb79: Pull complete 7aa958b41dfb: Pull complete 86c7a279e304: Pull complete 5730f6e2504a: Pull complete 7acd9a665350: Pull complete 9969766a5190: Downloading [============================>                      ]  76.23MB/135.8MB f987367b8966: Waiting 7fc6096e9352: Verifying Checksum 0511622369f5: Waiting ae2bd697888b: Waiting cdc17e08a583: Waiting 0e7691c27b92: Waiting code:200 Get "https://registry-1.docker.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

hugbio, 2025-05-26:
> 应该跟固件版本无关。重新安装KSpeeder试试

卸载了KSpeeder并重新安装后，pull，还是同样的问题。

jjm2473, 2025-05-26:
安装后启用了吗？执行`docker info 2>/dev/null | grep -A5 'Registry Mirrors:'`看看输出什么

closed 2025-05-28 by jjm2473