istoreos/istoreos #2361: R66S  24.10版本  passwall和passwall2  问题

cjw92114, 2025-05-24:
R66S  24.10版本  passwall和passwall2  TCP显示未运行  回退到22.03.7 2025051615  就没问题

bcseputetto, 2025-05-24:
提供更详细的信息，用的什么节点，passwall日志，和节点右侧的节点日志

cjw92114, 2025-05-24:
2025-05-24 22:39:43: 删除 iptables 规则完成。
2025-05-24 22:39:43: 清空并关闭相关程序和缓存完成。
2025-05-24 22:39:44: iptables透明代理基础依赖 iptables-mod-tproxy 未安装...
2025-05-24 22:39:44: iptables透明代理基础依赖 iptables-mod-socket 未安装...
2025-05-24 22:39:44: iptables透明代理基础依赖 iptables-mod-iprange 未安装...
2025-05-24 22:39:44: iptables透明代理基础依赖 iptables-mod-conntrack-extra 未安装...
2025-05-24 22:39:44: TCP节点：[IEPL·深港专线·HK0·空闲·2000M]，监听端口：1041
2025-05-24 22:39:44: UDP节点：与TCP节点相同
2025-05-24 22:39:44:   - Socks节点：[IEPL·深港专线·HK0·空闲·2000M]183.240.116.144:22745，启动 0.0.0.0:1070
2025-05-24 22:39:44: DNS域名解析：
2025-05-24 22:39:44:   - dns2socks(127.0.0.1#15353)，127.0.0.1:1070 -> tcp://1.1.1.1:53
2025-05-24 22:39:44:   - ChinaDNS-NG(127.0.0.1#15354)：直连DNS：192.168.5.2，可信DNS：127.0.0.1#15353
2025-05-24 22:39:44:   - 节点列表中的域名(vpslist)：192.168.5.2
2025-05-24 22:39:44:   - 域名白名单(whitelist)：192.168.5.2
2025-05-24 22:39:44:   - 代理域名表(blacklist)：127.0.0.1#15353
2025-05-24 22:39:44:   - 防火墙域名表(gfwlist)：127.0.0.1#15353
2025-05-24 22:39:44:   - 中国域名表(chnroute)：192.168.5.2
2025-05-24 22:39:44:   - 默认 DNS ：智能匹配
2025-05-24 22:39:44:   - ChinaDNS-NG已作为Dnsmasq上游，如果你自行配置了错误的DNS流程，将会导致域名(直连/代理域名)分流失效！！！
2025-05-24 22:39:45: 开始加载防火墙规则...
2025-05-24 22:39:45:   - [0]追加ISP IPv4 DNS到白名单：192.168.5.2
2025-05-24 22:39:45:   - [0]加入所有IPv4节点到ipset[passwall_vps]直连完成
2025-05-24 22:39:45:   - [0]加入所有IPv6节点到ipset[passwall_vps6]直连完成
2025-05-24 22:39:45:   - [0]追加WAN IP到iptables：192.168.5.1
2025-05-24 22:39:46:   - [0]追加直连DNS到iptables：192.168.5.2:53
2025-05-24 22:39:46:   - 【路由器本机】，屏蔽代理 UDP 端口[443]
2025-05-24 22:39:46:   - 访问控制：
2025-05-24 22:39:47:      - 【默认】，屏蔽代理 UDP 端口[443]
2025-05-24 22:39:47:      - 【默认】，使用 TCP 节点[IEPL·深港专线·HK0·空闲·2000M](TPROXY:1041)
2025-05-24 22:39:47:      - 【默认】，使用 UDP 节点[IEPL·深港专线·HK0·空闲·2000M](TPROXY:1041)
2025-05-24 22:39:47: 防火墙规则加载完成！
2025-05-24 22:39:47: 配置定时任务：自动更新规则。
2025-05-24 22:39:48: 配置定时任务：自动更新【西部世界】订阅。
2025-05-24 22:39:48: 运行完成！

cjw92114, 2025-05-24:
Error loading shared library libssl.so.1.1: No such file or directory (needed by /tmp/etc/passwall/bin/trojan-plus)
Error loading shared library libcrypto.so.1.1: No such file or directory (needed by /tmp/etc/passwall/bin/trojan-plus)
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_shutdown: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_read: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_free: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_default_verify_paths: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get_ex_data_X509_STORE_CTX_idx: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: ERR_clear_error: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_new_mem_buf: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: ERR_func_error_string: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_options: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_alpn_select_cb: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_get_verify_callback: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_ctrl_pending: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_get_verify_mode: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_set_ex_data: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_new: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_select_next_proto: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: CONF_modules_unload: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get_SSL_CTX: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: EVP_DigestInit_ex: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_set_session: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_VERIFY_PARAM_set_flags: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get_error: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: ERR_get_error: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: TLS_method: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: OpenSSL_version_num: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_alpn_protos: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_default_passwd_cb_userdata: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_STORE_CTX_get_ex_data: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_ctrl: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get_shutdown: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_get_ex_data: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set1_param: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_get_default_passwd_cb_userdata: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: EVP_DigestFinal_ex: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: PEM_read_bio_DHparams: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_ctrl: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_sess_set_new_cb: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_new: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get0_alpn_selected: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_connect: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_cipher_list: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_read: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: EVP_MD_CTX_new: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_use_PrivateKey_file: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_new_bio_pair: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_check_ip_asc: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_STORE_CTX_get_current_cert: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_default_passwd_cb: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_STORE_CTX_get_error_depth: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: X509_VERIFY_PARAM_new: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_sess_set_remove_cb: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: ERR_lib_error_string: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_ciphersuites: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_get_ex_data: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: BIO_write: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: OpenSSL_version: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_keylog_callback: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_session_reused: symbol not found
[... cut ...]
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_free: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_use_certificate_chain_file: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: ERR_reason_error_string: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_free: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: SSL_CTX_set_timeout: symbol not found
Error relocating /tmp/etc/passwall/bin/trojan-plus: EVP_DigestUpdate: symbol not found

cjw92114, 2025-05-24:
> 提供更详细的信息，用的什么节点，passwall日志，和节点右侧的节点日志

cjw92114, 2025-05-24:
+0000 2025-05-24 14:53:00 ERROR [1613927621 3ms] dns: exchange failed for n-devs-sgw.tplinkcloud.com.cn. IN A: (dial tcp [2409:8754:10:4:c000::198a]:53417: connect: network is unreachable | dial tcp [2409:8754:10:4:c000::70a]:53417: connect: network is unreachable | dial tcp [2409:8754:10:4:c000::a75a]:53417: connect: network is unreachable)
+0000 2025-05-24 14:53:00 ERROR [1613927621 4ms] dns: process packet connection: (dial tcp [2409:8754:10:4:c000::198a]:53417: connect: network is unreachable | dial tcp [2409:8754:10:4:c000::70a]:53417: connect: network is unreachable | dial tcp [2409:8754:10:4:c000::a75a]:53417: connect: network is unreachable)

cjw92114, 2025-05-24:
[image]

cjw92114, 2025-05-24:
> 提供更详细的信息，用的什么节点，passwall日志，和节点右侧的节点日志

都贴出来了

bcseputetto, 2025-05-24:
> [image]

高级设置里面防火墙怎么没切nftales？

jjm2473, 2025-05-26:
@bcseputetto 建议run直接删掉trojan-plus，甚至ss相关的都可以删掉，这样passwall应该就不会再使用老的实现，反正已经有xray，sing-box这些更好的实现了。

closed 2025-05-28 by jjm2473