istoreos/istoreos #2364: 24.10.1-2025052311 无法进入后台 昨晚全新安装的，今天出现两次了，需要重启才行

hy2023, 2025-05-25:
Unable to render any theme header template, last error was: Unable to compile 'themes/null/header' as Lua template:

/usr/lib/lua/luci/util.lua:645: Unable to establish ubus connection In assert(), file [C] called from function [anonymous function] (/usr/lib/lua/luci/util.lua:645) called from function ((tail call)) called from function get_all (/usr/lib/lua/luci/model/uci.lua:322) called from function [anonymous function] (/usr/lib/lua/luci/config.lua:12) called from function [anonymous function] (/usr/lib/lua/luci/template.lua:32) called from function require ([C]) called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:5) called from function [anonymous function] ([C])
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 55, byte 48:
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:103:32)
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:158:65)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:904:3)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `                this.L.invoke('require', 'luci.ucodebridge');`

jjm2473, 2025-05-26:
你是不是之前安装过其他主题？

执行这段脚本
```bash
uci set luci.themes.Argon=/luci-static/argon
uci set luci.main.mediaurlbase=/luci-static/argon
uci commit luci
```
恢复成默认主题试试。

Rich-xuguo, 2025-05-26:
昨天在线刷的新系统，也是进不去web界面，刚重启那会能登上，界面一开始是路由器界面，刷新两次恢复iStoreOS正常界面，然后半小时左右就再也进不去web界面了，但是挂载的smb还可以局域网进入。求大神救命啊，不想重新刷机。

jjm2473, 2025-05-28:
@Rich-xuguo 路由器界面？你确定不是ip冲突？