istoreos/istoreos #2365: {iStoreOS 24.10.1 2025052311]防火墙-自定义规则

Wakusei-79, 2025-05-25:
请问最新固件下，防火墙类别中自定义规则模块没有了么？

jjm2473, 2025-05-26:
确实没有了。不过24.10已经切换到nftables了，先看看ui上的配置能不能满足需求。之前的自定义规则也是iptables的，不适用。

yacilxyz, 2025-05-30:
只知道iptables的规则代码，以前都是直接复制粘贴到“自定义规则模块”里。ui上的配置，不会调啊？大神们，有没有办法让我继续使用iptables的规则代码？

dxfong, 2025-06-06:
我原来有一条ptables -t nat -A POSTROUTING -o br-lan -j MASQUERADE的规则，现在要怎么转换到 nftables或者恢复到iptables啊，可不可以对新手友好一点

jjm2473, 2025-06-07:
22.03 可以继续使用iptables规则

chxin579, 2025-06-08:
> 我原来有一条ptables -t nat -A POSTROUTING -o br-lan -j MASQUERADE的规则，现在要怎么转换到 nftables或者恢复到iptables啊，可不可以对新手友好一点

问了AI
nftables 版本（等效规则）

nft add table ip nat
nft add chain ip nat postrouting { type nat hook postrouting priority 100 ; }
nft add rule ip nat postrouting oifname "br-lan" masquerade

bistatic, 2025-08-19:
建议增加一个nftables版本的“自定义规则”界面，这样设置方便一点

jjm2473, 2025-08-20:
> 我原来有一条ptables -t nat -A POSTROUTING -o br-lan -j MASQUERADE的规则，现在要怎么转换到 nftables或者恢复到iptables啊，可不可以对新手友好一点

直接勾选防火墙LAN区域的“IP 动态伪装“

closed 2025-08-20 by jjm2473