istoreos/istoreos #2368: uhttpd进程总是挂掉

shixin-679, 2025-05-25:
升级到24.10后，https的web界面时常无法访问，查看/etc/init.d/uhttpd status显示not running，只有restart才恢复。
uhttpd中是配置了443端口和SSL证书的。
[image]

[image]

部分系统日志：
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for test
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for onion
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for localhost
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for local
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for invalid
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for bind
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: using only locally-known addresses for lan
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: read /etc/hosts - 12 names
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Mon May 26 01:42:41 2025 daemon.info dnsmasq[25536]: read /tmp/hosts/odhcpd - 4 names
Mon May 26 01:42:42 2025 cron.err crond[26032]: crond (busybox 1.36.1) started, log level 5
Mon May 26 01:44:06 2025 kern.info kernel: [ 1206.753424] uhttpd[18899]: segfault at 7fb04c74b050 ip 00007fb04c9814d0 sp 00007ffea9305bd8 error 4 in libcrypto.so.3[7fb04c943000+26a000] likely on CPU 0 (core 0, socket 0)
Mon May 26 01:44:06 2025 kern.info kernel: [ 1206.753442] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Mon May 26 01:51:40 2025 kern.info kernel: [ 1660.734120] uhttpd[28123]: segfault at 7fbfce173dd0 ip 00007fbfce3c64d0 sp 00007ffd3a962a88 error 4 in libcrypto.so.3[7fbfce388000+26a000] likely on CPU 3 (core 3, socket 0)
Mon May 26 01:51:40 2025 kern.info kernel: [ 1660.734161] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Mon May 26 01:53:12 2025 kern.info kernel: [ 1753.398682] uhttpd[7629]: segfault at 7f7a072ef290 ip 00007f7a075424d0 sp 00007ffc503061c8 error 4 in libcrypto.so.3[7f7a07504000+26a000] likely on CPU 0 (core 0, socket 0)
Mon May 26 01:53:12 2025 kern.info kernel: [ 1753.398720] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Mon May 26 02:02:52 2025 kern.warn kernel: [ 2332.655083] process '/usr/sbin/ddnstod' started with executable stack
Mon May 26 02:03:06 2025 kern.info kernel: [ 2347.282937] uhttpd[8307]: segfault at 7f2c9daa3b90 ip 00007f2c9dcf64d0 sp 00007fffc2f76e88 error 4 in libcrypto.so.3[7f2c9dcb8000+26a000] likely on CPU 0 (core 0, socket 0)
Mon May 26 02:03:06 2025 kern.info kernel: [ 2347.282963] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
Mon May 26 02:03:32 2025 daemon.err uhttpd[24537]: Error: Error opening /dev/sda: No medium found

部分内核日志：
[   21.007513] br-e12562a2d100: port 1(veth85f8918) entered blocking state
[   21.007516] br-e12562a2d100: port 1(veth85f8918) entered forwarding state
[   21.008653] safeline-ce: port 1(veth36932a4) entered disabled state
[   21.052959] safeline-ce: port 1(veth36932a4) entered disabled state
[   21.053469] veth36932a4 (unregistering): left allmulticast mode
[   21.053473] veth36932a4 (unregistering): left promiscuous mode
[   21.053475] safeline-ce: port 1(veth36932a4) entered disabled state
[   21.176057] safeline-ce: port 1(veth6a87cfc) entered blocking state
[   21.176062] safeline-ce: port 1(veth6a87cfc) entered disabled state
[   21.176069] veth6a87cfc: entered allmulticast mode
[   21.176110] veth6a87cfc: entered promiscuous mode
[   21.176355] safeline-ce: port 1(veth6a87cfc) entered blocking state
[   21.176358] safeline-ce: port 1(veth6a87cfc) entered forwarding state
[   21.547606] eth0: renamed from veth08bf7e4
[   21.617274] safeline-ce: port 1(veth6a87cfc) entered disabled state
[   21.618023] safeline-ce: port 1(veth6a87cfc) entered blocking state
[   21.618027] safeline-ce: port 1(veth6a87cfc) entered forwarding state
[  141.105007] uhttpd[10014]: segfault at 7f39cf46b710 ip 00007f39cf6be4d0 sp 00007fffb2340368 error 4 in libcrypto.so.3[7f39cf680000+26a000] likely on CPU 3 (core 3, socket 0)
[  141.105025] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  274.392959] uhttpd[25233]: segfault at 7f63739dc560 ip 00007f6373c2f4d0 sp 00007ffe437b5328 error 4 in libcrypto.so.3[7f6373bf1000+26a000] likely on CPU 0 (core 0, socket 0)
[  274.392999] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  535.053367] uhttpd[29227]: segfault at 7f357e04ec20 ip 00007f357e2a14d0 sp 00007ffc655503c8 error 4 in libcrypto.so.3[7f357e263000+26a000] likely on CPU 1 (core 1, socket 0)
[  535.053405] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  645.548369] uhttpd[7244]: segfault at 7f4dd439e8c0 ip 00007f4dd45d24d0 sp 00007ffd473a57c8 error 4 in libcrypto.so.3[7f4dd4594000+26a000] likely on CPU 1 (core 1, socket 0)
[  645.548393] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  738.413474] uhttpd[10748]: segfault at 7ff01b944830 ip 00007ff01bb7e4d0 sp 00007fff39653c58 error 4 in libcrypto.so.3[7ff01bb40000+26a000] likely on CPU 1 (core 1, socket 0)
[  738.413515] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  807.179170] uhttpd[12746]: segfault at 7f0a278f8cb0 ip 00007f0a27b4b4d0 sp 00007ffd494b0e88 error 4 in libcrypto.so.3[7f0a27b0d000+26a000] likely on CPU 2 (core 2, socket 0)
[  807.179210] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[  965.705978] uhttpd[16625]: segfault at 7efebc545170 ip 00007efebc7814d0 sp 00007fff841c5728 error 4 in libcrypto.so.3[7efebc743000+26a000] likely on CPU 3 (core 3, socket 0)
[  965.705988] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[ 1206.753424] uhttpd[18899]: segfault at 7fb04c74b050 ip 00007fb04c9814d0 sp 00007ffea9305bd8 error 4 in libcrypto.so.3[7fb04c943000+26a000] likely on CPU 0 (core 0, socket 0)
[ 1206.753442] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[ 1660.734120] uhttpd[28123]: segfault at 7fbfce173dd0 ip 00007fbfce3c64d0 sp 00007ffd3a962a88 error 4 in libcrypto.so.3[7fbfce388000+26a000] likely on CPU 3 (core 3, socket 0)
[ 1660.734161] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[ 1753.398682] uhttpd[7629]: segfault at 7f7a072ef290 ip 00007f7a075424d0 sp 00007ffc503061c8 error 4 in libcrypto.so.3[7f7a07504000+26a000] likely on CPU 0 (core 0, socket 0)
[ 1753.398720] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[ 2332.655083] process '/usr/sbin/ddnstod' started with executable stack
[... cut ...]
[ 2347.282963] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e
[ 2417.107548] uhttpd[24537]: segfault at 7f7dba8f4a70 ip 00007f7dbab474d0 sp 00007ffc1e4916a8 error 4 in libcrypto.so.3[7f7dbab09000+26a000] likely on CPU 1 (core 1, socket 0)
[ 2417.107586] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e

jjm2473, 2025-05-26:
先关闭https，软件包里面看看你之前安装的是哪个`libuhttpd-*ssl`（可能是 `libuhttpd-openssl` ），升级下。之后再打开https

shixin-679, 2025-05-26:
从22.03.7升级在线升级上来的，使用了smartdns，passwall2，Home Assistant，immich，nextcloud，Jellyfin，微信推送ServerChan，smartdns，Lucky（DDNS，WebDAV），Syncthing，还有docker部署的长亭WAF；

目前看来smartdns需要升级包，passwall2需要重新安装，uhttpd的ssl问题需要升级包，其他没遇到啥问题，运行正常；

升级前先修复了系统软件。

jjm2473, 2025-05-26:
luci-ssl-openssl 是个空包，单独升级这个可能没啥用

jjm2473, 2025-05-27:
@shixin-679 不用安装libuhttpd-openssl，应该跟这没关系。

shixin-679, 2025-05-27:
> [@shixin-679](https://github.com/shixin-679) 不用安装libuhttpd-openssl，应该跟这没关系。

嗯。。。还是出现了这个错误，
[image]

这是所有已安装的uhttpd相关的包
[image]

jjm2473, 2025-05-28:
@shixin-679 那试试执行命令`opkg update; opkg install libopenssl3`，然后重启uhttpd `/etc/init.d/uhttpd restart`

jjm2473, 2025-06-01:
那么再执行下 `opkg update; opkg install libustream-openssl20201210`

shixin-679, 2025-06-01:
> > [@shixin-679](https://github.com/shixin-679) 那试试执行命令`opkg update; opkg install libopenssl3`，然后重启uhttpd `/etc/init.d/uhttpd restart`
> 
> 好的，5.29 11:10 我安装上再观察一天

[image]

观察了2天，还是会有这个问题，现在只能写个脚本检测到uhttpd进程not running时就自动restart了

longforus, 2025-08-13:
iStoreOS 24.10.1 2025060614  存在同样的问题

jjm2473, 2025-08-21:
> 大佬我上商店里装了uhttpd这个，然后发现了经常过段时间就进不了后台，需要重启huttpd才行，然后我就在uhttpd界面把唯一的那个main删了，然后就再也进步了后台了。请问怎么恢复呀

/rom/etc/config里面有原始的配置文件

261032568, 2025-09-19:
> [@261032568](https://github.com/261032568) 更新到最新固件，恢复出厂设置，看看有没有问题
不行，还是要报错
Fri Sep 19 17:46:42 2025 kern.info kernel: [150985.885530] uhttpd[26845]: segfault at 7f26bf92c200 ip 00007f26bfb6f4d0 sp 00007ffc7db0e0e8 error 4 in libcrypto.so.3[7f26bfb31000+26a000] likely on CPU 30 (core 7, socket 0)
Fri Sep 19 17:46:42 2025 kern.info kernel: [150985.885554] Code: 00 00 31 c0 5b c3 0f 1f 40 00 31 c0 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 48 89 77 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 <48> 8b 47 40 c3 66 2e 0f 1f 84 00 00 00 00 00 90 89 77 28 c3 66 2e

jjm2473, 2025-09-19:
@261032568 恢复出厂设置什么都不配置，也有这个错误？什么机型？

jjm2473, 2025-09-26:
@longforus 根据其他用户反馈，如果是iOS系统，好像什么浏览器都能触发。电脑上就Safari才会触发。

shixin-679, 2025-09-26:
> 目前主要发现是x86机型，且使用Safari访问https时触发。 如果真正使用uhttpd的https，可以执行以下命令切换到mbedtls版本：
> 
> ```
> opkg update
> opkg install --force-overwrite libustream-mbedtls20201210
> ```

9.26 20:00我安装了，观察2天看看

shixin-679, 2025-10-01:
[image]

换成libustream-mbedtls20201210   5天后，没有报错了。