istoreos/istoreos #2372: [24.10] docker流水日志一直出现一个err错误

nanshihui, 2025-05-27:
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.099785364+08:00" level=warning msg="path in container /dev/dri already exists in privileged mode" container=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 spanID=68506636fd20279f traceID=b3762411b0ebfeb032c2d6bafc7b4361
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.221428135+08:00" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.221681015+08:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.221740517+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.222141859+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:22:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:22:02.531591971+08:00" level=warning msg="cleaning up after shim disconnected" id=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 namespace=moby
Tue May 27 11:23:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:02.679486206+08:00" level=warning msg="path in container /dev/dri already exists in privileged mode" container=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 spanID=67a82fb6c6b5d28e traceID=633af107b90d188a48ea01771c18fbfa
Tue May 27 11:23:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:02.802937922+08:00" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Tue May 27 11:23:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:02.803271305+08:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue May 27 11:23:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:02.803341598+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:23:02 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:02.803790775+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:23:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:23:03.123486007+08:00" level=warning msg="cleaning up after shim disconnected" id=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 namespace=moby
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.249383433+08:00" level=warning msg="path in container /dev/dri already exists in privileged mode" container=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 spanID=76826068c8bc78a7 traceID=a7e1250a1715c294c3af0940e8c8a58a
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.338324595+08:00" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.338577184+08:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.338638436+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.339034236+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.619251281+08:00" level=warning msg="cleaning up after shim disconnected" id=dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35 namespace=moby

nanshihui, 2025-05-27:
dfc1f44628a490aa8c0f74ca60fc80b3009af6738ac9ea69ec37374a79f54a35  是emby

jjm2473, 2025-05-28:
emby在不断重启吗？

nanshihui, 2025-05-28:
除了emby外
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.338577184+08:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.338638436+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Tue May 27 11:24:03 2025 daemon.err dockerd[9166]: time="2025-05-27T11:24:03.339034236+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1

这个日志也一直在加载。