istoreos/istoreos #2381: iptable或nftable的性能存在问题

HfSrJun, 2025-05-30:
在之前的issue #2011 中我提到过我的内网在通过特定路径访问时存在性能问题，在近期问题复现，并且我找到了问题的大概所在。

网络拓扑大概如下：光猫连接路由器（192.168.1.1）拨号，路由器上运行dockers容器。通过域名访问内网的webui（有些在路由器docker内，有些在其他设备）会先进入路由器的nginx容器，再根据子域名反代到内网ip+端口。

旧的问题总结就是，在内网，走nginx容器反代访问webui会很卡（建立连接的耗时久），直接用内网ip访问没问题。将nginx容器从bridge网络改到host网络则问题解决，或者nginx转发到容器IP（172.17.0.x）而非内网ip（192.168.1.1）也可以解决。**可以看出问题出现在路由器的LAN→Docker的转发**。

近期网络出现新的问题：我在内网设备上挂了大量的磁力下载（大约2000个磁链），制造了大量的链接。**随着挂下载的时间推移问题会加重**，不挂下载则不会出现问题（或者说问题发展缓慢），**重启路由器问题就会还原**。
问题表现为：**路由器LAN→Docker的路径访问十分缓慢**，且**不同设备的表现不同**。经过tcpdump抓包，发现在请求方发起第一次握手后，**主机的第二次握手要延迟很久才回复**。
“不同设备表现不同”具体而言，是**不同ip的表现不同**，而且**联网时间越久的ip状态越差**。将联网很久，访问变缓慢的电脑更改ip，访问速度显著提升。此时将另一台刚刚联网，访问很快的电脑改成刚才的ip，速度明显下降。

经过总结以及AI的分析，问题很可能出现在iptable或者nftable上。

HfSrJun, 2025-05-30:
性能问题似乎不止出现在路由器LAN→Docker，因为还有一个持续了数个月的问题。
我在内网的服务器上开了MC的服务器，通过外网访问时的效果很差，而之前的几年从未出现类似问题。通过排查后，发现用一直固定在内网的服务器以及电脑开服，连接问题存在，而不常在家的两台笔记本开服，就没有连接问题。与上面所说的ip随时间推移连接效果恶化是相符的。
但比较不同的是，我将内网服务器的流控算法从cubic改为bbr，网络问题得到了很大的缓解，原因不知。

jjm2473, 2025-06-06:
你的固件是什么版本？

这些恶化的ip，流量正常吗，有没有大量请求发到这些ip。另外，可以试试关掉防火墙中的“软件流量卸载”。

HfSrJun, 2025-06-06:
> 你的固件是什么版本？
> 
> 这些恶化的ip，流量正常吗，有没有大量请求发到这些ip。另外，可以试试关掉防火墙中的“软件流量卸载”。

固件保持更新，现在是最新的24.10版。我相信这个问题已经存在半年以上了
恶化的ip没有明显规律性，最明显的是运行于路由器docker中的dashy面板，此外也在笔记本电脑和台式机上观察到了类似现象
此外，昨天发现我在端口转发和防火墙规则中重复配置了一条规则（bt下载的端口，upnp打不通所以手动），规则重复会不会导致问题？我还没来得及对比测试

jjm2473, 2025-06-06:
你就按我说的配置试试，有问题再说

jjm2473, 2025-06-06:
有检查CPU/内存使用率吗？

HfSrJun, 2025-06-06:
> 有检查CPU/内存使用率吗？

这种比较基础的问题当然是第一时间检查了。虽然机器确实很老但对于路由器来说也绰绰有余，E3 1220 v3（之前也用过1275V3）+16G，CPU占用最多20%多一些
也考虑过会不会是cpu单核满载，但是命令行里观察到四个核心均有占用，且网卡中断在不同核心的分布也比较均匀

HfSrJun, 2025-06-06:
> 如果确实是连接数用光导致的问题，可以临时用这个命令`sysctl -w net.netfilter.nf_conntrack_max=131071`增大。

连接跟踪表是排查过的，DeepSeek的解答最核心的就是连接数问题。
机器的上限默认是65535，在命令行里观察，连接数在两万几千到一万几千之间波动

HfSrJun, 2025-06-06:
> 如果出问题时没满那应该不是这个原因。防火墙的Fullcone-NAT和软件流量卸载都关掉试试

全锥NAT没记错是这个版本加的，在升级24.10之前就有我说的问题了，关掉之后有一些若有若无的好转（延迟为几百到一千多，对内网来说还是很大）。软件卸载一直也没开，开了也没什么影响。
另外这个问题是随着开机时间而恶化的，现在开机一个半小时已经可以看到现象了。我现在并没有挂那么多bt，只是开着迅雷下几十个慢速种子，上传下载都是各位是MBps，依然会在一定程度上触发这个问题

jjm2473, 2025-06-06:
全锥NAT早就有了，21版本都加上了

HfSrJun, 2025-06-06:
> 全锥NAT早就有了，21版本都加上了

那我可能是之前没注意，大概是24版本的更新日志提了一嘴所以注意到了。
另外在23年底改用openwrt之前用的都是爱快，内网的东西也简单。前文提到的mc服务端连接问题好像也是伴随着迁移产生的，只不过大部分时间很不明显。会不会也是NAT类型问题

jjm2473, 2025-06-07:
全锥NAT可以自己关掉试试

closed 2025-10-16 by jjm2473