istoreos/istoreos #2383: r2c新系统24.10.1下WAN口失效

krkrchan, 2025-05-31:
在R2C中使用rufus全新安装系统R2S的24.10.1，未保留或恢复备份。
发现进入系统后，界面显示WAN口已断开，同时机器上WAN口灯未亮。LAN口正常。

重新安装22.03.7，发现WAN口工作正常。

krkrchan, 2025-06-07:
更新：使用rufus全新安装系统R2S的iStoreOS 24.10.1 2025060614，发现WAN口仍旧失效，前面板LAN口绿灯亮，WAN口灯不亮。网线插口LAN口灯亮，WAN口全都不亮。

[image]

nuka-so, 2025-06-09:
帮你提问了一下，好像是官方的问题
https://github.com/istoreos/istoreos/discussions/2411#discussioncomment-13398884

krkrchan, 2025-07-08:
iStoreOS 24.10.2 2025070410 内核日志

[core.log](https://github.com/user-attachments/files/21123277/core.log)

jjm2473, 2025-07-09:
可能要在uboot中判断机型，让内核根据机型加载不同的dtb文件

krkrchan, 2025-07-09:
使用 rufus 全新刷入 iStoreOS 24.10.2 2025070914，WAN口还是不能用，并且LAN口变成100M的了😂

[image]

[core0709.log](https://github.com/user-attachments/files/21142147/core0709.log)

jjm2473, 2025-07-09:
@krkrchan 看看系统信息是不是识别到R2C

jjm2473, 2025-07-09:
试试这个 https://sourceforge.net/projects/istoreos/files/iStoreOS/r2s/istoreos-24.10.2-2025070923-r2s-squashfs.img.gz/download

如果lan口还是协商到百兆，那执行`lsusb -tv; ethtool eth1`看看输出什么。

krkrchan, 2025-07-10:
还是不行。Rufus全新安装的。

[image]
[image]
[core0710.log](https://github.com/user-attachments/files/21162641/core0710.log)

```
root@iStoreOS:~# lsusb -tv; ethtool eth1
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
    |__ Port 001: Dev 002, If 0, Class=[unknown], Driver=r8152, 5000M
        ID 0bda:8153  
Settings for eth1:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Half 1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes
```

jjm2473, 2025-07-10:
lan口百兆是因为对面设备就是只提供了百兆速率。
```
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: No
```

wan口的问题，你要不先试试openwrt官方的r2c固件是不是正常 https://downloads.openwrt.org/releases/24.10.2/targets/rockchip/armv8/openwrt-24.10.2-rockchip-armv8-friendlyarm_nanopi-r2c-squashfs-sysupgrade.img.gz

jjm2473, 2025-07-11:
试试istoreos 0710，开机时wan口先别插网线，等到前面板的系统指示灯不闪了再插网线，然后把日志提取出来看看

jjm2473, 2025-07-11:
另外，再试试开机以后，wan口插上网线，执行这个命令`ip link set dev eth0 down; sleep 1 ; ip link set dev eth0 up; sleep 1; dmesg|grep eth0| tail -n 10`

jjm2473, 2025-07-11:
还有一个测试方法，就是先重刷istoreos 0710，然后在luci后台不保留配置升级到openwrt 24.10.2，看看wan口是不是正常的。这样可以排查uboot的影响，因为后台升级时不会修改uboot。或者啥时候有空我远程看看。

krkrchan, 2025-07-11:
没懂你说的 0710 是哪个版本

jjm2473, 2025-07-14:
@krkrchan 啥时候有空，让我远程看看

krkrchan, 2025-07-14:
怎么远程呢  @jjm2473

jjm2473, 2025-07-14:
把你的微信号发到 jjm2473@gmail.com ，我加你

jjm2473, 2025-07-15:
下载这个固件测试下，应该没问题了 https://sourceforge.net/projects/istoreos/files/iStoreOS/r2s/istoreos-24.10.2-2025071514-r2s-squashfs.img.gz/download

krkrchan, 2025-07-15:
iStoreOS 24.10.2 2025071514 确实可以。感谢。

[image]
[image]

closed 2025-07-15 by krkrchan