istoreos/istoreos #2392: 在线升级到24.10.1后，重启设备dockerd无法自启动

wz1st, 2025-06-03:
平台：x86

jjm2473, 2025-06-03:
系统-启动项里面看看是不是启用了dockerd

jjm2473, 2025-06-03:
那就看看系统日志，开机时dockerd有没有报错。

还有，确认开机以后dockerd无法启动吗？开机进后台等一分钟还是没启动？`ps ww|grep dockerd`有没有进程？

手动启动dockerd能成功吗？`/etc/init.d/dockerd boot`

wz1st, 2025-06-05:
> 那就看看系统日志，开机时dockerd有没有报错。
> 
> 还有，确认开机以后dockerd无法启动吗？开机进后台等一分钟还是没启动？`ps ww|grep dockerd`有没有进程？
> 
> 手动启动dockerd能成功吗？`/etc/init.d/dockerd boot`

1.dockerd 确认没起来，等了一周都没起来
2.用命令/etc/init.d/dockerd boot 和在web页面/cgi-bin/luci/admin/quickstart都能启动docker
3.只有在23升24存在这个问题(设备是：Dell Inc. Wyse 3040 Thin Client)，在fnos的qemu虚拟机直接安装的24没出现这个问题

wgxatc, 2025-06-07:
> `/etc/init.d/dockerd disable ; /etc/init.d/dockerd enable`重新启用看看。
> 
> 开机如果还是没自动启动，执行`ps ww|grep /etc/init.d/`看看是不是其他启动项卡死了，导致dockerd都还没执行到。

用了大佬的命令，重启一下可以正常启动了，感谢

jjm2473, 2025-07-18:
@wz1st 不要从线上重装docker，openwrt原版的docker是不安全的，可能会把docker的服务开放到wan口去。按这里第2点修复系统软件 https://github.com/istoreos/istoreos/wiki/FAQ